Ysurac/openmptcprouter #1540: Does OpenMPTCProuter do dynamic load balancing for each TCP connection?

xdaviddx, 2021-01-25:
I can see that OpenMPTCProuter does load balancing when doing a speed test over TCP.  Both of my WAN links are used to capacity.  However, it seems that when there is less traffic, it is only putting each TCP connection on one WAN link.  

I tested this by having a Cisco AnyConnect VPN connection to a corporate network and then rebooting one of my WAN devices.  Sometimes this reboot causes the AnyConnect VPN connection to fail until it reconnects on the other WAN link.

Is there a way to configure OpenMPTCProuter to use both links for TCP traffic when the traffic is low?

I see issue [#962](https://github.com/Ysurac/openmptcprouter/issues/962) mentions this same topic.  It was stated in there that a future version would provide this functionality.  If the functionality is there, I'd like to know how to enable it.

It seems like UDP traffic *does* load balancing dynamically in low traffic situations.  I've tested with a softphone application that uses UDP, and I can reboot WAN links without affecting the phone call, other than a momentary distortion in audio quality (presumably from the packets being lost/delayed that were sent to the link that was rebooted).  There is no audio silence or dropped calls.  This is the same experience I'd like for TCP traffic too.

Thanks!

Ysurac, 2021-01-25:
When VPS is active, OpenMPTCProuter doesn't do load balancing. OpenMPTCProuter aggregate links and link with lowest latency is used first.

xdaviddx, 2021-01-25:
Thank you for the prompt reply.

I want to be clear about what I mean by dynamic load balancing.  I mean balancing the aggregated connections.  In other words, for a TCP connection, if you have 2 WAN links, of equal speed, send half the packets for a TCP connection to WAN A and half the packets to WAN B.  If WAN A is twice as fast, send 2/3 of the packets to WAN A and 1/3 to WAN B.  It *seems* to do this for speed tests, unless the speed testing systems are using multiple TCP connections, so it just appears as if it is splitting up intra-TCP connections.  And it seems to do this for UDP traffic, even when the traffic is low.

Is there a way to use multiple links, concurrently, for the same TCP connection, for low traffic situations?  If not, are there any plans to do so?  The combined performance under heavy traffic situations is nice, but real-time resiliency of a link going down is my main purpose for wanting to use OpenMPTCProuter.  I want to use it for a corporate VPN and softphone calls over that corporate VPN.

Thanks again and thanks for creating a great product!

Ysurac, 2021-01-25:
There is Redundant or Round Robin Multipath TCP scheduler.

xdaviddx, 2021-01-25:
Thank you for pointing me in the right direction.  I found the settings page for MPTCP and see where the scheduler algorithm can be changed.

However, after doing more tests, I see that single TCP connections do seem to use both my WAN connections.  Therefore, if one WAN connection goes down, the TCP connection should not go down.  All packets should simply go to the remaining link, since that link now has the lowest latency (the link that is down has infinite latency).

The problem is that I'm not seeing the above behavior.  If I reboot one WAN router, it causes the end user application (CISCO VPN) to lose its connection.  Do you think this is a bug in the MPTCP code, where it is giving an error and causing the TCP connection to be shut down?

Ysurac, 2021-01-25:
As always when there is a problem, check log. It's in State->System log.

Ysurac, 2021-01-25:
And for the VPN, in most case it use UDP. By default, UDP is over a VPN that can me modified in wizard. Some are better in some case.

xdaviddx, 2021-01-25:
Thanks.  I will check the log.

I was assuming it was using TCP, based on the behavior.  I have a lot of things going on in the network, plus the OpenMPTCProuter router is double NAT'd, so it makes it harder to see which traffic is from which LAN machine.  I'll do more testing and try to simplify the network so that I can get more details about which protocols are being used.

xdaviddx, 2021-01-25:
Here is the section of the log from when I rebooted the LTE modem.  This is 192.168.20.1 in the log, on interface eth0.20 (a vlan).

Server 104.238.xxx.xxx is my VPS server.

...
`Mon Jan 25 14:17:47 2021 user.notice post-tracking-post-tracking: wan2 (eth0.20) switched off because ping from 192.168.20.2 error (114.114.114.114,1.1.1.1,4.2.2.1)
Mon Jan 25 14:17:47 2021 daemon.err /usr/bin/ss-local[3552]: getpeername: Socket not connected
Mon Jan 25 14:17:47 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.20.1 dev eth0.20
Mon Jan 25 14:18:02 2021 daemon.err /usr/bin/ss-local[3552]: getpeername: Socket not connected
Mon Jan 25 14:18:12 2021 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact via http 74.82.42.42, 176.103.130.130, 1.0.0.1)
Mon Jan 25 14:18:12 2021 user.notice Shadowsocks: Rules DOWN
Mon Jan 25 14:18:17 2021 daemon.err /usr/bin/ss-local[3552]: getpeername: Socket not connected
Mon Jan 25 14:18:23 2021 daemon.info omr-tracker-ss: Server (104.238.xxx.xxx) seems down, no answer to ping
Mon Jan 25 14:18:43 2021 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 198.27.92.1)
Mon Jan 25 14:18:43 2021 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Mon Jan 25 14:18:43 2021 user.notice Shadowsocks: Rules UP
Mon Jan 25 14:18:43 2021 daemon.err /usr/bin/ss-local[3552]: getpeername: Socket not connected
Mon Jan 25 14:18:43 2021 user.notice Shadowsocks: Reload omr-bypass rules
Mon Jan 25 14:18:43 2021 user.notice omr-bypass: Starting OMR-ByPass...
Mon Jan 25 14:18:48 2021 user.notice post-tracking-post-tracking: wan2 (eth0.20) switched up
Mon Jan 25 14:18:48 2021 user.notice omr-bypass: Reload dnsmasq...
Mon Jan 25 14:18:48 2021 daemon.info dnsmasq[4030]: read /etc/hosts - 4 addresses
Mon Jan 25 14:18:48 2021 daemon.info dnsmasq[4030]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Mon Jan 25 14:18:48 2021 daemon.info dnsmasq-dhcp[4030]: read /etc/ethers - 0 addresses
Mon Jan 25 14:18:48 2021 user.notice omr-bypass: OMR-ByPass is running`
...

Also, is there a link for donating money to the project?  I thought I saw one, but now I can't find it.

xdaviddx, 2021-01-25:
To help clarify my test in the previous message, I have a DSL connection (WAN 1) and an LTE connection (WAN2).  I rebooted the LTE modem, but the DSL connection was still there.  I don't know why this caused the VPS connection to go down.

If there is anything else I can provide to help troubleshoot things, I'm happy to provide it.

xdaviddx, 2021-01-26:
More testing...

I verified that when I reboot the LTE modem, I can't ping the VPS server from the OMR commandline.  I can ping the VPS server at the same time from the DSL router (which is the remaining WAN link).

However, if I reboot the DSL router, then I can still ping the VPS server from the OMR commandline.

I'm not sure where to look for which configuration might cause this.  I have the LTE router set as Master and the DSL router set as Enabled.  Those are the only differences I'm aware of.  If there are other settings I should check, please let me know.

Thanks.

Ysurac, 2021-01-26:
Connection the more stable, so should be DSL, need to be set as master.
What do you have in System->OpenMPTCProuter, "Advanced settings" tab at "Master interface selection" ?

Adorfer, 2021-01-26:
My assumption: the question of this issue is a mixup between "load balancing" and "link aggregation". 
OMR does load balancing as long as there is no VPS (reachable), and as soon as the conneciton to the VPS is established, it uses (differend methods of) link aggregation.

xdaviddx, 2021-01-26:
> Connection the more stable, so should be DSL, need to be set as master.
I am only using the DSL connection temporarily.  Once I can prove that OMR works for my needs, I'll be getting rid of the DSL connection and getting another LTE connection.

Does OMR not work when the master goes down?  I had hoped it would keep sending packets on the non-master.

> What do you have in System->OpenMPTCProuter, "Advanced settings" tab at "Master interface selection" ?
It is set to "Balancing".

xdaviddx, 2021-01-26:
> My assumption: the question of this issue is a mixup between "load balancing" and "link aggregation".

My question was about load balancing *within* a TCP connection.  In other words, balancing the packets across all WAN interfaces when *doing* link aggregation.  I changed the subject to help people understand better.

I assumed it was doing packet balancing, but my results show that when one link goes down, TCP connections are closed.  I'm trying to figure out if that is normal behavior.  If it is not normal behavior, then I need to find how to configure my OMR to keep connections open when one of the WAN links goes down.

Ysurac, 2021-01-26:
Master is the WAN used in priority for initial connection. It's the default route when balancing mode is not used.
When a connection is down this shouldn't put Shadowsocks down. What is the router route table ? (```ip r``` via SSH)

xdaviddx, 2021-01-26:
104.238.xxx.xxx is the VPS address.
192.168.10.1 is the DSL router.
192.168.20.1 is the LTE router (and set as Master).
eth0.10, ethe0.20, and eth0.2 are the vlans I set up.  

I am connected to a managed switch, with the OMR (running on a Raspberry Pi  4 B) having a single ethernet connection to the switch.  Tagging is used on the port that the OMR is connected to on the switch.  No tagging is used on the other ports, so they don't need to be vlan aware.

eth0.2 = The vlan for the LAN side of OMR.  That is tagged in the switch port that OMR is connected to.  Another port is assigned for that vlan too, untagged, and that port is connected to my normal LAN router.  The subdomain is 192.168.2.0.

eth0.10 = The vlan for the DSL router.  This is tagged in the switch port that OMR is connected to.  Another port is assigned for that vlan too, untagged, and that port is connected to the DSL router.  The subdomain is 192.168.10.0.

eth0.20 = The vlan for the LTE router.  This is tagged in the switch port that OMR is connected to.  Another port is assigned for that vlan too, untagged, and that port is connected to the LTE router.  The subdomain is 192.168.20.0.

Here is the information you requested, run through an SSH session to the OMR router.  

root@OpenMPTCProuter:~# ip r
default via 10.255.255.1 dev tun0
default via 192.168.10.1 dev eth0.10 metric 3
default via 192.168.20.1 dev eth0.20 metric 4
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2
10.255.255.2 dev tun0 scope link metric 10
10.255.255.2 dev tun0 scope link metric 1200
104.238.xxx.xxx
        nexthop via 192.168.10.1 dev eth0.10 weight 1
        nexthop via 192.168.20.1 dev eth0.20 weight 10
127.0.0.0/8 dev lo proto static scope link metric 5
192.168.2.0/24 dev eth0.2 proto static scope link metric 8
192.168.10.0/24 dev eth0.10 scope link metric 3
192.168.20.0/24 dev eth0.20 scope link metric 4

If you need screenshots of my interface screens or firewall screens in OMR, I can provide those too.

Ysurac, 2021-01-27:
For now I have all need informations.
I'm making some test to try to reproduce the issue, for now without success.
Maybe an issue that happen only with VLANs, so I'm setting VLANs to test this.

xdaviddx, 2021-01-27:
Okay, thanks.  Please let me know if you'd like me to do any testing.

One thing I had wondered is if there is something I have set up wrong in the interfaces and/or firewall settings for OMR.  However, it can reach the VPS when both WAN links are up, so I'm not sure why it only can't reach the VPS when one is down.  I am not sure what happens if the DSL link is down instead of the LTE link.  I will test that and report back.

The other things I did notice that may be related to this is that the status screen that draws the diagram of the network lists the wrong subnet for the vlan2 interface.  If you look at the screenshot I sent earlier, it shows the following:

Lan address: 192.168.100.1
Lan range:  192.168.100.10 - 192.168.100.13
VLAN2 address:  **192.168.100.1**
VLAN2 range:  192.168.2.10 - 192.168.2.13

I thought it was just a bug in how it was displaying the VLAN2 address.  But if something in the code is confused between the Lan subnet and the vlan subnet, then maybe that is causing problems.

Thanks.

xdaviddx, 2021-01-27:
I happened to run ip r today and there is one thing different from yesterday, even though I haven't made any configuration changes:

root@OpenMPTCProuter:~# ip r
default via 10.255.255.1 dev tun0
default via 192.168.10.1 dev eth0.10 metric 3
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2
10.255.255.2 dev tun0 scope link metric 10
10.255.255.2 dev tun0 scope link metric 1200
104.238.xxx.xxx
        nexthop via 192.168.10.1 dev eth0.10 weight 1
        nexthop via 192.168.20.1 dev eth0.20 weight 10
127.0.0.0/8 dev lo proto static scope link metric 5
192.168.2.0/24 dev eth0.2 proto static scope link metric 8
192.168.10.0/24 dev eth0.10 scope link metric 3
192.168.20.0/24 dev eth0.20 scope link metric 4

It is missing a default route for my LTE router (192.168.20.1) on the vlan interface eth.20.  OMR with the VPS is operational now, however, and sending packets across both WAN interfaces.

xdaviddx, 2021-01-27:
Here is what more testing has shown.

1)  When my LTE router is rebooted, the connection to the VPS goes down (as discussed previously), and it deletes the default route to that router.  Here is the system log showing it going down and coming back.  When it comes back, it doesn't re-add the default route.  However, the system does work when it comes back.  Both WAN links are used.

...
Wed Jan 27 16:41:53 2021 daemon.err /usr/bin/ss-local[16398]: getpeername: Socket not connected
Wed Jan 27 16:41:54 2021 user.notice post-tracking-post-tracking: wan2 (eth0.20) switched off because ping from 192.168.20.2 error (1.2.4.8,80.67.169.40,114.114.114.114)
**Wed Jan 27 16:41:54 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.20.1 dev eth0.20**
Wed Jan 27 16:41:57 2021 daemon.err /usr/bin/ss-redir[16400]: remote recv: Network unreachable
Wed Jan 27 16:41:57 2021 daemon.err /usr/bin/ss-redir[16400]: remote recv: Network unreachable
Wed Jan 27 16:42:03 2021 daemon.err /usr/bin/ss-local[16398]: getpeername: Socket not connected
Wed Jan 27 16:42:05 2021 daemon.err /usr/bin/ss-local[16398]: getpeername: Socket not connected
Wed Jan 27 16:42:08 2021 daemon.err /usr/bin/ss-local[16398]: getpeername: Socket not connected
Wed Jan 27 16:42:18 2021 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact via http 176.103.130.130, 1.0.0.1, 212.27.48.10)
Wed Jan 27 16:42:18 2021 user.notice Shadowsocks: Rules DOWN
Wed Jan 27 16:42:22 2021 daemon.err /usr/bin/ss-local[16398]: getpeername: Socket not connected
Wed Jan 27 16:42:23 2021 daemon.err /usr/bin/ss-local[16398]: getpeername: Socket not connected
Wed Jan 27 16:42:24 2021 daemon.err /usr/bin/ss-local[16398]: getpeername: Socket not connected
Wed Jan 27 16:42:26 2021 daemon.err /usr/bin/ss-local[16398]: getpeername: Socket not connected
Wed Jan 27 16:42:28 2021 daemon.err /usr/bin/ss-local[16398]: getpeername: Socket not connected
Wed Jan 27 16:42:28 2021 daemon.info omr-tracker-ss: Server (104.238.xxx.xxx) seems down, no answer to ping
Wed Jan 27 16:42:49 2021 daemon.err /usr/bin/ss-local[16398]: getpeername: Socket not connected
Wed Jan 27 16:42:52 2021 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 151.101.129.164)
Wed Jan 27 16:42:52 2021 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Wed Jan 27 16:42:52 2021 user.notice Shadowsocks: Rules UP
Wed Jan 27 16:42:52 2021 user.notice Shadowsocks: Reload omr-bypass rules
Wed Jan 27 16:42:52 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed Jan 27 16:42:53 2021 user.notice omr-bypass: Reload dnsmasq...
Wed Jan 27 16:42:53 2021 daemon.info dnsmasq[15874]: read /etc/hosts - 4 addresses
Wed Jan 27 16:42:53 2021 daemon.info dnsmasq[15874]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Wed Jan 27 16:42:53 2021 daemon.info dnsmasq-dhcp[15874]: read /etc/ethers - 0 addresses
Wed Jan 27 16:42:53 2021 user.notice omr-bypass: OMR-ByPass is running
Wed Jan 27 16:42:56 2021 user.notice post-tracking-post-tracking: wan2 (eth0.20) switched up
...

2)  If I reboot my DSL router, the VPS stays up.  However, a different thing happens.  It reports on the status page (where it shows a diagram of current state), VPN is down, for a short while.  

Also, like when the LTE router was rebooted, the default route to the DSL router is deleted and not re-added when the DSL router comes back.

Here is what that looks like in the system log:

...
Wed Jan 27 16:44:29 2021 daemon.err glorytun[10415]: read: Operation timed out
**Wed Jan 27 16:44:29 2021 daemon.info glorytun[10415]: STOPPED tun0**
Wed Jan 27 16:44:29 2021 daemon.err glorytun[10415]: 104.238.xxx.xxx.65001: connected
Wed Jan 27 16:44:35 2021 daemon.info glorytun[10415]: STARTED tun0
Wed Jan 27 16:44:46 2021 user.notice post-tracking-post-tracking: wan1 (eth0.10) switched off because gateway down and ping from 192.168.10.2 error (8.8.8.8,80.67.169.12,8.8.4.4)
**Wed Jan 27 16:44:46 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.10.1 dev eth0.10**
Wed Jan 27 16:46:10 2021 daemon.err /usr/bin/ss-redir[16400]: remote recv: Operation timed out
Wed Jan 27 16:46:12 2021 daemon.err /usr/bin/ss-redir[16401]: remote recv: Operation timed out
Wed Jan 27 16:46:12 2021 daemon.err /usr/bin/ss-redir[16402]: remote recv: Operation timed out
Wed Jan 27 16:46:12 2021 daemon.err /usr/bin/ss-redir[16400]: remote recv: Operation timed out
Wed Jan 27 16:46:19 2021 daemon.err /usr/bin/ss-redir[16400]: remote recv: Operation timed out
Wed Jan 27 16:46:23 2021 daemon.err /usr/bin/ss-redir[16402]: remote recv: Operation timed out
Wed Jan 27 16:46:23 2021 daemon.err /usr/bin/ss-redir[16402]: remote recv: Operation timed out
Wed Jan 27 16:46:25 2021 daemon.err /usr/bin/ss-redir[16402]: remote recv: Operation timed out
Wed Jan 27 16:46:25 2021 daemon.err /usr/bin/ss-redir[16400]: remote recv: Operation timed out
Wed Jan 27 16:46:25 2021 daemon.err /usr/bin/ss-redir[16400]: remote recv: Operation timed out
Wed Jan 27 16:46:25 2021 daemon.err /usr/bin/ss-redir[16400]: remote recv: Operation timed out
Wed Jan 27 16:46:25 2021 daemon.err /usr/bin/ss-redir[16400]: remote recv: Operation timed out
Wed Jan 27 16:46:27 2021 daemon.err /usr/bin/ss-redir[16402]: remote recv: Operation timed out
Wed Jan 27 16:46:33 2021 daemon.err /usr/bin/ss-redir[16400]: remote recv: Operation timed out
Wed Jan 27 16:46:43 2021 daemon.err /usr/bin/ss-redir[16400]: remote recv: Operation timed out
[... cut ...]
default via 10.255.255.1 dev tun0 metric 1200
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2
10.255.255.2 dev tun0 scope link metric 10
10.255.255.2 dev tun0 scope link metric 1200
104.238.xxx.xxx
        nexthop via 192.168.10.1 dev eth0.10 weight 1
        nexthop via 192.168.20.1 dev eth0.20 weight 10
127.0.0.0/8 dev lo proto static scope link metric 5
192.168.2.0/24 dev eth0.2 proto static scope link metric 8
192.168.10.0/24 dev eth0.10 scope link metric 3
192.168.20.0/24 dev eth0.20 scope link metric 4

xdaviddx, 2021-01-28:
One more test.

- Rebooted OMR.
- Rebooted DSL router.

Results:

- The system log shows different results rebooting the DSL router after a fresh reboot of OMR than it does rebooting the DSL router after rebooting the LTE router.  In other words, if my WAN2 is rebooted and comes back, and then WAN 1 is rebooted, that acts differently than if only WAN 1 is rebooted.  The reason the OMR is being rebooted for this test is because it puts the default routes back for each WAN interface (and possibly resets state for other things that I'm unaware of).
- The default routes are put back in this test.  They are not put back when the LTE router is rebooted and then the DSL router is rebooted.

Here is the system log, showing the reboot of the DSL router.  It shows the router going down, a time delay, the router coming up, and then after another delay, the VPN goes down for some unknown reason.  Neither WAN interface went down at that time.  When it rebuilds all the VPN stuff, it also recreates default paths for the 2 WAN interfaces

...
Wed Jan 27 21:02:40 2021 user.notice post-tracking-post-tracking: wan1 (eth0.10) switched off because gateway down and ping from 192.168.10.2 error (8.8.4.4,9.9.9.9,1.0.0.1)
Wed Jan 27 21:02:40 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.10.1 dev eth0.10
Wed Jan 27 21:05:16 2021 user.notice post-tracking-post-tracking: wan1 (eth0.10) switched up
**Wed Jan 27 21:05:42 2021 user.notice post-tracking-post-tracking: omrvpn down because gateway down**
Wed Jan 27 21:05:43 2021 user.notice post-tracking-post-tracking: Glorytun VPN down, restart it
Wed Jan 27 21:05:43 2021 daemon.info glorytun[10727]: STOPPED tun0
Wed Jan 27 21:05:43 2021 daemon.notice netifd: Network device 'tun0' link is down
Wed Jan 27 21:05:43 2021 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Wed Jan 27 21:05:43 2021 daemon.notice netifd: Interface 'omrvpn' is now down
Wed Jan 27 21:05:43 2021 daemon.debug dnsmasq[3182]: stopped listening on tun0(#21): 10.255.255.2 port 53
Wed Jan 27 21:05:43 2021 daemon.notice netifd: Interface 'tun0' has link connectivity loss
Wed Jan 27 21:05:43 2021 daemon.notice netifd: Interface 'tun0' is now down
Wed Jan 27 21:05:43 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Wed Jan 27 21:05:43 2021 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Wed Jan 27 21:05:43 2021 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Wed Jan 27 21:05:43 2021 daemon.info glorytun[30480]: INITIALIZED tun0
Wed Jan 27 21:05:43 2021 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed Jan 27 21:05:43 2021 daemon.notice netifd: Interface 'tun0' is disabled
Wed Jan 27 21:05:43 2021 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed Jan 27 21:05:43 2021 daemon.notice netifd: Interface 'tun0' is enabled
Wed Jan 27 21:05:43 2021 daemon.notice netifd: Network device 'tun0' link is up
Wed Jan 27 21:05:43 2021 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Wed Jan 27 21:05:43 2021 daemon.notice netifd: Interface 'omrvpn' is setting up now
Wed Jan 27 21:05:43 2021 daemon.notice netifd: Interface 'omrvpn' is now up
Wed Jan 27 21:05:43 2021 daemon.notice netifd: Interface 'tun0' has link connectivity
Wed Jan 27 21:05:43 2021 daemon.notice netifd: Interface 'tun0' is setting up now
Wed Jan 27 21:05:43 2021 daemon.notice netifd: Interface 'tun0' is now up
Wed Jan 27 21:05:43 2021 user.notice ModemManager: hotplug: add network interface tun0: event processed
Wed Jan 27 21:05:43 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Wed Jan 27 21:05:43 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan 27 21:05:43 2021 user.notice firewall: Reloading firewall due to ifup of tun0 (tun0)
Wed Jan 27 21:05:43 2021 daemon.err glorytun[30480]: 104.238.xxx.xxx.65001: connected
Wed Jan 27 21:05:43 2021 daemon.info glorytun[30480]: STARTED tun0
Wed Jan 27 21:05:44 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed Jan 27 21:05:45 2021 user.notice omr-bypass: Reload dnsmasq...
Wed Jan 27 21:05:45 2021 daemon.info dnsmasq[3182]: read /etc/hosts - 4 addresses
Wed Jan 27 21:05:45 2021 daemon.info dnsmasq[3182]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Wed Jan 27 21:05:45 2021 daemon.info dnsmasq-dhcp[3182]: read /etc/ethers - 0 addresses
Wed Jan 27 21:05:45 2021 user.notice omr-bypass: OMR-ByPass is running
Wed Jan 27 21:05:45 2021 daemon.debug dnsmasq[3182]: listening on tun0(#22): 10.255.255.2 port 53
**Wed Jan 27 21:05:48 2021 user.notice post-tracking-post-tracking: Set ip route replace default scope global  nexthop via 192.168.10.1 dev eth0.10 weight 1 nexthop via 192.168.20.1 dev eth0.20 weight 2**
Wed Jan 27 21:05:49 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Wed Jan 27 21:05:50 2021 user.notice mptcp: Reloading mptcp config due to ifup of tun0 (tun0)
Wed Jan 27 21:05:56 2021 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Wed Jan 27 21:05:57 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed Jan 27 21:05:58 2021 user.notice omr-bypass: Reload dnsmasq...
Wed Jan 27 21:05:58 2021 daemon.info dnsmasq[3182]: read /etc/hosts - 4 addresses
Wed Jan 27 21:05:58 2021 daemon.info dnsmasq[3182]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
[... cut ...]
        nexthop via 192.168.20.1 dev eth0.20 weight 2**
default via 192.168.10.1 dev eth0.10 metric 3
default via 192.168.20.1 dev eth0.20 metric 4
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2
104.238.xxx.xxx
        nexthop via 192.168.10.1 dev eth0.10 weight 1
        nexthop via 192.168.20.1 dev eth0.20 weight 10
127.0.0.0/8 dev lo proto static scope link metric 5
192.168.2.0/24 dev eth0.2 proto static scope link metric 8
192.168.10.0/24 dev eth0.10 scope link metric 3
192.168.20.0/24 dev eth0.20 scope link metric 4
...

xdaviddx, 2021-02-07:
Checking in on this issue.  Is it something you're still working on?  Is there anything you need from my end to help diagnose?

Thanks.

Ysurac, 2021-02-07:
I've made some changes and I'm making tests with VLAN.

xdaviddx, 2021-02-07:
Thanks for the update, Yannick.

xdaviddx, 2021-03-02:
Hi Yannick,

I see that a new version was released and it mentions VLAN support in the wizard.  Do you think the new version would fix anything related to this issue? (1540)

Thanks.

closed 2021-06-06 by github-actions[bot]