Ysurac/openmptcprouter #1541: Redémarrages/plantages du routeur

mathchab, 2021-01-25:
## Excpectations

Fonctionnement en continu du routeur.

## Current Behavior

Le routeur plante dans les 10 minutes qui suivent l’allumage.
La configuration semble OK, l’accès internet est OK, les 2 WAN sont connectées.

J’ai tenté tous les VPN.
Avec ou sans WAN.
Clean install (reformatage SD).
RPi différent.
Autre version de l’archive.
Autre alimentation.
Je n’ai rien en USB.

Cela se produit depuis un matin, sans que la config physique comme logicielle n’ai été modifiée.

Merci de m’indiquer la démarche pour vous fournir un log...

Merci de votre aide.

## Specifications

  - OpenMPTCProuter version: 0.57.3 (idem sur 0.57.2)
  - OpenMPTCProuter VPS version: 0.1024 5.4.74
  - OpenMPTCProuter platform: RPI3+

Ysurac, 2021-01-25:
Les logs sont dans Etat->Journal système et journal noyau.

mathchab, 2021-01-25:
> Les logs sont dans Etat->Journal système et journal noyau.

Ils ne sont pas remis à zéro au reboot ? Donc perdu après plantage ?
En gros j’attends qu’il plante et au redémarrage je vous copie/colle les 2 pages ?

Merci

Ysurac, 2021-01-25:
Dans Système->Système, onglet journalisation, le log peut être enregistré sur le disque.
Et non on me colle pas les 2 pages, on lit les logs sachant que si il y a quelque chose ça a des chances d'être vers l'heure du plantage.

mathchab, 2021-01-25:
Les voici.

System :
Mon Jan 25 20:52:39 2021 daemon.info unbound: [6323:0] info: service stopped (unbound 1.13.0).
Mon Jan 25 20:52:39 2021 daemon.info unbound: [6323:0] info: server stats for thread 0: 76 queries, 13 answers from cache, 63 recursions, 0 prefetch, 0 rejected by ip ratelimiting
Mon Jan 25 20:52:39 2021 daemon.info unbound: [6323:0] info: server stats for thread 0: requestlist max 23 avg 10.1587 exceeded 0 jostled 0
Mon Jan 25 20:52:39 2021 daemon.info unbound: [6323:0] info: average recursion processing time 0.298767 sec
Mon Jan 25 20:52:39 2021 daemon.info unbound: [6323:0] info: histogram of recursion processing times
Mon Jan 25 20:52:39 2021 daemon.info unbound: [6323:0] info: [25%]=0.111177 median[50%]=0.219738 [75%]=0.454071
Mon Jan 25 20:52:39 2021 daemon.info unbound: [6323:0] info: lower(secs) upper(secs) recursions
Mon Jan 25 20:52:39 2021 daemon.info unbound: [6323:0] info:    0.016384    0.032768 1
Mon Jan 25 20:52:39 2021 daemon.info unbound: [6323:0] info:    0.032768    0.065536 5
Mon Jan 25 20:52:39 2021 daemon.info unbound: [6323:0] info:    0.065536    0.131072 14
Mon Jan 25 20:52:39 2021 daemon.info unbound: [6323:0] info:    0.131072    0.262144 17
Mon Jan 25 20:52:39 2021 daemon.info unbound: [6323:0] info:    0.262144    0.524288 14
Mon Jan 25 20:52:39 2021 daemon.info unbound: [6323:0] info:    0.524288    1.000000 12
Mon Jan 25 20:52:39 2021 daemon.notice unbound: [15415:0] notice: init module 0: iterator
Mon Jan 25 20:52:40 2021 daemon.info unbound: [15415:0] info: start of service (unbound 1.13.0).
Mon Jan 25 20:52:40 2021 daemon.info vnstatd[4242]: Error: Exec step failed (8: attempt to write a readonly database): "insert or ignore into fiveminute (interface, date, rx, tx) values (2, datetime(datetime(1611527700, 'unixepoch'), ('-' || (strftime('%M', datetime(1611527700, 'unixepoch'))) || ' minutes'), ('-' || (strftime('%S', datetime(1611527700, 'unixepoch'))) || ' seconds'), ('+' || (round(strftime('%M', datetime(1611527700, 'unixepoch'))/5,0)*5) || ' minutes'), 'localtime'), 0, 0)"
Mon Jan 25 20:52:40 2021 daemon.info vnstatd[4242]: Error: Fatal database error detected, exiting.
Mon Jan 25 20:52:42 2021 user.notice omr-bypass: Reload dnsmasq...
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq[13254]: exiting on receipt of SIGTERM
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq[16402]: started, version 2.82 cachesize 150
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq[16402]: DNS service limited to local subnets
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq[16402]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq-dhcp[16402]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq[16402]: using only locally-known addresses for domain test
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq[16402]: using only locally-known addresses for domain onion
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq[16402]: using only locally-known addresses for domain localhost
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq[16402]: using only locally-known addresses for domain local
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq[16402]: using only locally-known addresses for domain invalid
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq[16402]: using only locally-known addresses for domain bind
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq[16402]: using nameserver 127.0.0.1#5353
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq[16402]: using only locally-known addresses for domain use-application-dns.net
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq[16402]: using only locally-known addresses for domain lan
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq[16402]: read /etc/hosts - 4 addresses
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq[16402]: read /tmp/hosts/dhcp.cfg01411c - 15 addresses
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq-dhcp[16402]: read /etc/ethers - 0 addresses
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq[16402]: read /etc/hosts - 4 addresses
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq[16402]: read /tmp/hosts/dhcp.cfg01411c - 15 addresses
Mon Jan 25 20:52:42 2021 daemon.info dnsmasq-dhcp[16402]: read /etc/ethers - 0 addresses
Mon Jan 25 20:52:42 2021 user.notice omr-bypass: OMR-ByPass is running
Mon Jan 25 20:52:43 2021 user.notice post-tracking-post-tracking: Check API configuration... Done
Mon Jan 25 20:52:43 2021 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jan 25 20:52:43 2021 user.notice Shadowsocks: Rules DOWN
Mon Jan 25 20:52:44 2021 daemon.err omr-tracker[11159]: Command failed: Not found
Mon Jan 25 20:52:44 2021 user.notice omr-bypass: Restart dnsmasq...
Mon Jan 25 20:52:44 2021 daemon.info dnsmasq[16402]: exiting on receipt of SIGTERM
Mon Jan 25 20:52:45 2021 user.notice Shadowsocks: Rules UP
Mon Jan 25 20:52:45 2021 daemon.info dnsmasq[17409]: started, version 2.82 cachesize 150
Mon Jan 25 20:52:45 2021 daemon.info dnsmasq[17409]: DNS service limited to local subnets
Mon Jan 25 20:52:45 2021 daemon.info dnsmasq[17409]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Mon Jan 25 20:52:45 2021 daemon.info dnsmasq-dhcp[17409]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
[... cut ...]
[   17.737909] IPv6: ADDRCONF(NETDEV_CHANGE): wan1: link becomes ready
[   17.747156] IPv6: ADDRCONF(NETDEV_CHANGE): wan2: link becomes ready
[   18.172074] EXT4-fs error (device mmcblk0p2): ext4_mb_generate_buddy:747: group 1, block bitmap and bg descriptor inconsistent: 12991 vs 12992 free clusters
[   18.194582] EXT4-fs (mmcblk0p2): Remounting filesystem read-only
[   18.206717] EXT4-fs (mmcblk0p2): ext4_writepages: jbd2_start: 2147483646 pages, ino 1002; err -30

Dites-moi si cela suffit, c'est pas simple de tomber au bon moment...
Merci !

Ysurac, 2021-01-25:
Comme c'est indiqué le système de fichier est monté en lecture seule. Ce qui n'est pas bon du tout. Soucis de carte SD ou redémarrage violent.
Sinon rien indique un plantage du routeur.

Kalimeiro, 2021-01-31:
Bonjour,

Cela semble être le même problème que j'ai déjà remonté il y a un moment avec les RPI3B+, pour ma part, et ce n'est que mon avis : OMR ne fonctionne **PAS** avec un RPI3B+ à cause d'un violent bug du noyau/eeprom/driver/jensaisrien de la carte réseau, ce bug est apparu de mémoire depuis la 0.53.x

Comme décrit, le système en lecture seule et normal compte tenu des redémarrages violent, d'ailleurs sur ce sujet l'option "verifier les systèmes de fichiers avant le montage" devait être modifié et activé par défaut dans la 0.57.3 et ce n'est toujours pas le cas, cette option activé par défaut évite que les partitions soient monté en lecture seule lors d'un redémarrage violent, cela clean le système de fichier ... quand il est activé ça fonctionne bien.

mathchab, 2021-01-31:
Bonjour,

Du coup, malgré les réinstallations, impossible de retrouver un système fonctionnel (malgré 2 ans d’utilisation).

J’ai switché sur une VM ESXi 6.7 et ça marche vraiment bien, facilement duplicable avant update (au cas où), etc.

On peut clore le sujet, même si je me doute que quelqu’un d’autre se retrouvera dans cette situation.

Merci !

Ysurac, 2021-01-31:
L'option "verifier les systèmes de fichiers avant le montage" est bien mise par défaut dans la configuration, mais elle est bizarrement supprimée à un moment par le boot sans que j'en sache le responsable.
Pour le RPI3B+, les patches sont ceux par défaut d'OpenWrt maintenant. Je pourrais reprendre les patches officiels venant de raspberry mais ça a déjà posé des soucis et nécessite beaucoup de tests. La solution serait d'ajouter uniquement le patch nécessaire, mais faut le trouver si il existe et arriver à reproduire le soucis qui ne se produit pas systématiquement.

closed 2021-05-07 by github-actions[bot]