Ysurac/openmptcprouter #1542: OMR-Bypass

sonnyxp2008, 2021-01-26:
health to you, i am new and only recently using omptcpr. I would like to know if there is a valid alternative to bypass some sites without changing the dns or mac address. in practice I want to pass netflix on my master regardless of mac or dns. Thanks and sorry for my english

Ysurac, 2021-01-26:
Maybe by bypassing ASN.

sonnyxp2008, 2021-01-26:
can you give me a quick tip on how to set it up?

Ysurac, 2021-01-26:
You need to find AS used by Netflix using peeringdb.com for example and add them in OMR-Bypass ASN section.

sonnyxp2008, 2021-01-26:
and by doing so, all traffic is diverted regardless of the dns entered? EXCELLENT, I try it, thanks

sonnyxp2008, 2021-01-30:
ho provato tutto ma se non inserisco i dns non funziona

sonnyxp2008, 2021-01-30:
i tried everything but if i don't insert the dns it doesn't work

closed 2021-02-01 by sonnyxp2008