Ysurac/openmptcprouter #1543: VPS selection

rifaterdemsahin, 2021-01-28:
i was using hostworld.uk than switched to ovh but the connection started dropping yesterday
now I consider changing but how do I know it is the IPS not my solution ?

[image]

[image]

@Ysurac thanks for the anonymous the IP feature mate...you are a life saver.

rifaterdemsahin, 2021-01-28:
my alternatives
https://www.fasthosts.co.uk/virtual-private-servers

kams19, 2021-01-28:
i am using fasthosts without any issues so far, its been more than 2 months. I have the cheapest VPS £1.20 per month with 512M RAM 1vCPU, it has unlimited data just that the bandwidth is 400M max

rifaterdemsahin, 2021-01-28:
@kams19 thanks for the response. I need a Gigabit connection

rifaterdemsahin, 2021-01-28:
why do started to get these error more and more ?
[image]

Ysurac, 2021-01-29:
As always, read log in State->System log. This will tell you if a problem is detected and what is the problem.

rifaterdemsahin, 2021-01-29:
makes sense
[image]

rifaterdemsahin, 2021-01-29:
I would love emoji support here (: maybe the only aspie in this thread is me

Ysurac, 2021-01-29:
Many "Connection reset by peer"...
You can try another TCP Congestion Control in Network->MPTCP

closed 2021-05-04 by github-actions[bot]