Ysurac/openmptcprouter #1544: upgrading from webui

rifaterdemsahin, 2021-01-29:
i picked the top x64 and flashing.
is that the way to go
[image]
I got this
https://protectli.com/vault-6-port/

rifaterdemsahin, 2021-01-29:
[image]
[image]

[image]

rifaterdemsahin, 2021-01-29:
x86-64 (64-bits)
ext4 image (EFI): openmptcprouter-v0.57.3-r0+15225-bfc433efd4-x86-64-generic-ext4-combined-efi.img.gz (2021/01/05 - 69.40Mo - sha256sum)

Ysurac, 2021-01-29:
As the platform doesn't seems to support EFI, it's not the good image.

rifaterdemsahin, 2021-01-29:
it worked though and I am getting no issues today
which image is the better image?

[image]
monitoring daily ˆˆˆ

**Suggestions**
1.it would be great if we have a daily VPS and OMR fail count in a page
2.If there is another dhcp the OMR should mention it in the console...it is hard to find other DHCPs in the network

spacedream, 2021-02-17:
> Labels **question**

But what is your question was exactly? i not understand.

> it worked though and I am getting no issues today

if your question is not actual anymore please close your ticket, ty.

closed 2021-05-24 by github-actions[bot]