Ysurac/openmptcprouter #1547: omr-test-speed returns "syntax error".

hrmtc, 2021-01-29:
## Expected Behavior
Get a wan speed test using `omr-test-speed`

## Current Behavior

The command returns a syntax error and the selected IP to test appears to be an invalid IP.

```
root@OMR:~# omr-test-speed wan1
Select best test server...
host: speedtest.frankfurt.linode.com - ping: 231
host: speedtest.tokyo2.linode.com - ping: 292
host: speedtest.singapore.linode.com - ping: 359
host: speedtest.newark.linode.com - ping: 147
host: speedtest.atlanta.linode.com - ping: 147
host: speedtest.dallas.linode.com - ping: 165
host: speedtest.fremont.linode.com - ping: 199
host: speedtest.tele2.net - ping: 238
host: www.ovh.net - ping:
host: speed.hetzner.de - ping: 231
host: ipv4.bouygues.testdebit.info - ping: 229
host: par.download.datapacket.com - ping: 233
host: nyc.download.datapacket.com - ping: 144
host: ams.download.datapacket.com - ping: 225
host: fra.download.datapacket.com - ping: 237
host: lon.download.datapacket.com - ping: 228
host: mad.download.datapacket.com - ping: 235
host: prg.download.datapacket.com - ping: 240
host: sto.download.datapacket.com - ping: 236
host: vie.download.datapacket.com - ping: 253
host: war.download.datapacket.com - ping: 254
host: atl.download.datapacket.com - ping: 147
host: chi.download.datapacket.com - ping: 153
host: lax.download.datapacket.com - ping: 197
host: mia.download.datapacket.com - ping: 123
host: nyc.download.datapacket.com - ping: 144
Best server is http://mia.download.datapacket.com/10000mb.bin, running test:
ipset v7.6: Syntax error: cannot parse 195.181.162.27195.181.162.489.187.173.11: resolving to IPv4 address failed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:46 --:--:--     0
curl: (28) Failed to connect to mia.download.datapacket.com port 80: Operation timed out

ipset v7.6: Syntax error: cannot parse 195.181.162.27195.181.162.489.187.173.11: resolving to IPv4 address failed
root@OMR:~#
```

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: 0.1024.5.4.86
  - OpenMPTCProuter platform: x86

hrmtc, 2021-01-29:
Runs fine if I execute `omr-test-speed` not specifying which wan connection.

```
root@OMR:~# omr-test-speed
Select best test server...
host: speedtest.frankfurt.linode.com - ping: 236
host: speedtest.tokyo2.linode.com - ping: 302
host: speedtest.singapore.linode.com - ping: 387
host: speedtest.newark.linode.com - ping: 144
host: speedtest.atlanta.linode.com - ping: 166
host: speedtest.dallas.linode.com - ping: 170
host: speedtest.fremont.linode.com - ping: 176
host: speedtest.tele2.net - ping: 247
host: www.ovh.net - ping:
host: speed.hetzner.de - ping: 240
host: ipv4.bouygues.testdebit.info - ping: 249
host: par.download.datapacket.com - ping: 237
host: nyc.download.datapacket.com - ping: 142
host: ams.download.datapacket.com - ping: 242
host: fra.download.datapacket.com - ping: 235
host: lon.download.datapacket.com - ping: 226
host: mad.download.datapacket.com - ping: 239
host: prg.download.datapacket.com - ping: 245
host: sto.download.datapacket.com - ping: 253
host: vie.download.datapacket.com - ping: 237
host: war.download.datapacket.com - ping: 248
host: atl.download.datapacket.com - ping: 165
host: chi.download.datapacket.com - ping: 160
host: lax.download.datapacket.com - ping: 198
host: mia.download.datapacket.com - ping: 136
host: nyc.download.datapacket.com - ping: 142
Best server is http://mia.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 15 9536M   15 1516M    0     0  4932k      0  0:32:59  0:05:14  0:27:45 5280k
```

Ysurac, 2021-01-29:
Thanks.
This will be solved in next release.

closed 2021-02-01 by Ysurac

Adorfer, 2021-03-04:
same goes for ping/traceroute from UI: would be very helpful to specify the WAN for traceroute,  when WAN seems "no ping to server" despite beeing "online".

Ysurac, 2021-03-04:
Not related to the issue.