Ysurac/openmptcprouter #1549: IPv6 Connection not working.

iamfreaky, 2021-01-29:
## Expected Behavior
ping for IPv6 on OpenMPTCP Router should work, same with clients who got an IPv6 Address from the Router.

## Current Behavior
IPv6 Addresses On the Internet are not Ping able, while the VPS does have a IPv6 Connection. Also the Dashboard says that IPv6 works. Ping from the VPS works (IPv6). Also everything in IPv4 works. 

## Specifications

  - OpenMPTCProuter version: 0.57.3
  - OpenMPTCProuter VPS version: 0.1024
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-01-30:
If VPN is up, a 6in4 tunnel is mounted and ping in IPv6 work.

iamfreaky, 2021-01-30:
Yes Both VPNs IPv4 and 6 are Up, but there is still no IPv6 Access

iamfreaky, 2021-01-30:
[image]
[image]

Or is there a wrong configuration ?
[image]
[image]
[image]

Ysurac, 2021-01-30:
You shouldn't give the /48 prefx to OpenMPTCProuter, this may have conflict with VPS own IP (or you should check that IP are not already the VPS IP).
What is the VPN used ? the default one (glorytun tcp) ?

iamfreaky, 2021-01-30:
So i gave the Router now the /52 Prefix and the Interface /56. 

Default one Glorytun TCP. Its basically a fresh install, both on VPS and OpenMPTCProuter

iamfreaky, 2021-01-30:
Looks like there is some traffic going out. But nothing comes back.

[image]

Ysurac, 2021-01-30:
You can check on VPS if you have answer, using tcpdump for example.

iamfreaky, 2021-01-30:
It Does Appear on the VPS on gt-tun0 but doesnt get a answer

IP6 2605:XXXX:XXXX::2 (Router) > zrh11s03-in-x0e.1e100.net: ICMP6, echo request, seq 289, length 64

iamfreaky, 2021-01-30:
On the Outgoing Device ens3 ( Default Network Adapter) i can see the ping also (request) but no answer.

Ysurac, 2021-01-30:
Who is the VPS provider ?
Does it work with default prefix (fdxx...) ?

iamfreaky, 2021-01-30:
VPS Provider is BuyVM

And that i didnt try. but i will.

iamfreaky, 2021-01-30:
with fdxx it works a bit not everything.

Ysurac, 2021-01-30:
What is not working ?

iamfreaky, 2021-01-30:
Port Opening from the Outside.

Ysurac, 2021-02-01:
You checked https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding#debug on shorewall6 ?

iamfreaky, 2021-02-08:
Check that the VPN is running in System->OpenMPTCProuter and Status tab. ( Is Working )
Check on the VPS in /etc/shorewall6/rules that rules you set are available (Nothing shows up, on the IPv4 Version i can see it.)
Check on the VPS in /etc/shorewall6/params.vpn that OMR_ADDR is the router side VPN IP (10.255.255.2 for glorytun TCP, the default VPN) ( Is Set [FE80 Address)
Check on the router if packets are coming on port 8080 from the VPN for example: tcpdump -i tun0 port 8080 ( didnt see anything even on IPv4 Change. )

iamfreaky, 2021-02-08:
`
\#
\# Shorewall version 4.0 - Sample Rules File for two-interface configuration.
\# Copyright (C) 2006-2014,2007 by the Shorewall Team
\#
\# This library is free software; you can redistribute it and/or
\# modify it under the terms of the GNU Lesser General Public
\# License as published by the Free Software Foundation; either
\# version 2.1 of the License, or (at your option) any later version.
\#
\# See the file README.txt for further details.
\#------------------------------------------------------------------------------
\# For information about entries in this file, type "man shorewall-rules"
\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#\#
\#ACTION         SOURCE          DEST            PROTO   DEST    SOURCE          ORIGINAL        RATE            USER/   MARK    CONNLIMIT       TIME            HEADERS         SWITCH          HELPER
\#                                                       PORT    PORT(S)         DEST            LIMIT           GROUP
?SECTION ALL
?SECTION ESTABLISHED
?SECTION RELATED
?SECTION INVALID
?SECTION UNTRACKED
?SECTION NEW

\#       Don't allow connection pickup from the net
\#
Invalid(DROP)   net             all             tcp
\#
\#       Accept DNS connections from the firewall to the network
\#
DNS(ACCEPT)     $FW             net
\#
\#       Allow Ping from/to the VPN
\#
ACCEPT          vpn             $FW             ipv6-icmp
ACCEPT          vpn             net             ipv6-icmp
ACCEPT          $FW             vpn             ipv6-icmp
\#
\#       Allow Ping from the firewall to the network
\#
ACCEPT          $FW             net             ipv6-icmp
\#
\# Drop Ping from the "bad" net zone.. and prevent your log from being flooded..
\#
\#DROP           net             $FW             ipv6-icmp
ACCEPT          net             $FW             ipv6-icmp
\#
\#       Accept connection from port > 65000 for shadowsocks and glorytun on the firewall
\#
ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535
\#
\#       Accept connection from SSH to the firewall
\#
ACCEPT          net             $FW             tcp     65222
\#
\#       DHCP forward to the VPN from the firewall
\#
ACCEPT          $FW             vpn             udp     53
ACCEPT          vpn             net             udp     53
\#
\#       Redirect all port from 1 to 64999 to the VPN client from the network
\#
\#DNAT           net             vpn:10.0.0.2    tcp     1-64999
`

Ysurac, 2021-02-08:
What is the firewall config you added to set IPv6 port forward ?

iamfreaky, 2021-02-08:
`
config rule
        option name 'Forward-ALL-IPv6-To-pfSense'
        option src 'vpn'
        option dest 'lan'
        list dest_ip '2605:XXXX:IPV6-ADDRESS-FROM-INTERFACE'
        option target 'ACCEPT'
        option family 'ipv6'
`

Ysurac, 2021-02-08:
ports must be specified.

iamfreaky, 2021-02-08:
Like This ? Because this isnt working. 
`config rule
        option name 'Forward-ALL-IPv6-To-pfSense'
        option src 'vpn'
        option dest 'lan'
        list dest_ip '2605:XXXX:IPV6-ADDRESS-FROM-INTERFACE'
        option target 'ACCEPT'
        option family 'ipv6'
        option dest_port '1-64999'
        option src_port '1-64999'
`

Ysurac, 2021-02-08:
It need "src_dport", and this shouldn't be the case. I will fix that for next release.

iamfreaky, 2021-02-08:
if i add src_dport to config rule it also doesnt work. 

But thats actually not the only problem. ICMP Packets are Blocked from Enterring the Server. 
I can Ping google from my OpenMPTCP router instance, but if try to ping from a second vps with vps6 access i cant ping the router. if i try it from the Vps itself it works.

Ysurac, 2021-02-08:
Yes ping is blocked to internal network.
You can edit /etc/shorewall6/rules to allow that if you want.

iamfreaky, 2021-02-08:
Could i also set the port manually open ?

Ysurac, 2021-02-08:
Yes, in /etc/shorewall6/rules on the VPS.

closed 2021-05-15 by github-actions[bot]