Ysurac/openmptcprouter #1551: openmptcprouter sur Overthebox plus

gblanchet03, 2021-01-30:
Je possède une OTB plus. J'ai résilié l'abonnement chez ovh à compter du 1er février 2021.
J'ai téléchargé 
openmptcprouter-v0.57.3-r0+15225-bfc433efd4-x86-64-generic-squashfs-combined-efi.img.gz
qui ressemble au firmware actuel de l'OTB+
Après m'être battu contre grub et l'EFI j'ai réussi à mettre le firmware OMR ci-dessus à partir d'une clef usb.
Mais si eth0 wan1 et wan2 apparaissent le dhcp n'attribue aucune adresse à la machine connectée
Je précise que je vois tout ça par un clavier et du hdmi connectés à l'OTB+
Où est le blemme ??
J'ai ouvert un VPS chez OVH et le script s'est bien déroulé.
Mais je suis bloqué localement.
Je précise que j'ai deux réseaux fibres : freebox pop 192.168.3.254 et livebox5 192.168.1.1 que je connecte soit sur les ports wan13 et wan14 de l'OTB+ mais ça ne change rien. J'ai aussi essayé de désactiver le dhcp sur les deux box.
Cordialement.
Gérard

Ysurac, 2021-01-30:
Il faudrait voir le type d'interface que sont wan1 et wan2, il est possible que ça soit des macvlan, ce qui n'est certainement pas bon pour l'OTB+. A voir dans Réseau->Interfaces puis sur paramètres physiques.

gblanchet03, 2021-01-30:
Je n'ai pas accès à https://192.168.100.1 il semble que Luci ne soit pas lancé.
D'autre part j'ai aussi essayé de mettre mes 2 boxes sur le switch de l'OTB+ et pas sur les wan.
J'ai aussi installé OpenMptcpRouter sur un raspi4 et aussi sur Virtual Box, ça bloque au même endroit 
eth0 a bien pour adresse 192.168.100.1 mais route n'a rien, quelque fois 127.0.0.1 et c'est tout.
Merci je vais attendre le 2 février la fin de mon contrat chez ovh pour investiguer plus.

gblanchet03, 2021-01-30:
Je viens de réussir sur mon raspi4. Effectivement wan1 et wan2 sont bien des MacVlan par défaut.
Bon j'avance sur le raspberrypi. Je verrais mardi sur l'OTB+

gblanchet03, 2021-02-01:
J'ai mis le firmware OMR sur une clef usb et je demande au bios-EFI de l'OTB+ de booter sur la clef.
Il semble qu'il y ait un problème sur eth0. J'ai le message suivant 
r8169 eth0 : no native access to PCI extended config space, falling back to CSI
puis plus loin : flow control off
J'ai rebooté sur le DD sur le firmware OVH et je n'ai pas ce problème.
Je vais attendre qu'OVH me coupe les tunnels.

gblanchet03, 2021-02-03:
J'ai mis une debian 10 sur une clef usb; OMR sur une autre clef et OTB sur le DD en configurant les grub-efi
eth0 n'arrive pas à se connecter sous debian donc c'est le même problème que pour OMR.
Je vais étudier les /etc/config/network des deux Openwrt : l'OTB et l'OMR.
En attendant le switch de l'OTB+ se comporte comme un ... switch
J'ai aussi commandé une carte réseau usb pour essayer de contourner le problème au moins pour la debian

Adorfer, 2021-02-03:
On mine luci works perfect "on the internal network connector of the atom-cpu-pcb", but since the VLAN-switch requires some "magic" for initial config, nothing to see at the frontpanel eths.

gblanchet03, 2021-02-03:
Bon pas de /etc/config/network sur le firmware OMR. Je laisse tomber, ça dépasse mes compétences.
Dommage un firmware OMR adapté à l'OTB+ pourrait intéresser ceux qui ont, comme moi, un otb+ sur les bras.
A 249€ la bête ça fait cher pour en faire juste un switch.

Ysurac, 2021-02-04:
Il y a bien un fichier /etc/config/network sur OMR créé après le premier démarrage.
Je pense que l'OTB+ nécessite une configuration particulière à ce niveau mais sans en avoir un c'est compliqué de tester.
Dans le code d'Overthebox j'ai trouvé ça : https://github.com/ovh/overthebox/blob/master/ext/otb-v2b/files/etc/uci-defaults/20_20-network-v2b

gblanchet03, 2021-02-04:
@Ysurac dit : Il y a bien un fichier /etc/config/network sur OMR créé après le premier démarrage.
J'ai vu ça ce matin
Concernant le code je vais essayer, dans la mesure de mes compétences, de voir ce que je peux faire.
Du coup je me relance.
Merci

gblanchet03, 2021-02-07:
@Ysurac je pense y être arrivé.
Sur les deux ports usb arrière de l'OTB+ une clef usb avec debian 10 (sdb); une clef usb avec OMR x86-64 ext4-efi (sda)
Sur un port usb avant un hub usb avec 3 entrées pour : une clef radio pour un clavier sans fil; une carte réseau usb (eth1) et
un emplacement pour connexion d'un android (usb0). Le 2e port usb avant est masqué par le hub usb.
Un écran connecté sur la prise HDMI
Sur le port 1 du switch câble provenant de la freebox, sur le port 2 un câble vers un switch desservant toutes mes machines du réseau 192.168.3.0 ; sur le port 13 (en bas en principe wan 13) un câble vers la livebox5
Au démarrage quand le bios indique de presser la touche sup le faire pour entrer dans le bios et choisir sur quoi booter.
Bon on boote sur la clef sda (OMR)
On crée un mot de passe pour root puis vi /etc/config/network où on met le lan sur eth1 (puisque eth0 n'est pas reconnu)
On relance le réseau par : /etc/init.d/network restart
On met la carte réseau d'un ordi sur 192.168.100.241 masque 255.255.255.0 passerelle et DNS 192.168.100.1
Puis https://192.168.100.1 (message pour accepter le certificat). On désactive le DHCP pour le lan.
On configure wan1 en adresse fixe 192.168.3.2 passerelle 192.168.3.254
Puis on configure l'accès au VPS en cochant la case "forcer la récupération des paramètres". Bien sûr le VPS doit avoir été préalablement configuré.
Normalement l'etat dans OpenMPTCProuter devrait indiquer que tout marche.
C'est à partir de là qu'il faut :
1) mettre dans /lib/network/ à la place du switch.sh d'origine (le renommer) le switchOTB.txt (en le renommant switch.sh) suivant :
[switchOTB.txt](https://github.com/Ysurac/openmptcprouter/files/5938970/switchOTB.txt)
2) rajouter à la fin de /etc/config/network le contenu suivant
[networkOTB-sanswan14.txt](https://github.com/Ysurac/openmptcprouter/files/5938978/networkOTB-sanswan14.txt)
où le port 14 n'est plus le wan14 mais fait partie du lan (on devrait pouvoir y connecter le câble provenant de la freebox mais je n'ai pas encore essayé)
3) on croise les doigts et on fait dans la console (ou par ssh sur 192.168.100.1) la commande suivante
/etc/init.d/network restart
4) j'ai aussi connecté mon android sur le port usb du hub usb (modem usb) et j'ai ceci par la commande route 
default         10.255.255.1    0.0.0.0         UG    0      0        0 tun0
default         192.168.3.254   0.0.0.0         UG    3      0        0 wan1
default         192.168.1.1     0.0.0.0         UG    12     0        0 eth0.3
default         192.168.42.129  0.0.0.0         UG    14     0        0 usb0
default         10.255.255.1    0.0.0.0         UG    1200   0        0 tun0
10.255.255.1    *               255.255.255.255 UH    0      0        0 tun0
10.255.255.2    *               255.255.255.255 UH    1200   0        0 tun0
51.*.*.162   192.168.3.254   255.255.255.255 UGH   0      0        0 wan1
127.0.0.0       *               255.0.0.0       U     5      0        0 lo
192.168.1.0     *               255.255.255.0   U     12     0        0 eth0.3
192.168.3.0     *               255.255.255.0   U     3      0        0 wan1
192.168.42.0    *               255.255.255.0   U     14     0        0 usb0

J'ai masqué l'IP du VPS. Le tunnel est bien établi mais etat de OpenMPTCProuteur indique une erreur.
eth0.3 c'est wan13
Fing que je possède dans mon réseau m'indique
Download                        Upload   
134.4 Mbps                     95.4 Mbps
et c'est considéré comme étant le réseau OVH ???
Avec Free seul j'avais :
92.6 Mbps                       96,3 Mbps
Et avec OTB amalgamant mes deux box
88.4 Mbps                       95,2 Mbps

Bon je vais essayer de faire un tutoriel.

gblanchet03, 2021-02-07:
Je viens de configurer les deux redirections de ports 22 et 443 entrée tunnel sortie 192.168.100.1
et ça marche à partir de l'adresse de mon VPS donc le tunnel est bien actif.
je pense qu'il doit y avoir un problème pour l'usb. 
Le dashboard m'indique  boardinfo.release is undefined (on définit ça où ?)
et etat m'indique Pas de sortie. Pourtant wan1 est bien coché en vert.

gblanchet03, 2021-02-07:
J'ai commis ça dans la journée 👍 
https://gblanchet03.monsite-orange.fr/page-602007dd51e95.html

gblanchet03, 2021-02-09:
@Ysurac pourriez-vous intégrer ces éléments dans une image spécifique pour l'OTB+
Je déconseille de flasher le firmware de l'OTB+ afin de conserver la possibilité de reprendre un abonnement à OVH, mais de la mettre sur une clef usb préparée par Etcher.
ça marche parfaitement depuis dimanche et j'ai amélioré mes débits. Le wan13 ne doit pas être activé au démarrage, il sert juste de support physique eth0.3 pour wan2, le wan1 étant sur le port 1 (mais pas port 14 ) et support physique eth1, la carte réseau usb qui coûte 13,97 chez CdDiscount est indispensable. Le problème de reconnaissance de l'Eth0 doit être un problème de module du kernel qui doit être spécifique à OVH.
Je n'ai pas réussi le raccordement usb0 à mon smartphone en modem usb, ça perturbe le tunnel.

Ysurac, 2021-02-09:
Difficile de supporter sans avoir le matériel.
Pour eth0, normalement tout est quelque part sur https://github.com/ovh/overthebox
Si il n'y a ensuite qu'un besoin de configuration réseau particulière, je pourrais l'ajouter à l'image.

Adorfer, 2021-02-09:
if needed i can set it up with the stock image and give you ssh access to obtain the neccesary stuff. 
it's sitting here on the shelf since a year.

gblanchet03, 2021-02-09:
@Ysurac @Adorfer sorry my english is near of 0
j'ai regardé 
1- sur otb
/lib/modules/4.14.74/r8169.ko est là
sur github otb CONFIG_package_kmod-r8169=y est présent
dans /etc/modules.d/r8169 est là
dans log du kernel on a r8169 Gigabit Ethernet driver 2.3LK-NAPI loaded

2- sur OMR
/lib/modules/5.4.83/r8169.ko est là
sur github OMR config_package_kmod-r8169=y est présent
dans /etc/modules.d/r8169 est là
Par contre dans le log du kernel le driver 2.3LK-NAPI de r8169 Gigabit Ethernet driver n'est pas chargé (not loaded)

Donc à part le kernel différent 4.14.74 pour OTB et 5.4.83 pour OMR il me semble que c'est cette question de driver qui pose problème.

Mais là j'ai vraiment atteint mes limites. Tel que j'ai bidouillé (là @Adorfer ne va plus rien comprendre) les choses
ça convient à mes besoins.
Dommage j'aurai souhaité partager et aboutir à un firmware propre avec la seule carte réseau eth0
Cordialement
Gérard

gblanchet03, 2021-02-14:
@Ysurac @Adorfer 
J'ai réessayé avec seulement eth0 et comme OTB de mettre le lan en bridge
mais ça ne marche pas.
Du coup j'ai toujours eth1.
J'ai remis actif le wan14 dans /etc/config/network et j'ai réussi le tethering avec mon android GalaxyS10e
J'ai masqué les adresses publiques 👍 
[image]

Adorfer, 2021-02-15:
sorry, i do not get it.
please describe 1) expected behaviour versus 2) observed behaviour. The screenshot you are sharing does not show soemthing i would consider as faulty.

gblanchet03, 2021-02-15:
@Adorfer 
I summarize :
- I didn't take a note when I tried to bridge the lan on the eth0 network card only but nothing worked
- with a usb eth1 network card everything works.
Two elements:
*** /lib/network/switch.sh
*** / etc / config / network
that I can send you privately.
Translated by Madam Google: when I say OK Google it's a female voice.
@Ysurac  je pense qu'on peut clore l'issue. La solution c'est une carte réseau usb eth1
1) mettre le firmware OMR x86_64 ext4 efi sur une clef usb
2) mettre 1ere box (freebox) sur switch port2 et deuxième box (livebox) sur port 13 (wan13) le wan14 étant laissé libre
3) remplir le switch avec tous les appareils du réseau 1 y compris l'ordinateur dont la carte réseau sera configurée en adresse statique 192.168.100.242 masque 255.255.255.0 passerelle et dns 192.168.100.1
4) mettre aussi la carte réseau usb eth1 sur ce switch et sur un port usb à l'avant de l'OTB+
5) dès le boot lancer le setup du bios efi et choisir de booter sur la clef usb 
6) clavier et hdmi en place mettre un passwd pour root
7) se connecter à https://192.168.100.1 sur l'ordinateur 192.168.100.242 et immédiatement décocher autoritaire et ignorer l'interface lan pour éviter pertubations dans le réseau sur le switch
8) aller mettre dans /lib/network le switch.sh qui va bien en sauvegardant l'autre
9) idem avec /etc/config/network 
10) depuis la console directe sur l'OTB+ faire un /etc/init.d/network restart
Voili, voilou !!
Gérard

gblanchet03, 2021-02-15:
En fait il n'y a rien de secret dans ces fichiers. les voici
@Ysurac verra que mettre dans l'image spéciale OTB+à créer

[networkOTB+.txt](https://github.com/Ysurac/openmptcprouter/files/5982211/networkOTB%2B.txt)
[switchOTB+.txt](https://github.com/Ysurac/openmptcprouter/files/5982234/switchOTB%2B.txt)

gblanchet03, 2021-02-17:
@Ysurac 
Tout fonctionne parfaitement sauf que j'ai, le plus souvent lorsque je met mon smartphone en modem usb
ou que je l'enlève, mais aussi de manière aléatoire. C'est la dernière ligne du journal du noyau ci-dessous
Je ne sais pas si c'est lié à l'OTB+

journal du noyau 
[189480.803472] usb 1-2.4: new high-speed USB device number 14 using xhci_hcd
[189481.011650] rndis_host 1-2.4:1.0 usb0: register 'rndis_host' at usb-0000:00:14.0-2.4, RNDIS device, d6:2c:11:61:cd:db
[189481.014742] sdhci-acpi INT33BB:01: GPIO lookup for consumer (null)
[189481.014745] sdhci-acpi INT33BB:01: using ACPI for GPIO lookup
[189481.014749] acpi INT33BB:01: GPIO: looking up gpios
[189481.014753] acpi INT33BB:01: GPIO: looking up gpio
[189481.014755] acpi INT33BB:01: GPIO: looking up 0 in _CRS
[189481.014851] sdhci-acpi INT33BB:01: No GPIO consumer (null) found
[189481.015682] sdhci-acpi INT33BB:01: GPIO lookup for consumer (null)
[189481.015686] sdhci-acpi INT33BB:01: using ACPI for GPIO lookup
[189481.015689] acpi INT33BB:01: GPIO: looking up gpios
[189481.015693] acpi INT33BB:01: GPIO: looking up gpio
[189481.015695] acpi INT33BB:01: GPIO: looking up 0 in _CRS
[189481.015791] sdhci-acpi INT33BB:01: No GPIO consumer (null) found
[189680.583653] traps: logread[2276] general protection fault ip:7f08e41cc4e9 sp:7ffe92a70d30 error:0 in libc.so[7f08e418b000+49000]

Adorfer, 2021-02-18:
for me it's not yet clear how the OTB+ firmware controlls the vlan switch. off course the /etc/config/network deals with the vlan tagging. 
But the mapping between vlans and the physical ports on the PCB of the switch is unclear for me.
[image]

closed 2021-05-24 by github-actions[bot]

hercut, 2021-12-30:
Bonsoir,

Je suis sur une otb it v1.
Pensiez-vous que ça soit le même fonctionnement ?
De mon côté je n'arrive pas du tout à faire quoi que se soit, je ne comprends pas le fonctionnement de l'efi.

Merci de votre aide @gblanchet03

gblanchet03, 2022-01-02:
Bonsoir,
Pourriez-vous dire ceci à hercut :

Il faudrait demander à OVH quel est le bios de otb it v1.
Si ça se trouve il n'y a pas d'EFI et c'est du 32 bits donc l'OMR à mettre
sur clef usb
sera différent de celui que j'ai mis sur l'OTB+ (efi et 64 bits)
Cordialement
Gérard

hercut, 2022-01-03:
Bonjour @gblanchet03 ,

Merci du retour, je suis bien sur un EFi de chez intel, et le système est bien en 64bits.
Peut-être serait-il possible d'échanger en direct d'une manière ou d'une autre avec vous ?

Merci.

gblanchet03, 2022-01-03:
Bonjour,

Je ne sais pas si @Ysurac autorise cet échange direct en donnant à l'un
l'email de l'autre.
Pour moi pas de problème c'est OK
Il faut que votre machine otb puisse accepter un clavier et ait un visuel
sur un écran par hdmi.
Inspirez-vous alors de ceci :
https://slaes.net/Utiliser_OVB+_sans_abonnement.html
J'ai pu abandonné eth1 et configurer br-lan le 29 juin 2021 (voir la fin du
tuto)
Cordialement et bonne année 2021
Gérard