Ysurac/openmptcprouter #1552: Multiple MPTCP over VPN

thothbot, 2021-01-30:
I have multiple LTE WANs and all of them blocks MPTCP

When I enable "MPTCP over VPN" it works for one interface only, 
is it possible to enable "MPTCP over VPN" for multiple WANs?

Ysurac, 2021-01-30:
Not on all wan in current release.
Are you sure all your connections block MPTCP ? It's not blocked so often.

thothbot, 2021-01-30:
I Use Vodafone.de and I have the following on all WANs:

```
1: 3.126.59.111 TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize -TCPOptionMPTCPCapable 

Result:
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 44 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 150 , Protocol = 0x6 , CheckSum = 0x6d24 , SourceIP = 3.126.59.111 , DestinationIP = 100.99.212.87 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 56376 , SeqNumber = 2226634050 , AckNumber = 1870939950 , DataOffset = 6 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 65535 , CheckSum = 0x4d3b , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1460 , >
```

How is it possible to use MLVPN? I mean where can I turn in on? if any chance to get aggregated data

Ysurac, 2021-01-30:
The "TCPOptionMPTCPCapable" on first hop ? strange...
Glorytun UDP and MLVPN have own aggregation system and can be turned on on System->OpenMPTCProuter, advanced settings checkbox.

closed 2021-05-06 by github-actions[bot]