Ysurac/openmptcprouter #1554: Multi USB bonding

Nickmcl024, 2021-01-30:
Hey so have some questions
Can I use a raspberry pi 4 and plug in 8 lte huawei usb Modems
The modem will all be dhcp, 

I have test 2 modem and then used dhcp, but I can get very good speeds is there thing that I need to setup in the interface

Thanks

Ysurac, 2021-01-31:
Please use the template, and give more infos.

Nickmcl024, 2021-02-02:
hey so just put in a second fiber line now 
200dl200up and 200dl100up 
fresh install
set up via wiz all default 
all green on the stats page 

but only getting a max dl200 and up200 

is there any esle to set in the config 

system log 

Tue Feb  2 12:17:25 2021 user.notice ModemManager: hotplug: add network interface bondwan2: event processed
Tue Feb  2 12:17:25 2021 user.notice ModemManager: hotplug: event reported: action=add, name=bondwan2, subsystem=net
Tue Feb  2 12:17:25 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Feb  2 12:17:28 2021 user.notice omr-tracker: Launched
Tue Feb  2 12:17:28 2021 daemon.info vnstatd[7715]: Info: SIGTERM received, exiting.
Tue Feb  2 12:17:28 2021 daemon.info vnstatd[5639]: Info: vnStat daemon 2.6 started. (pid:5639 uid:0 gid:0 64-bit)
Tue Feb  2 12:17:28 2021 daemon.info vnstatd[5639]: Info: Monitoring (2): wan2 (1000 Mbit) wan1 (1000 Mbit)
Tue Feb  2 12:17:28 2021 daemon.info v2ray: DNS disabled: main_dns
Tue Feb  2 12:17:28 2021 daemon.info v2ray: Setting transparent proxy on port: 1897
Tue Feb  2 12:17:28 2021 daemon.info v2ray: Transparent proxy mode: default
Tue Feb  2 12:17:28 2021 user.notice v2ray: add rules
Tue Feb  2 12:17:28 2021 user.notice v2ray: Reload omr-bypass rules
Tue Feb  2 12:17:28 2021 user.notice omr-bypass: Starting OMR-ByPass...
Tue Feb  2 12:17:29 2021 user.notice omr-bypass: Reload dnsmasq...
Tue Feb  2 12:17:30 2021 daemon.info dnsmasq[4231]: read /etc/hosts - 4 addresses
Tue Feb  2 12:17:30 2021 daemon.info dnsmasq[4231]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Tue Feb  2 12:17:30 2021 daemon.info dnsmasq-dhcp[4231]: read /etc/ethers - 0 addresses
Tue Feb  2 12:17:30 2021 user.notice omr-bypass: OMR-ByPass is running
Tue Feb  2 12:17:30 2021 daemon.info v2ray[6764]: V2Ray 4.31.0 (V2Fly, a community-driven edition of V2Ray.) OpenWrt R1 (go1.15.2 linux/arm64)
Tue Feb  2 12:17:30 2021 daemon.info v2ray[6764]: A unified platform for anti-censorship.
Tue Feb  2 12:17:30 2021 daemon.info v2ray[6764]: 2021/02/02 12:17:30 [Info] v2ray.com/core/common/platform/ctlcmd: <v2ctl message>
Tue Feb  2 12:17:30 2021 daemon.info v2ray[6764]: v2ctl> Read config:  /var/etc/v2ray/v2ray.main.json
Tue Feb  2 12:17:33 2021 daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 212.27.48.10)
Tue Feb  2 12:17:35 2021 kern.info kernel: [ 1412.217707] bonding-omrvpn: (slave bondwan1): Enslaving as an active interface with an up link
Tue Feb  2 12:17:35 2021 kern.info kernel: [ 1412.229528] bonding-omrvpn: (slave bondwan1): link status definitely up, 10 Mbps full duplex
Tue Feb  2 12:17:35 2021 kern.info kernel: [ 1412.238088] bonding-omrvpn: active interface up!
Tue Feb  2 12:17:35 2021 daemon.notice netifd: Network device 'bonding-omrvpn' link is up
Tue Feb  2 12:17:35 2021 kern.info kernel: [ 1412.325842] bonding-omrvpn: (slave bondwan2): Enslaving as an active interface with an up link
Tue Feb  2 12:17:35 2021 kern.info kernel: [ 1412.361882] bonding-omrvpn: (slave bondwan2): link status definitely up, 10 Mbps full duplex
Tue Feb  2 12:17:36 2021 user.notice omr-tracker: Launching...
Tue Feb  2 12:17:39 2021 user.notice omr-tracker: Launched
Tue Feb  2 12:17:41 2021 daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 1.0.0.1)
Tue Feb  2 12:17:43 2021 user.notice omr-bypass: Starting OMR-ByPass...
Tue Feb  2 12:17:45 2021 daemon.info dnsmasq-dhcp[4231]: DHCPDISCOVER(eth0) 00:18:ae:8a:4e:e0
Tue Feb  2 12:17:45 2021 daemon.info dnsmasq-dhcp[4231]: DHCPOFFER(eth0) 192.168.100.100 00:18:ae:8a:4e:e0
Tue Feb  2 12:17:45 2021 daemon.info dnsmasq-dhcp[4231]: DHCPDISCOVER(eth0) 00:18:ae:8a:4e:e0
Tue Feb  2 12:17:45 2021 daemon.info dnsmasq-dhcp[4231]: DHCPOFFER(eth0) 192.168.100.100 00:18:ae:8a:4e:e0
Tue Feb  2 12:17:45 2021 user.notice omr-bypass: Reload dnsmasq...
Tue Feb  2 12:17:45 2021 daemon.info dnsmasq[4231]: read /etc/hosts - 4 addresses
Tue Feb  2 12:17:45 2021 daemon.info dnsmasq[4231]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Tue Feb  2 12:17:45 2021 daemon.info dnsmasq-dhcp[4231]: read /etc/ethers - 0 addresses
Tue Feb  2 12:17:45 2021 user.notice omr-bypass: OMR-ByPass is running
Tue Feb  2 12:17:46 2021 daemon.err omr-tracker-v2ray[8664]: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Tue Feb  2 12:17:46 2021 daemon.info omr-tracker-v2ray: Reload V2Ray rules
Tue Feb  2 12:17:47 2021 daemon.info dnsmasq-dhcp[4231]: DHCPDISCOVER(eth0) 00:18:ae:8a:4e:e0
Tue Feb  2 12:17:47 2021 daemon.info dnsmasq-dhcp[4231]: DHCPOFFER(eth0) 192.168.100.100 00:18:ae:8a:4e:e0
Tue Feb  2 12:17:49 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.248.1 dev bonding-omrvpn
Tue Feb  2 12:17:51 2021 user.notice omr-tracker: Launching...
Tue Feb  2 12:17:54 2021 user.notice omr-tracker: Launched
Tue Feb  2 12:17:55 2021 daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 1.0.0.1)
Tue Feb  2 12:18:01 2021 user.notice MPTCP: No master multipath defined, setting it to wan1
Tue Feb  2 12:18:02 2021 daemon.err openvpn(omr_bonding_wan1)[4971]: event_wait : Interrupted system call (code=4)
Tue Feb  2 12:18:02 2021 daemon.err openvpn(omr_bonding_wan2)[4944]: event_wait : Interrupted system call (code=4)
Tue Feb  2 12:18:02 2021 kern.info kernel: [ 1438.833044] bonding-omrvpn: (slave bondwan1): Releasing backup interface
[... cut ...]
Tue Feb  2 12:28:57 2021 daemon.info dnsmasq-dhcp[31183]: DHCPOFFER(eth0) 192.168.100.100 00:18:ae:8a:4e:e0
Tue Feb  2 12:29:19 2021 daemon.info dnsmasq-dhcp[31183]: DHCPDISCOVER(eth0) 00:18:ae:8a:4e:e0
Tue Feb  2 12:29:19 2021 daemon.info dnsmasq-dhcp[31183]: DHCPOFFER(eth0) 192.168.100.100 00:18:ae:8a:4e:e0
Tue Feb  2 12:29:22 2021 daemon.info dnsmasq-dhcp[31183]: DHCPDISCOVER(eth0) 00:18:ae:8a:4e:e0
Tue Feb  2 12:29:22 2021 daemon.info dnsmasq-dhcp[31183]: DHCPOFFER(eth0) 192.168.100.100 00:18:ae:8a:4e:e0

closed 2021-06-06 by github-actions[bot]