Ysurac/openmptcprouter #1555: Latence sur smartphone sur internet

JeedHome44, 2021-01-31:
## Expected Behavior

Depuis plus d'une 15ène de jours, les membres de ma famille avons des latences sur nos smartphones en wifi.
Je n'ai rien changé à mon LAN et rien aux AP.
Mes box domotiques et autres appareils fonctionnent correctement.

J'ai l'impression que c'est OMR qui ne transmet pas correctement les sites aux smartphones.
J'ai fait des tests de débit et ils sont bons.
J'ai testé de by-passer sur l'une ou l'autres de connexions et ca fait pareil.

Pourtant, Netflix ou autre charge très vite avec le by-pass.

## Current Behavior

Y a-t-il quelque chose sur OMR qui pourrait faire un ralentissement sur les sites web?
J'ai même testé de désactivé Pi-hole et ça ne change rien.

Est-ce la conjoncture actuelle qui rend les data center surchargé et qui perturbe internet comme on le connaissait avant, avec moi de même en temps normal dessus?
Télétravail à gogo et tout peuvent-ils en être la cause?

Je cherche un moyen d'identifier le problème.
OMR fournit-il un diagnostic réseau?

Si oui, comment l'utiliser?
Merci

## Specifications

  - OpenMPTCProuter version:  57.3
  - OpenMPTCProuter VPS version:  1024
  - OpenMPTCProuter platform:  Pi 4B 2Go

Kalimeiro, 2021-01-31:
Il faut aller du côté de OMR-DSCP qui de base possède quelques règles de classification mais mal définie... (à mon sens...) en tout cas j'ai modifié comme suit pour obtenir quelque chose de plus correcte et qui me convient :

- Protocole ICMP (ping) = Classe 7 - Latency Sensitive
- Protocole UDP/TCP pour les ports DNS/NTP (d'ailleurs, il manque le 5053, 5054, 5353) = Classe 4 - Real-Time Interactive
- Les deux derniers concernant les ports 65xxx doivent rester en Classe 6 - Routage réseau.

Ensuite il y a l'onglet DSCP Domaines, qui possède quelques domaines pré-configurés, là aussi je trouve que les Classes sont mal définies, et, il ne peut pas y avoir tout les domaines... d'ailleurs je cherche une liste si ça existe...

JeedHome44, 2021-01-31:
Merci beaucoup pour ces précisions. J'avais que OMR me dépasse beaucoup.
J'ai compris la base mais c'est sur, je vais me renseigner plus en détail de tout ce qui le compose.

Petite question. Quand on va dans Système > OpenMPTCProuteur > Plus de paramètres, il y a une option pour gérer la latence plutôt que la bande passante. Que faut-il faire avec cette option?

Kalimeiro, 2021-01-31:
la laissez cocher.

kevin39, 2021-01-31:
Pour une utilisation personnelle standard vous n'êtes pas censé toucher à ces paramètres à mon sens.
J'ai bien peur que @JeedHome44  s'attire plus d'ennuis qu'autre chose en jouant avec OMR-DSCP, pense surtout aux backups ! Si tu fais des backups, tu peux toucher à tout sans risque :)

Ysurac, 2021-01-31:
Il y a peu de risques de toucher à OMR-DSCP mais ça ne change pas forcément grand chose.
Pour ce qui est des latences il faudrait voir si ça vient des requêtes DNS (majoritairement en UDP), ou une latence en TCP.
Il doit être possible de voir ça avec les options développeur du navigateur, mais je ne sais pas si il y en a sur smartphone. 
Cependant, si la latence est présente que via wifi et pas autrement, le soucis est à voir à ce niveau. Le soucis peut être au niveau de l'AP ou d'un soucis de fréquence surchargée.

Kalimeiro, 2021-02-01:
> 
> 
> Pour une utilisation personnelle standard vous n'êtes pas censé toucher à ces paramètres à mon sens.
> J'ai bien peur que @JeedHome44 s'attire plus d'ennuis qu'autre chose en jouant avec OMR-DSCP, pense surtout aux backups ! Si tu fais des backups, tu peux toucher à tout sans risque :)

Oui évidemment, je suis un amateur qui fait n'importe quoi cela va de soit...

kevin39, 2021-02-01:
Hola.... On va se calmer, ici on s'entre-aide calmement :)

Et je ne parlais même pas à toi...

closed 2021-04-21 by JeedHome44