Ysurac/openmptcprouter #1556: Glorytun-tcp restarts / reconnect all 6 sec.

homicida2000, 2021-01-31:
## Expected Behavior

Glorytun restarts every 6 sec - don't know why

I've 3 WAN Connections
1. VDSL 100
2. VDSL 100
3. LTE

On the OMR there is that output from dmesg
`[72321.562956] __mptcp_init4_subsockets: token 0xb651631f bind() to 10.255.255.2 index 9105 failed, error -99
[72474.932266] __mptcp_init4_subsockets: token 0x7259ad61 bind() to 10.255.255.2 index 9127 failed, error -99
[72489.309112] __mptcp_init4_subsockets: token 0x5d29b04b bind() to 10.255.255.2 index 9129 failed, error -99
[72608.729263] __mptcp_init4_subsockets: token 0x9da6797c bind() to 10.255.255.2 index 9145 failed, error -99
[72645.845831] traps: logread[28386] general protection fault ip:7f7c131b24e9 sp:7ffcfa9e6bb0 error:0 in libc.so[7f7c13171000+49000]
[72714.217800] __mptcp_init4_subsockets: token 0x19ee80f1 bind() to 10.255.255.2 index 9159 failed, error -99
[72778.783706] __mptcp_init4_subsockets: token 0x75f5f553 bind() to 10.255.255.2 index 9168 failed, error -99
[73064.146108] __mptcp_init4_subsockets: token 0x9d39510b bind() to 10.255.255.2 index 9207 failed, error -99
[73455.511752] __mptcp_init4_subsockets: token 0x15b9b55 bind() to 10.255.255.2 index 9260 failed, error -99
`
10.255.255.2 is the glorytun dev

PS output :

> 
'
 5996 root      4264 S    /sbin/rngd -f -r /dev/urandom
 6198 root         0 IW   [kworker/1:2-eve]
 6224 root      1156 S    /usr/sbin/crond -f -c /etc/crontabs -l 5
 6325 root      3968 S    /usr/sbin/snmptrapd -Lf /dev/null -f
 6979 root      3072 S    /usr/sbin/uhttpd -f -h /www -r OpenMPTCProuter -x /cgi-bin -u /ubus -t 60 -T 30 -k 20 -A 1 -n 3 -N 100 -R
 7219 root      1332 S    /usr/bin/dbus-daemon --system
 7332 root      2232 S    /usr/sbin/vnstatd --nodaemon
 7399 root     11264 S    /usr/sbin/ModemManager
 7426 root         0 Z    [sh]
 7684 root      1008 S    /sbin/blockd
 8739 root         0 IW   [kworker/u4:2-fl]
 9894 root         0 IW   [kworker/0:0-rcu]
13910 root         0 IW   [kworker/0:2-eve]
14318 root       928 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
14362 root      4648 S    /usr/sbin/usbmuxd --systemd
15792 root      1164 S    -ash
16416 root         0 IW   [kworker/u4:7-mp]
17384 root       928 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
19872 root      1168 S    -ash
20235 root      2196 S    /usr/bin/ss-local -c /var/etc/shadowsocks-libev/ss_local.tracker.json
20236 root      2016 S    /usr/bin/ss-redir -c /var/etc/shadowsocks-libev/ss_redir.hi.json
20237 root     12536 S    /usr/bin/ss-redir -c /var/etc/shadowsocks-libev/ss_redir.hi2.json
20323 root         0 IW   [kworker/0:1-eve]
20722 root      1196 S    /usr/sbin/glorytun keyfile /tmp/glorytun-vpn.key port 65001 host xxx.xxx.xxx.xxx dev tun0 mptcp chacha20 re
20793 root         0 IW   [kworker/1:0-eve]
20907 root      1224 S    /bin/sh /sbin/hotplug-call iface
21198 root      1572 S    /bin/sh /etc/rc.common /etc/init.d/mptcp reload
21433 root      1572 S    /bin/sh /etc/rc.common /etc/init.d/mptcp reload
21434 root      1156 S    traceroute -m1 -i tun0 8.8.8.8
21435 root      1168 S    awkFNR ==2{ print $2 }
21546 root      1152 S    sleep 4
22081 root         0 IW   [kworker/u4:4-ev]
22517 root         0 IW   [kworker/u4:1-mp]
22531 root      1152 S    sleep 2
22870 root      1152 S    sleep 5
23314 root      1224 S    /bin/sh /sbin/hotplug-call iface
23315 root      1488 S    /bin/sh /etc/rc.common /etc/init.d/mptcp enabled
23329 root      1152 S    flock 1000
23451 root      1152 S<   /usr/sbin/ntpd -n -N -l -S /usr/sbin/ntpd-hotplug -p 0.openwrt.pool.ntp.org -p 1.openwrt.pool.ntp.org -p
23495 root      1256 S    /bin/sh /etc/rc.common /etc/rc.d/S99adb-enablemodem boot
23498 root      3812 S    adb wait-for-device
23500 root      4108 S    adb -P 5037 fork-server server
23565 root      1152 S    sleep 2
23576 root       688 S    ping -B -I eth1.200 -w 5 -c 2 -Q 184 192.168.200.1
[... 203 lines cut ...]

By the way -- 
when glorytun starts up - it hooks up a network 192.168.1.0 on the vps with endpoint 10.255.255.2

>
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         gw2             0.0.0.0         UG    0      0        0 ens192
10.255.248.0    0.0.0.0         255.255.255.0   U     0      0        0 omr-bonding
10.255.248.2    0.0.0.0         255.255.255.255 UH    0      0        0 omr-bonding
10.255.250.0    10.255.250.2    255.255.255.0   UG    0      0        0 tun1
10.255.250.2    0.0.0.0         255.255.255.255 UH    0      0        0 tun1
10.255.251.2    0.0.0.0         255.255.255.255 UH    0      0        0 dsvpn0
10.255.252.0    10.255.252.2    255.255.255.0   UG    0      0        0 tun0
10.255.252.2    0.0.0.0         255.255.255.255 UH    0      0        0 tun0
10.255.253.0    0.0.0.0         255.255.255.0   U     0      0        0 mlvpn0
10.255.254.0    0.0.0.0         255.255.255.252 U     0      0        0 gt-udp-tun0
10.255.255.0    0.0.0.0         255.255.255.252 U     0      0        0 gt-tun0
gw2             0.0.0.0         255.255.255.255 UH    0      0        0 ens192
192.168.1.0     10.255.255.2    255.255.255.0   UG    0      0        0 gt-tun0

## Specifications

  - OpenMPTCProuter version: v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: Ubuntu 20.04.2 LTS  / V1024
  - OpenMPTCProuter platform: x86_64 vmware

Ysurac, 2021-01-31:
Check that MPTCP is disabled on OMRVPN interface on the router.
Check also router log in State->System log.

Ysurac, 2021-01-31:
Also if you have choice on VPS, use Debian. Ubuntu is far less tested.

homicida2000, 2021-01-31:
MPTCP is disabled on OMRVPN interface on the router.

Syslog output snippet :
>
Sun Jan 31 20:19:41 2021 user.notice post-tracking-post-tracking: Set ip route replace default scope global  nexthop via 192.168.200.1 dev eth1.200 weight 2 nexthop via 192.168.201.1 dev eth2.200 weight 1 nexthop via 192.168.202.1 dev eth3.200 weight 1
Sun Jan 31 20:19:42 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Jan 31 20:19:42 2021 user.notice omr-bypass: Reload dnsmasq...
Sun Jan 31 20:19:42 2021 daemon.info dnsmasq[22066]: read /etc/hosts - 4 addresses
Sun Jan 31 20:19:42 2021 daemon.info dnsmasq[22066]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Sun Jan 31 20:19:42 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Jan 31 20:19:42 2021 daemon.debug dnsmasq[22066]: listening on tun0(#89): 10.255.255.2 port 53
Sun Jan 31 20:19:46 2021 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sun Jan 31 20:19:46 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Sun Jan 31 20:19:47 2021 user.notice OMR-VPS: Restart glorytun...
Sun Jan 31 20:19:47 2021 daemon.info glorytun[13674]: STOPPED tun0
Sun Jan 31 20:19:47 2021 daemon.notice netifd: Network device 'tun0' link is down
Sun Jan 31 20:19:47 2021 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Sun Jan 31 20:19:47 2021 daemon.notice netifd: Interface 'omrvpn' is now down
Sun Jan 31 20:19:47 2021 daemon.debug dnsmasq[22066]: stopped listening on tun0(#89): 10.255.255.2 port 53
Sun Jan 31 20:19:47 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Jan 31 20:19:47 2021 daemon.info glorytun[17719]: INITIALIZED tun0
Sun Jan 31 20:19:47 2021 daemon.notice netifd: Interface 'omrvpn' is disabled
Sun Jan 31 20:19:47 2021 daemon.notice netifd: Interface 'omrvpn' is enabled
Sun Jan 31 20:19:47 2021 daemon.notice netifd: Interface 'omrvpn' is disabled
Sun Jan 31 20:19:47 2021 daemon.notice netifd: Interface 'omrvpn' is enabled
Sun Jan 31 20:19:47 2021 daemon.notice netifd: Network device 'tun0' link is up
Sun Jan 31 20:19:47 2021 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Sun Jan 31 20:19:47 2021 daemon.notice netifd: Interface 'omrvpn' is setting up now
Sun Jan 31 20:19:47 2021 daemon.notice netifd: Interface 'omrvpn' is now up
Sun Jan 31 20:19:47 2021 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Sun Jan 31 20:19:47 2021 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Sun Jan 31 20:19:47 2021 user.notice ModemManager: hotplug: add network interface tun0: event processed
Sun Jan 31 20:19:47 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Sun Jan 31 20:19:47 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Jan 31 20:19:47 2021 kern.err kernel: [  510.927020] __mptcp_init4_subsockets: token 0xdba558c bind() to 10.255.255.2 index 89 failed, error -99
Sun Jan 31 20:19:47 2021 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Sun Jan 31 20:19:47 2021 daemon.err glorytun[17719]: 82.165.48.124.65001: connected
Sun Jan 31 20:19:47 2021 daemon.info glorytun[17719]: STARTED tun0
Sun Jan 31 20:19:49 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Jan 31 20:19:49 2021 user.notice omr-bypass: Reload dnsmasq...
Sun Jan 31 20:19:49 2021 daemon.info dnsmasq[22066]: read /etc/hosts - 4 addresses
Sun Jan 31 20:19:49 2021 daemon.info dnsmasq[22066]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Sun Jan 31 20:19:49 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Jan 31 20:19:49 2021 daemon.debug dnsmasq[22066]: listening on tun0(#90): 10.255.255.2 port 53
Sun Jan 31 20:19:50 2021 user.notice post-tracking-post-tracking: Set ip route replace default scope global  nexthop via 192.168.200.1 dev eth1.200 weight 2 nexthop via 192.168.201.1 dev eth2.200 weight 1 nexthop via 192.168.202.1 dev eth3.200 weight 1
Sun Jan 31 20:19:53 2021 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sun Jan 31 20:19:53 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Sun Jan 31 20:19:54 2021 user.notice OMR-VPS: Restart glorytun...
Sun Jan 31 20:19:54 2021 daemon.info glorytun[17719]: STOPPED tun0
Sun Jan 31 20:19:54 2021 daemon.notice netifd: Network device 'tun0' link is down
Sun Jan 31 20:19:54 2021 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Sun Jan 31 20:19:54 2021 daemon.notice netifd: Interface 'omrvpn' is now down
Sun Jan 31 20:19:54 2021 daemon.debug dnsmasq[22066]: stopped listening on tun0(#90): 10.255.255.2 port 53
Sun Jan 31 20:19:54 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Jan 31 20:19:54 2021 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Sun Jan 31 20:19:54 2021 daemon.info glorytun[22194]: INITIALIZED tun0
Sun Jan 31 20:19:54 2021 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Sun Jan 31 20:19:54 2021 daemon.notice netifd: Interface 'omrvpn' is disabled
Sun Jan 31 20:19:54 2021 daemon.notice netifd: Interface 'omrvpn' is enabled
Sun Jan 31 20:19:54 2021 daemon.notice netifd: Interface 'omrvpn' is disabled
Sun Jan 31 20:19:54 2021 daemon.notice netifd: Interface 'omrvpn' is enabled
Sun Jan 31 20:19:54 2021 daemon.notice netifd: Network device 'tun0' link is up
[... cut ...]
Sun Jan 31 20:20:11 2021 daemon.info dnsmasq[22066]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Sun Jan 31 20:20:11 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Jan 31 20:20:11 2021 daemon.debug dnsmasq[22066]: listening on tun0(#93): 10.255.255.2 port 53
Sun Jan 31 20:20:11 2021 user.notice post-tracking-post-tracking: Set ip route replace default scope global  nexthop via 192.168.200.1 dev eth1.200 weight 2 nexthop via 192.168.201.1 dev eth2.200 weight 1 nexthop via 192.168.202.1 dev eth3.200 weight 1

I'll test debian 9 or 10 ??? and kernel ?

Ysurac, 2021-01-31:
Debian 10. What you mean by kernel ? VPS script replace default kernel.
If there is no changes after using Debian, what is the result of ```ip r``` on the router via SSH ? and the result of ```uci show network``` and ```uci show glorytun``` ?

homicida2000, 2021-01-31:
same ...

ip r vps :

root@localhost:~# ip r
default via 10.255.255.1 dev ens192
10.255.248.0/24 dev omr-bonding scope link
10.255.248.2 dev omr-bonding scope link src 10.255.248.1
10.255.250.0/24 via 10.255.250.2 dev tun1
10.255.250.2 dev tun1 proto kernel scope link src 10.255.250.1
10.255.251.2 dev dsvpn0 proto kernel scope link src 10.255.251.1
10.255.252.0/24 via 10.255.252.2 dev tun0
10.255.252.2 dev tun0 proto kernel scope link src 10.255.252.1
10.255.253.0/24 dev mlvpn0 proto kernel scope link src 10.255.253.1
10.255.254.0/30 dev gt-udp-tun0 proto kernel scope link src 10.255.254.1
10.255.255.0/30 dev gt-tun0 proto kernel scope link src 10.255.255.1
**10.255.255.1 dev ens192 scope link < THIS could be a problem, it's the default gw of the vps, and glorytun has the same ip na is not be changeable from the webif**
192.168.1.0/24 via 10.255.255.2 dev gt-tun0
-----------------------------------------------------------------
Router : 

root@OpenMPTCProuter:~# ip r
default
        nexthop via 192.168.200.1 dev eth1.200 weight 2
        nexthop via 192.168.201.1 dev eth2.200 weight 1
        nexthop via 192.168.202.1 dev eth3.200 weight 1
default via 192.168.200.1 dev eth1.200 metric 3
default via 192.168.201.1 dev eth2.200 metric 4
default via 192.168.202.1 dev eth3.200 metric 9
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2
82.165.48.124
        nexthop via 192.168.200.1 dev eth1.200 weight 10
        nexthop via 192.168.201.1 dev eth2.200 weight 1
        nexthop via 192.168.202.1 dev eth3.200 weight 1
127.0.0.0/8 dev lo proto static scope link metric 6
172.16.0.0/16 via 192.168.255.2 dev eth0
192.168.200.0/24 dev eth1.200 scope link metric 3
192.168.201.0/24 dev eth2.200 scope link metric 4
192.168.202.0/24 dev eth3.200 scope link metric 9
------------------------------------------------------------

root@OpenMPTCProuter:~# uci show network
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.macaddr='00:00:00:00:00:00'
network.loopback.metric='6'
network.globals=globals
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='blest'
network.globals.congestion='cubic'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.ula_prefix='fd11:badf:1949::/48'
network.globals.mptcp_syn_retries='2'
network.lan=interface
network.lan.ifname='eth0'
network.lan.proto='static'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.ipv6='0'
network.lan.delegate='0'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.macaddr='00:0c:29:3e:98:11'
network.lan.metric='7'
network.lan.addlatency='0'
network.lan.ipaddr='192.168.255.1'
network.lan.defaultroute='0'
network.lan.peerdns='0'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.multipath='master'
network.wan1.defaultroute='0'
network.wan1.macaddr='00:0c:29:3e:98:f3'
network.wan1.ipv6='0'
network.wan1.metric='3'
network.wan1.peerdns='0'
network.wan1.ifname='eth1.200'
network.wan1.addlatency='0'
network.wan1.ipaddr='192.168.200.4'
network.wan1.netmask='255.255.255.0'
network.wan1.gateway='192.168.200.1'
network.wan1.dns='192.168.200.1'
network.wan2=interface
network.wan2.proto='static'
network.wan2.ip4table='wan'
network.wan2.multipath='on'
network.wan2.defaultroute='0'
network.wan2.macaddr='00:0c:29:3e:98:fd'
network.wan2.ipv6='0'
network.wan2.metric='4'
network.wan2.peerdns='0'
network.wan2.ifname='eth2.200'
network.wan2.addlatency='0'
network.wan2.ipaddr='192.168.201.4'
network.wan2.netmask='255.255.255.0'
network.wan2.gateway='192.168.201.1'
network.wan2.dns='192.168.201.1'
network.omrvpn=interface
network.omrvpn.ifname='tun0'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omrvpn.type='tunnel'
network.omrvpn.txqueuelen='100'
network.omrvpn.metric='1200'
network.omrvpn.ipv6='0'
network.omrvpn.proto='none'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.ipaddr='10.255.255.2'
network.omr6in4.peeraddr='10.255.255.1'
network.omr6in4.auto='0'
network.omr6in4.metric='1201'
network.omr6in4.ip6addr='fe80::a00:2/126'
network.omr6in4.gateway='fe80::a00:1/126'
network.wan3=interface
network.wan3.proto='static'
network.wan3.ifname='eth3.200'
network.wan3.addlatency='0'
network.wan3.ipaddr='192.168.202.4'
network.wan3.netmask='255.255.255.0'
network.wan3.gateway='192.168.202.1'
network.wan3.dns='192.168.202.1'
network.wan3.multipath='on'
network.wan3.defaultroute='0'
network.wan3.peerdns='0'
network.wan3.metric='9'
network.wan3.ipv6='0'
network.wan3.macaddr='00:0c:29:3e:98:07'

Ysurac, 2021-02-01:
Yes, it's a big problem and this can't work with this VPS gateway. You can try another VPN in System->OpenMPTCProuter, "Advanced settings" checkbox but I don't think this will work well.

homicida2000, 2021-02-02:
OK
everything works now.
I've changend the glorytun-vpn network to 172.30.255.1(server) and 172.30.255.2(router).

btw. the vps ist hosted at ionos(1&1) germany

closed 2021-02-02 by homicida2000