Ysurac/openmptcprouter #1557: Routing update on shadowsocks-libev

nitrique, 2021-01-31:
Hey @Ysurac, 
good to see a project originating from a "Bressan" 😊 ( I live in Vonnas ) !

When following this tutorial : [Multi IP VPS](https://github.com/Ysurac/openmptcprouter/wiki/VPS-Multi-IPs)
I need to reboot the router to apply changes on redirects.

## Expected Behavior

Apply immediately changes when clicking on "save and apply"

## Current Behavior

Changes are saved but not applied immediately on shadowsocks-libev. 

## Possible Solution

Changes are applied when doing `service shadowsocks-libev restart` on SSH shell,

## Steps to Reproduce the Problem

  1. Set MPTCP on  Debian 10 with multiple IPs
  2. Wait for IP failover discover
  3. Follow tutorial

## Context (Environment)

OMR Running on Promox Cluster, 
OMR VPS running in Online Datacenter, with 5 IP.

## Specifications

  - OpenMPTCProuter version: v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: 0.1025-test
  - OpenMPTCProuter platform: x86_64 (qEmu)
  

Thanks,
Nicolas

Ysurac, 2021-02-01:
It's a know issue, this will be solved in next VPS script release.

nitrique, 2021-02-01:
> It's a know issue, this will be solved in next VPS script release.

So maybe you should add it to the wiki page

closed 2021-05-08 by github-actions[bot]