Ysurac/openmptcprouter #1559: 2 vdsl + 1 4g problem

sonnyxp2008, 2021-02-01:
Goodmorning everyone. i have a problem with the 4g connection i entered on omr. Basically as soon as I insert the lan cable on the port, the connection drops and becomes slow and unstable. I am using a Mikrotik antenna. The assistance tells me that there are no problems, in fact if I insert it directly on the PC, it surfs easily, as soon as I insert it on omr it dies. I send a photo. Thank you
[image]

Ysurac, 2021-02-01:
Please use the template.

sonnyxp2008, 2021-02-01:
Ysurac did you mean this model? Excuse me but I'm new around here

Ysurac, 2021-02-01:
Yes.
You can check on State->System log if there is any info.
How is connected the 4g, to a own port or using macvlan or vlan ?

sonnyxp2008, 2021-02-01:
these are the mistakes. I am currently using it with an upstream router. I have also tried everything to change the server where to send the pings.
Sun Jan 31 19:46:56 2021 user.notice post-tracking-post-tracking: 4g (eth2) switched off because ping from 192.168.0.102 error (1.1.1.1,8.8.4.4,1.1.1.1)
Sun Jan 31 19:46:56 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.0.1 dev eth2
Sun Jan 31 19:47:46 2021 daemon.err /usr/bin/ss-redir[13605]: server recv: Operation timed out
Sun Jan 31 19:47:56 2021 daemon.err /usr/bin/ss-redir[13604]: server recv: Operation timed out
Sun Jan 31 19:48:42 2021 daemon.err /usr/bin/ss-redir[13607]: send: Operation timed out
Sun Jan 31 19:48:43 2021 daemon.err /usr/bin/ss-redir[13604]: server recv: Host is unreachable
Sun Jan 31 19:50:50 2021 daemon.err /usr/bin/ss-redir[13604]: remote recv: Connection reset by peer
Sun Jan 31 19:51:36 2021 daemon.err /usr/bin/ss-redir[13604]: server recv: Operation timed out
Sun Jan 31 19:52:35 2021 daemon.err /usr/bin/ss-redir[13606]: server recv: Operation timed out
Sun Jan 31 19:53:06 2021 daemon.err /usr/bin/ss-redir[13604]: server recv: Host is unreachable
Sun Jan 31 19:54:47 2021 daemon.err /usr/bin/ss-redir[13605]: server recv: Host is unreachable
Sun Jan 31 19:54:58 2021 daemon.err /usr/bin/ss-redir[13606]: server recv: Host is unreachable
Sun Jan 31 19:55:00 2021 daemon.err /usr/bin/ss-redir[13606]: send: Connection reset by peer
Sun Jan 31 19:55:00 2021 daemon.err /usr/bin/ss-redir[13608]: send: Connection reset by peer

Ysurac, 2021-02-01:
Can you do a ```ping -I 192.168.0.102 192.168.0.1``` via SSH or Services->Terminal on the router ?
If not, check 4G router IP.

sonnyxp2008, 2021-02-01:
the ping is there, in truth as soon as I restart the antenna the connection appears up, after a few minutes it goes down. I am using a Mikrotik 4g antenna. Another fact is that I put a router upstream to see if I can surf safely. The fact is that as soon as I insert the cable to the assigned port, the connection is unstable. Only you tube opens and every now and then some speed tests start. If I unplug the cable from OMR everything works normally. I asked Mikrotik for assistance and they replied that there is no conflict. But I believe that there is something wrong or that does not support something.

Ysurac, 2021-02-01:
Do you have something with ```ping -I 192.168.0.102 8.8.8.8``` ?
What are the references of the material ?

sonnyxp2008, 2021-02-01:
then OMR ping side on 0.1 is ok 8.8.8.8 after 2-3 minutes it goes ko.
On a pc directly on a router that manages pc and OMR we have unstable connection if I connect the cable to OMR, unstable connection to the PC but from the pc I get the ping on 8.8.8.8 It is difficult for me to explain it to you because such a thing has never happened to me. The 0.1 router I put it just to see if the connection went ko. i could delete it but i have the same problem. Could it be an MPTCP problem between OMR and Mikrotik? believe me now I've gone crazy

Ysurac, 2021-02-01:
If there is a firewall on miktotik or something that block MPTCP yes. There is an option that block MPTCP on some Cisco.

sonnyxp2008, 2021-02-01:
I have disabled the firewall I have modified the nat but the results are bad. can't disable MPTCP on a single wan? Or any other solution is welcome

Ysurac, 2021-02-01:
In Network->MPTCP you can disable MPTCP on a wan. You can try and then test ping 8.8.8.8, if same issue than it's not related to MPTCP.

sonnyxp2008, 2021-02-01:
i think i don't understand, if i disable MPTCP on a single wan it obviously falls, i send you a picture of my current configuration
[image]

Ysurac, 2021-02-01:
You can try to disable Multipath TCP on 4G interface. Then you will see if the interface can be used directly via the ping -I previous command or if the result is the same.

sonnyxp2008, 2021-02-01:
So if I disable MPTCP on the wan we don't have any ping. Maybe I found the solution after a week and the problem lies in the NAT. As soon as I'm done freaking out I will write my configuration here so that someone can be of help as these MIkrotik antennas have become more commercial

Ysurac, 2021-02-01:
You can also paste here the configuration (without keys/vps ip), available in System->OpenMPTCProuter, "show all settings" tab, so I can check if there is no error somewhere.

sonnyxp2008, 2021-02-01:
nat refers to the mikrotik routerboard. Now due to an error I can no longer enter the Mikrotik interface. I should reset everything but it's cold now and I don't want to go up and reset the antenna. Tomorrow I will reset it and configure the NAT. I keep you updated

closed 2021-05-14 by github-actions[bot]