Ysurac/openmptcprouter #1560: Create image for unsupported platform

extart, 2021-02-01:
In wiki:
If it's a not yet supported platform, select your arch in the menu and build it like this:

cd myplatform/source
make menuconfig 
make -j6

But I did not understand exactly how I should enter into (cd myplatform/source), because at the root there is no such directory ...
I want to try to build for mikrotik RBM33G 
how to do it?

Ysurac, 2021-02-01:
It's created by previous command.

extart, 2021-02-04:
> It's created by previous command.

Please tell me which previous command?
I have read all the text from here several times
https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform
I didn't understand ...

Ysurac, 2021-02-04:
By ```OMR_TARGET="myplatform" OMR_FEED_SRC="master" ./build.sh```

extart, 2021-02-07:
> By `OMR_TARGET="myplatform" OMR_FEED_SRC="master" ./build.sh`

I'm starting to understand ... What parameter to use in OMR_TARGET if the platform is unsupported? For example, for mikrotik RBM33G. In OpenWRT, target is "ramips", subtarget is "mt7621", Package architecture: "mipsel_24kc". What value to enter instead of "myplatform"?

Ysurac, 2021-02-07:
You can set what you want. It's only used to create the directory for unsupported platform.

extart, 2021-02-07:
Thanks! I'm going to try now.

spacedream, 2021-02-12:
@extart did you tried yet? i don't wanna created another topic with same subject, so will ask here (hope you don't mind)
dear @Ysurac, i trying to compile for mips_24kc but its braked at ndpi-netfilter2 package. its any chance you can help with debug and fix that?
i'm using this config 
[config-gl_mifi.zip](https://github.com/Ysurac/openmptcprouter/files/5972824/config-gl_mifi.zip)
thank you for doing this awesome project

Ysurac, 2021-02-12:
Do you use develop branches ? I compile it for an mt7621 based arch using develop branches (both build scripts and OMR_FEED_SRC="develop").

spacedream, 2021-02-12:
i'll try! ty! <3

spacedream, 2021-02-12:
git clone https://github.com/Ysurac/openmptcprouter.git
wget https://github.com/Ysurac/openmptcprouter/files/5972824/config-gl_mifi.zip
unzip config-gl_mifi.zip && cp config-gl_mifi openmptcprouter/
cd openmptcprouter
git checkout master
OMR_TARGET="gl_mifi" OMR_FEED_SRC="develop" ./build.sh -j1 V=s

extart, 2021-02-13:
Still trying to build. 
There is hope that it will work out. Since there is a compiled package here - https://download.openmptcprouter.com/develop/5.4/ubnt-erx/targets/ramips/mt7621/
ramips, mt7621 - like RBM33G
Now I will start the build from "develop"...

spacedream, 2021-02-13:
after 
OMR_TARGET="gl_mifi" OMR_FEED_SRC="develop" ./build.sh -j1 V=s

compilation process disrupted by Kernel configuration few questions (which i answered all by hit "enter" key, including this one:

> * Platform support for MikroTik RouterBoard virtual devices
> *
> Platform support for MikroTik RouterBoard virtual devices (MIKROTIK) [N/y/?] (NEW) 

so this part explained your problem @extart. so do command exactly like i showed above, with "-j1 V=s" part 
but change OMR_TARGET= on your own. you need to take your config (maybe named as config.microtik_mt7621) from your device and put to openmptcprouter folder.

@Ysurac, can you add "--debug_log" option (or so) to your build.sh script, please? 
so run with this option it will grab all logs and info you maybe need, to easier to learn and understand what problem is and how to solve it, and put them in zip file. if its not hard to do. i guess its can be useful in some cases in future too.

Ysurac, 2021-02-13:
I don't need log as I don't support unsupported platforms :)

spacedream, 2021-02-13:
just a tip: i did make my config.gl_mifi for my router (so you can make in same way for any another models, i guess) by make this commands first (from instruction for unsupported platform):

cd **gl_mifi**/source
make menuconfig 
cp **.config** config.gl_mifi
cp config.gl_mifi ../../ && cd ../../

folder "**gl_mifi**/source" created after first time try above **build.sh** command (even with no luck, but its will download source code first, its what we need).

after this its will be possible to compile by command from above:

**OMR_TARGET="gl_mifi" OMR_FEED_SRC="develop" ./build.sh -j1 V=s**

[image]

spacedream, 2021-02-13:
> I don't need log as I don't support unsupported platforms :)

uh... i did missed that comment. 

is no hope? 

[image]

spacedream, 2021-02-13:
so my compilation is showing the same error with bunch of messages like this:
````
  CC [M]  /home/sd/wrk/openmptcprouter-dev/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-25a5c2e1d619aa2d819b18bed8276b3bd7eb4c5b/ndpi-netfilter/src/../../src/lib/protocols/fasttrack.o
In file included from /home/sd/wrk/openmptcprouter-dev/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-25a5c2e1d619aa2d819b18bed8276b3bd7eb4c5b/ndpi-netfilter/src/../../src/include/ndpi_main.h:30,
                 from /home/sd/wrk/openmptcprouter-dev/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-25a5c2e1d619aa2d819b18bed8276b3bd7eb4c5b/ndpi-netfilter/src/../../src/include/ndpi_api.h:28,
                 from /home/sd/wrk/openmptcprouter-dev/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-25a5c2e1d619aa2d819b18bed8276b3bd7eb4c5b/ndpi-netfilter/src/../../src/lib/protocols/fasttrack.c:29:
/home/sd/wrk/openmptcprouter-dev/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-25a5c2e1d619aa2d819b18bed8276b3bd7eb4c5b/ndpi-netfilter/src/../../src/include/ndpi_define.h:66:21: warning: "__LITTLE_ENDIAN" is not defined, evaluates to 0 [-Wundef]
 #if __BYTE_ORDER == __LITTLE_ENDIAN
````
console log here: [gl_mifi_log.txt.zip](https://github.com/Ysurac/openmptcprouter/files/5975879/gl_mifi_log.txt.zip)

spacedream, 2021-02-13:
somewhere missing line, maybe?
````
#include <endian.h>
````
is what the error complaining?

spacedream, 2021-02-13:
someone pointed me to this missing file: 
**skbuff.h: No such file or directory**
why this file is missing?

````

make[4]: Entering directory '/home/sd/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-3.2-0a03249da911f4033fd2a0d38a97cdba72eee2b6/ndpi-netfilter/src'
grep: /home/sd/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/source/include/linux/skbuff.h: No such file or directory
$xt_ndpi-y is [main.o ndpi_strcol.o ndpi_proc_parsers.o ndpi_proc_generic.o ndpi_proc_info.o  ndpi_proc_flow.o ndpi_proc_hostdef.o ndpi_proc_ipdef.o ../libre/regexp.o ../../src/lib/third_party/src/ndpi_md5.o ../../src/lib/third_party/src/ndpi_sha1.o ../../src/lib/third_party/src/libahocorasick.o ../../src/lib/third_party/src/libcache.o ../../src/lib/third_party/src/ndpi_patricia.o ../../src/lib/third_party/src/ht_hash.o ../../src/lib/third_party/src/btlib.o ../../src/lib/ndpi_main.o ../../src/lib/ndpi_utils.o ../../src/lib/ndpi_serializer.o ../../src/lib/protocols/*.o] -I/../../src/include -I/../../src/lib -I/../libre -I/../../src/lib/third_party/include  -DHAVE_CONFIG_H -DNDPI_LIB_COMPILATION -DOPENDPI_NETFILTER_MODULE -g
````

spacedream, 2021-02-13:
i checked - its even folder '/home/sd/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/source' **not exists**, but folder '/home/sd/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/' is there and full of files.

spacedream, 2021-02-13:
i did created folder source and symlinked
source/include to ../include
so i will try again to compile now.

spacedream, 2021-02-14:
it didn't help. same error happened: 
[compilation.log.txt](https://github.com/Ysurac/openmptcprouter/files/5977386/compilation.log.txt)

spacedream, 2021-02-14:
arm is a BIG_ENDIAN but looks like its not recognized somewhere in headers maybe.

spacedream, 2021-02-17:
so i guess its was Gentoo problem. i trying right now in virtualbox + ubuntu-20. i'll let you know later of results.
ps: strangely enough but i assembled many packages before on same system and even many images for same router, but its breaks specifically at ndpi package.

Ysurac, 2021-02-17:
It's not Gentoo as it's the distribution I'm using ;)

spacedream, 2021-02-17:
well, can you try then assemble for ar71xx, [exactly as i did](https://github.com/Ysurac/openmptcprouter/issues/1560#issuecomment-778299375)? you will see the error i mentioned, i'm sure. 
ps: or did you installed some packages additionally for compile? like crossdev, cross-arm-softfloat-linux-mips_24kc_musl/linux-headers or cross-arm-softfloat-linux-uclibceabi/linux-headers? in some reason first two didn't installed for me, i tried yestarday. its why i gave up and trying to build in virtualbox now.

spacedream, 2021-02-17:
Ok, i been not lucky again :( 
now compilation is breaking (in virtualbox now with xubuntu-20) on package protobuf

> mptcprouter/feeds/openmptcprouter/glorytun clean-build
 make[3] -C /home/user/wrk/openmptcprouter/feeds/openmptcprouter/glorytun compile
 make[3] -C /home/user/wrk/openmptcprouter/feeds/openmptcprouter/glorytun-udp clean-build
 make[3] -C /home/user/wrk/openmptcprouter/feeds/openmptcprouter/glorytun-udp compile
 make[3] -C /home/user/wrk/openmptcprouter/feeds/openmptcprouter/golang/golang host-compile
 make[3] -C /home/user/wrk/openmptcprouter/feeds/openmptcprouter/golang/golang-golang-x-text clean-build
 make[3] -C /home/user/wrk/openmptcprouter/feeds/openmptcprouter/golang/golang-golang-x-text compile
 make[3] -C /home/user/wrk/openmptcprouter/feeds/openmptcprouter/golang/golang-golang-x-net clean-build
 make[3] -C /home/user/wrk/openmptcprouter/feeds/openmptcprouter/golang/golang-golang-x-net compile
 make[3] -C /home/user/wrk/openmptcprouter/feeds/openmptcprouter/golang/golang-golang-x-sys clean-build
 make[3] -C /home/user/wrk/openmptcprouter/feeds/openmptcprouter/golang/golang-golang-x-sys compile
 make[3] -C /home/user/wrk/openmptcprouter/feeds/openmptcprouter/golang/golang-golang-x-crypto clean-build
 make[3] -C /home/user/wrk/openmptcprouter/feeds/openmptcprouter/golang/golang-golang-x-crypto compile
 make[3] -C /home/user/wrk/openmptcprouter/feeds/openmptcprouter/protobuf host-compile
    ERROR: package/feeds/openmptcprouter/protobuf [host] failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/user/wrk/openmptcprouter/gl_mifi/source/include/toplevel.mk:242: world] Error 1
user@user:~/wrk/openmptcprouter$ 

running now with -j1 V=s to see more details but not expecting happy ending in general. don't know why i'm so not lucky :\

spacedream, 2021-02-17:
its breaks at Kernel config. this is normal? i don't think so... how patch this behavior for future skip this break?
also i noticed DEFAULT_OLIA changes previous settings in new 5.4.83 kernel, did you noticed it?
how about TCP_CONG_BBR2? we need it? and few more new options... I don't know for sure how to correctly answer all follows questions its asks. so far i got this:

> .target /dev/null /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/.config.override > /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/.config.set
mv /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/.config.set /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/.config.old
grep -v INITRAMFS /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/.config.old > /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/.config.set
rm -rf /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/modules
cmp -s /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/.config.set /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/.config.prev || { cp /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/.config.set /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/.config; cp /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/.config.set /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/.config.prev; }
export MAKEFLAGS= ; [ -d /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/user_headers ] || make -C /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83 KCFLAGS="-fmacro-prefix-map=/home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl=target-mips_24kc_musl" HOSTCFLAGS="-O2 -I/home/user/wrk/openmptcprouter/gl_mifi/source/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mips-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Dec 15 19:35:00 2020" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/user/wrk/openmptcprouter/gl_mifi/source/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/user/wrk/openmptcprouter/gl_mifi/source/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.83 CC="mips-openwrt-linux-musl-gcc" INSTALL_HDR_PATH=/home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83/user_headers headers_install
make[5]: Entering directory '/home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.83'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/unifdef
  WRAP    arch/mips/include/generated/uapi/asm/bpf_perf_event.h
  WRAP    arch/mips/include/generated/uapi/asm/ipcbuf.h
  SYSTBL  arch/mips/include/generated/asm/syscall_table_32_o32.h
  SYSTBL  arch/mips/include/generated/asm/syscall_table_64_n32.h
  SYSTBL  arch/mips/include/generated/asm/syscall_table_64_n64.h
  SYSTBL  arch/mips/include/generated/asm/syscall_table_64_o32.h
  SYSHDR  arch/mips/include/generated/uapi/asm/unistd_n32.h
  SYSHDR  arch/mips/include/generated/uapi/asm/unistd_n64.h
  SYSHDR  arch/mips/include/generated/uapi/asm/unistd_o32.h
  SYSNR   arch/mips/include/generated/uapi/asm/unistd_nr_n32.h
  SYSNR   arch/mips/include/generated/uapi/asm/unistd_nr_n64.h
  SYSNR   arch/mips/include/generated/uapi/asm/unistd_nr_o32.h
  HOSTCC  arch/mips/tools/elf-entry
  HOSTCC  arch/mips/boot/tools/relocs_32.o
  HOSTCC  arch/mips/boot/tools/relocs_64.o
  HOSTCC  arch/mips/boot/tools/relocs_main.o
  HOSTLD  arch/mips/boot/tools/relocs
  UPD     include/generated/uapi/linux/version.h
  HDRINST usr/include/video/edid.h
  HDRINST usr/include/video/uvesafb.h
  HDRINST usr/include/video/sisfb.h
  HDRINST usr/include/drm/virtgpu_drm.h
  HDRINST usr/include/drm/lima_drm.h
  HDRINST usr/include/drm/tegra_drm.h
  HDRINST usr/include/drm/mga_drm.h
  HDRINST usr/include/drm/drm.h
  HDRINST usr/include/drm/vc4_drm.h
  HDRINST usr/include/drm/r128_drm.h
  HDRINST usr/include/drm/vmwgfx_drm.h
  HDRINST usr/include/drm/i810_drm.h
  HDRINST usr/include/drm/sis_drm.h
  HDRINST usr/include/drm/drm_sarea.h
  HDRINST usr/include/drm/etnaviv_drm.h
  HDRINST usr/include/drm/vgem_drm.h
  HDRINST usr/include/drm/v3d_drm.h
  HDRINST usr/include/drm/drm_fourcc.h
  HDRINST usr/include/drm/via_drm.h
  HDRINST usr/include/drm/amdgpu_drm.h
  HDRINST usr/include/drm/omap_drm.h
[... cut ...]
  TCP Vegas (TCP_CONG_VEGAS) [N/m/y/?] n
  TCP NV (TCP_CONG_NV) [N/m/y/?] n
  Scalable TCP (TCP_CONG_SCALABLE) [N/m/y/?] n
  TCP Low Priority (TCP_CONG_LP) [N/m/y/?] n
  TCP Veno (TCP_CONG_VENO) [N/m/y/?] n
  YeAH TCP (TCP_CONG_YEAH) [N/m/y/?] n
  TCP Illinois (TCP_CONG_ILLINOIS) [N/m/y/?] n
  DataCenter TCP (DCTCP) (TCP_CONG_DCTCP) [N/m/y/?] n
  CAIA Delay-Gradient (CDG) (TCP_CONG_CDG) [N/m/y/?] n
  BBR TCP (TCP_CONG_BBR) [Y/n/m/?] y
  NANGINLANG TCP (TCP_CONG_NANQINLANG) [M/n/y/?] m
  BBR2 TCP (TCP_CONG_BBR2) [N/m/y/?] (NEW)

spacedream, 2021-02-17:
i just hit "Enter" all questions, now process continues...

spacedream, 2021-02-18:
Ok, now i got this:

> touch -r /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/golang-golang-x-net-20180620-afe8f62b/.built /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/golang-golang-x-net-20180620-afe8f62b/.autoremove 2>/dev/null >/dev/null
find /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/golang-golang-x-net-20180620-afe8f62b -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -and -not -name '.pkgdir' | xargs -r rm -rf
make[3]: Leaving directory '/home/user/wrk/openmptcprouter/feeds/openmptcprouter/golang/golang-golang-x-net'
time: package/feeds/openmptcprouter/golang-golang-x-net/compile#11.71#7.45#22.60
make[3]: Entering directory '/home/user/wrk/openmptcprouter/feeds/openmptcprouter/golang/golang-golang-x-sys'
mkdir -p /home/user/wrk/openmptcprouter/gl_mifi/source/dl
SHELL= flock /home/user/wrk/openmptcprouter/gl_mifi/source/tmp/.golang-golang-x-sys-20180621-63fc586f.tar.xz.flock -c '  	/home/user/wrk/openmptcprouter/gl_mifi/source/scripts/download.pl "/home/user/wrk/openmptcprouter/gl_mifi/source/dl" "golang-golang-x-sys-20180621-63fc586f.tar.xz" "3afe7936fb9fb291ef9b9cfa88f51576cdc19abbd34240232ce284958ac7dbaf" "" || (  /home/user/wrk/openmptcprouter/gl_mifi/source/scripts/dl_github_archive.py --dl-dir="/home/user/wrk/openmptcprouter/gl_mifi/source/dl" --url="https://github.com/golang/sys.git" --version="63fc586f45fe72d95d5240a5d5eb95e6503907d3" --subdir="golang-golang-x-sys-20180621-63fc586f" --source="golang-golang-x-sys-20180621-63fc586f.tar.xz" --hash="3afe7936fb9fb291ef9b9cfa88f51576cdc19abbd34240232ce284958ac7dbaf" || ( 	echo "Checking out files from the git repository..."; mkdir -p /home/user/wrk/openmptcprouter/gl_mifi/source/tmp/dl && cd /home/user/wrk/openmptcprouter/gl_mifi/source/tmp/dl && rm -rf golang-golang-x-sys-20180621-63fc586f && [ \! -d golang-golang-x-sys-20180621-63fc586f ] && git clone  https://github.com/golang/sys.git golang-golang-x-sys-20180621-63fc586f && (cd golang-golang-x-sys-20180621-63fc586f && git checkout 63fc586f45fe72d95d5240a5d5eb95e6503907d3 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd golang-golang-x-sys-20180621-63fc586f && git log -1 --format='\''@%ct'\''` && rm -rf golang-golang-x-sys-20180621-63fc586f/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c golang-golang-x-sys-20180621-63fc586f | 	xz -zc -7e > /home/user/wrk/openmptcprouter/gl_mifi/source/tmp/dl/golang-golang-x-sys-20180621-63fc586f.tar.xz && mv /home/user/wrk/openmptcprouter/gl_mifi/source/tmp/dl/golang-golang-x-sys-20180621-63fc586f.tar.xz /home/user/wrk/openmptcprouter/gl_mifi/source/dl/ && rm -rf golang-golang-x-sys-20180621-63fc586f; );  )    '
touch /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/golang-golang-x-sys-20180621-63fc586f/.prepared_162f7d2fa7ec674dc7bcbc41f4fc0200_18f1e190c5d53547fed41a3eaa76e9e9_check
. /home/user/wrk/openmptcprouter/gl_mifi/source/include/shell.sh; xzcat /home/user/wrk/openmptcprouter/gl_mifi/source/dl/golang-golang-x-sys-20180621-63fc586f.tar.xz | tar -C /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/golang-golang-x-sys-20180621-63fc586f/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/golang-golang-x-sys-20180621-63fc586f
touch /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/golang-golang-x-sys-20180621-63fc586f/.prepared_162f7d2fa7ec674dc7bcbc41f4fc0200_18f1e190c5d53547fed41a3eaa76e9e9
rm -f /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/golang-golang-x-sys-20180621-63fc586f/.configured_*
rm -f /home/user/wrk/openmptcprouter/gl_mifi/source/staging_dir/target-mips_24kc_musl/stamp/.golang-golang-x-sys_installed
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/user/wrk/openmptcprouter/gl_mifi/source/.go-build" GO_MOD_CACHE_DIR="/home/user/wrk/openmptcprouter/gl_mifi/source/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="golang.org/x/sys" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="1" GO_BUILD_PKG="golang.org/x/sys/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/golang-golang-x-sys-20180621-63fc586f" GO_BUILD_DIR="/home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/golang-golang-x-sys-20180621-63fc586f/.go_work/build" GO_BUILD_BIN_DIR="/home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/golang-golang-x-sys-20180621-63fc586f/.go_work/build/bin/linux_mips" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/user/wrk/openmptcprouter/gl_mifi/source/staging_dir/target-mips_24kc_musl/usr/share/gocode/src" /usr/bin/env bash ..//golang-build.sh configure
Copying files from /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/golang-golang-x-sys-20180621-63fc586f into /home/user/wrk/openmptcprouter/gl_mifi/source/build_dir/target-mips_24kc_musl/golang-golang-x-sys-20180621-63fc586f/.go_work/build/src/golang.org/x/sys
cpu/cpu.go
cpu/cpu_arm.go
cpu/cpu_arm64.go
cpu/cpu_gc_x86.go
cpu/cpu_gccgo.c
cpu/cpu_gccgo.go
cpu/cpu_mips64x.go
cpu/cpu_mipsx.go
cpu/cpu_ppc64x.go
cpu/cpu_s390x.go
cpu/cpu_test.go
[... cut ...]
make[2]: Leaving directory '/home/user/wrk/openmptcprouter/gl_mifi/source'
make[1]: *** [package/Makefile:108: /home/user/wrk/openmptcprouter/gl_mifi/source/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/wrk/openmptcprouter/gl_mifi/source'
make: *** [/home/user/wrk/openmptcprouter/gl_mifi/source/include/toplevel.mk:242: world] Error 2
user@user:~/wrk/openmptcprouter$ 

ps: i did everything by your instruction, step-by-step. something is missing?

spacedream, 2021-02-18:
[image]

spacedream, 2021-03-22:
I found it too late, just for reference:
https://github.com/Ysurac/openmptcprouter/issues/1343

closed 2021-06-25 by github-actions[bot]