Ysurac/openmptcprouter #1563: I have 4 modems connected with multiple lan connections

dooley1981, 2021-02-02:
I have 4 modems but when I look at the lan connections on each modem. It shows that it's connected to 10 to 11 devices. The devices are connecting to each other though. The DHCP is turned off on the modems so it's not sending out IP addresses to other devices other than each other modems. 
[image]

hairydogltd, 2021-02-15:
A screencapture of the relevant status page (/cgi-bin/luci/admin/system/openmptcprouter/status) would be a thousand times more useful than a photo of four modems. 
The modems are reporting that they can detect ten devices on the network. Mine currently reports there are thirteen, though Fing reports thirty-seven. I suggest that you don't worry about it.
However, as you have posted that picture, I suggest that is is a slightly daft setup. I'd put them at least a metre apart from each other.

Adorfer, 2021-02-16:
judging by the picute: mounting the nighthawks without at least 50cm of distance (preferrbly 1m) will heavily degrade performance due to interference in the nearfield of the modems. 
(And this even applies even if there are external antennas in use)

dooley1981, 2021-02-17:
I spaced out the modems about 3 feet apart. They do not perform any different for what I can tell. My speeds have been terrible lately. I should be getting around 200+ down between 4 modems. Now i'm getting just 60 down with 4 modems. When I test one single modem without the OpenMPTCProuter, I get around 70 down on each of the 3 modems, the 4 modem is around 30 down which I expect.

Ysurac, 2021-02-17:
If all modems are connected to same provider/antenna it can be logical.
To test speed use ```omr-test-speed``` on the router via SSH (ctrl+c to stop after at least 2 minutes). Check when during test bandwidth usage in Network->MPTCP

spacedream, 2021-02-17:
maybe external antennas will/may helps too.

dooley1981, 2021-02-17:
> If all modems are connected to same provider/antenna it can be logical.
> To test speed use `omr-test-speed` on the router via SSH (ctrl+c to stop after at least 2 minutes). Check when during test bandwidth usage in Network->MPTCP

> maybe external antennas will/may helps too.

I don't have any issues with signal nor speed, only when connected to the Openmptcprouter.  I even tried several VPS servers to rule that out as well.

Ysurac, 2021-02-17:
So we can close this issue if there is no more issue ?

dooley1981, 2021-02-17:
I have the issue with speed when going through Openmptcprouter only.

Ysurac, 2021-02-17:
So it's why I tell you to use omr-test-speed.

dooley1981, 2021-02-18:
you can close this issue. I tested it with omr-test-speed. It was pretty fast. On some test I got failed ipv4. But it did eventually work.

closed 2021-05-24 by github-actions[bot]