Ysurac/openmptcprouter #1564: OMR VPS in Orange status No result No answer

vinceducat, 2021-02-02:
## Expected Behavior
OMR status in green

## Current Behavior
OMR status in orange
No result - No script answer

## Possible Solution
dont know restart OMR and OMR VPS

## Steps to Reproduce the Problem

  1. do nothing special
  2. status become orange ...
  3.

## Context (Environment)
not really a problem but this appear so i want to understant

## Specifications

  - OpenMPTCProuter version : 0.57.3
  - OpenMPTCProuter VPS version: 0.1024 5.4.86-mtcp
  - OpenMPTCProuter platform: x86_64

vinceducat, 2021-02-02:
thanks to Ycarus the problem was a corrupt file on the omr vps :  /etc/openmptcprouter-vps-admin/omr-admin-config.json  restore via wget from develop https and adding server key in the script .

can be close ;-)

rlopessilva, 2021-02-03:
I had the same "bug":
user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, server API answer)
Like you, I changed the user_password for user openmptcprouter in /etc/openmptcprouter-vps-admin/omr-admin-config.json with the server key from /root/openmptcprouter_config.txt and everithing is green again.
Thank you :)

vinceducat, 2021-02-03:
glade that is help you ;-)

closed 2021-02-04 by vinceducat