Ysurac/openmptcprouter #1565: Suggestion to include "nano" as a default package

milankragujevic, 2021-02-03:
**Is your enhancement proposal related to a problem? Please describe.**
In some situations, especially with multiple Ethernet adapters on VirtualBox, it is necessary to manually edit config files from the console (either VBox RDP or actual display and keyboard attached to device) to get the network up and running properly. 

Currently only **vi** is included with the OpenWrt install, so while it is possible, it's also not ergonomic for those not using EN keyboard layout, as **vi** is infamously complicated (_we all know the "exit vim" meme_) and relies heavily on keyboard shortcuts, some of which don't work over WebVNC.

**Describe the solution you'd like**
I would appreciate the **nano** package being included in the default distro. It is not big, for example years ago when I used to run OpenWrt on TP-Link WR740N I would add nano even though the device only had 4 MB of SPI Flash

spacedream, 2021-02-06:
i voting for nano too! :)

Ysurac, 2021-02-06:
nano is added in develop branch, so this will be in next release.
midnight commander is available and should help in most case in current release.

closed 2021-02-06 by milankragujevic

spacedream, 2021-02-12:
thank you! 
ps: mcedit if useful but in some cases nano is prefer.