Ysurac/openmptcprouter #1569: Latest build error on tcp_bbr2.o

adricoues, 2021-02-03:
Bonjour à tous, je me permets de venir vers vous car je rencontre un problème pendant la compilation de la dernière revision. Le problème serait situé au niveau de la congestion tcp bbr. Voici les logs: [http://pastebin.fr/83925](url)

Merci pour votre aide :) 

In English 
----

Hello everyone, I take the liberty of coming to you because I encountered a problem while compiling the latest revision. The problem would be located in the tcp bbr congestion. Here are the logs: http://pastebin.fr/83925

Thanks for your help :)

Ysurac, 2021-02-03:
BBR2 only compile with 64bits processor, so it must not be enabled on any 32bits processor like this one.

adricoues, 2021-02-03:
Thanks for your reply. 

Is it possible to enable this or bypass this error ?

Thanks in advance

Ysurac, 2021-02-03:
As I said, you need to disable BBR2. You may have to use ```make kernel_menuconfig```.

adricoues, 2021-02-03:
I'm testing this, thank you

closed 2021-05-10 by github-actions[bot]