Ysurac/openmptcprouter #1571: route/firewall issues for lan zone (i think)

zulu14, 2021-02-05:
## Expected Behavior
omr supply internet to all letwork
## Current Behavior
omr router has internet with diagnostic page or ping in console, but not the lan network when is in direct output 

## Possible Solution
the route seem to be ok,
[image]

ip a command:
[image]

luci value:

config settings 'settings'
	option enabled '1'
	option version '0.57.3'
	option scaling_min_freq '1500000'
	option scaling_max_freq '1500000'
	option scaling_governor 'performance'
	option disable_ipv6 '1'
	option check_ipv4_website 'http://ip.openmptcprouter.com'
	option check_ipv6_website 'http://ipv6.openmptcprouter.com'
	option status_vps_timeout '2'
	option status_getip_timeout '2'
	option enable_nodelay '1'
	option menu 'OpenMPTCProuter'
	option serial '1000000031e1a37e'
	option vpn 'glorytun_tcp'
	option firstboot '0'
	option proxy 'shadowsocks'
	option apilc '1612563545'
	option disable_fastopen '0'
	option external_check '1'
	option disablegwping '0'
	option disableintfrename '0'
	option defaultgw '1'
	option tracebox '1'
	option disableserverping '0'
	option shadowsocksudp '0'
	option master 'change'
	option vnstat_backup '0'
	list master_lcintf 'wan2'
	list master_lcintf 'wan1'
	option debug '1'

config router 'omr'
	option detected_public_ipv4 '<ip corespond to any of wan1 or wan2>'

config interface 'wan1'
	option metric '3'
	option multipathvpn '0'
	option state 'up'
	option publicip '<ip has ok>'
	option asn 'IP2000-ADSL-BAS'
	option mtu '1500'
	option lc '1612564615'
	option multipath 'master'
	option latency_previous '29'
	option latency '28'

config interface 'wan2'
	option metric '4'
	option multipathvpn '0'
	option state 'up'
	option publicip '<ip has ok>'
	option asn 'FR-PROXAD-ADSL'
	option mtu '1500'
	option manufacturer 'huawei'
	option lc '1612564550'
	option multipath 'on'
	option latency_previous '35'
	option latency '36'

config interface 'omrvpn'
	option multipath 'off'
	option metric '1200'

config interface 'loopback'
	option multipath 'off'
	option metric '5'

config interface 'lan'
	option multipath 'off'
	option metric '6'
	option multipathvpn '0'

config interface 'omr6in4'
	option multipath 'off'
	option metric '1201'

config latest_versions 'latest_versions'
	option omr '0.57.3'
	option vps '0.1024'
	option lc '1612563341'

config server 'cbm1'
	option username 'openmptcprouter'
	option password '<key>'
	option ip '51.178.122.227'
	option port '65500'
	option master '1'
	option backup '0'
	option lastchange '1612562977'
	option token '<key>'
	option redirect_ports '0'
	list available_vpn 'glorytun_tcp'
	list available_vpn 'glorytun_udp'
	list available_vpn 'dsvpn'
	list available_vpn 'openvpn'
	list available_vpn 'openvpn_bonding'
	list available_vpn 'mlvpn'
	option get_config '0'
	option pihole '0'
	option kernel '5.4.86-mptcp'
	option machine 'x86_64'
	option omr_version '0.1024'
	option admin_error '1'
	option nofwredirect '0'

## Context (Environment)
whe he make a setup wizard

## Specifications

  - OpenMPTCProuter version: 57.3
  - OpenMPTCProuter VPS version: 1024
  - OpenMPTCProuter platform: <rpi4 etx4 factory img

Ysurac, 2021-02-06:
Should work, but I think the problem is shadowsocks in an unknown state and shadowsocks rules still active.
What is the result of  ```ps w | grep ss``` and ```iptables-save | grep ss``` on the router via SSH or via Services->Terminal ?

zulu14, 2021-02-06:
for ps:
[image]

for iptables 3screen):
[image]
[image]
[image]

zulu14, 2021-02-06:
config:

shadowsocks +glorytun tcp 
= state wan1 wan2 green omr red (ss nit working)  vps yellow (direct output) omr has internal connectivity (cmd ping 1.1.1.1) device on 192.168.100.1 network has not connectivity

v2ray +glorytun tcp 
= state wan1 wan2 green omr green.  vps yellow (direct output) omr has internal connectivity (cmd ping 1.1.1.1) device on 192.168.100.1 network has not connectivity

none proxy  +glorytun tcp 
= state wan1 wan2 green omr yellow( normal proxy disabled but glorytun has up)
vps yellow (direct output) omr has internal connectivity (cmd ping 1.1.1.1) device on 192.168.100.1 network has not connectivity

the same test with glorytudp make same result. 

after this i have tested with mlvpn

mlvpn without proxy all things working

mlvpn with ss has no internet connectivity for device but omr shell has connectivity (ping 1.1.1.1) 

mlvpn with v2ray all things working

after this i have retry the same glorytun process and now is working  only with v2ray proxy or none proxy

for remove the maxium of failure possibilites each proxy setup has make with no encryption method

every time i setup the wizard some init.d service restart but its not included into the openmptcprouter controller wizard_add function,
 and the first issues is each time it logout luci interfaces when you apply wizard (multiple browser tested chrome edge opera) and when you have ss activated it logout aprox each minutes, but when i logout aprox each minutes the system not restart the service just shadowsocks are restart by ss-trackee

to resume i think he has an issues into the post-tracking  or wizardadd in a specific case , mlvpn setup seem to fix it for other tunnel after it for v2ray and none case

in each case when ss are activated the connectivity are loss for external device omr internal are ok

we think ss  or ss tracker or omrtracker make a failure into the luci session 
when ss is setup is logout, and when you apply wizard you restart in controller shadowsock its correspond  to the issues

on new post you will find the shadowsocks uci config:

Ysurac, 2021-02-06:
It's not a good idea to put keys and IP in issue...

zulu14, 2021-02-06:
its a test environnement its not a problem

Ysurac, 2021-02-06:
There is a problem with the multiples IPs on the VPS. Server are set to "192.168.1.3", it's the default setting and should not be here. But having multiples IPs may explain the issue, it's still a really beta feature.

zulu14, 2021-02-06:
ok i see

can omrip0 / 1 failure block sss0 and h1 / 2/3/4?

can i manually remove omrip1 / 2 or is the function that autosetup them to temporarily disable it?

anyway he has another issues but verry hard to find it now mlvpn has fixed it,

and luci logout issues is no strange

zulu14, 2021-02-06:
rectification it also disconects the web interface when we add or remove an interface from the wizard add when we reconnect we have an http error 405 and the url is always on
https://192.168.100.1/cgi-bin/luci/admin/system/openmptcprouter/wizard_add
In other words, the problem happens before the buildurl therefore probably at the level of network interfaces? (uci commit ? )

zulu14, 2021-02-06:
another bug;
when you add an interface the firewall not asign it to wan zone
[image]

the issues it is add another list and not add the value to actual list
[image]

zulu14, 2021-02-06:
for you a fix for firewall:
not tested: https://codeshare.io/amnqg0

Ysurac, 2021-02-06:
wan6 is normal, it's a default value. I may remove it.
But the fact that it's not a list is not normal, but an "issue" in OpenWrt code... I will try to fix it when I add an interface.

closed 2021-05-12 by github-actions[bot]