Ysurac/openmptcprouter #1572: Le VPN n'est pas lancé

babyss69, 2021-02-08:
Hello ,
i installed OpenMPTCProuter, but it tells me the VPN is not running, I am using a raspberry pi3, whatever I do it does not work.
I believed it was from VPS, I tried with debian 9 and 10.
on 2 different Vps one che OVH and the other at vultr.
It always gives me the same error message.

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform: 

OpenMPTCProuter (192.168.100.1)

Version 0.57.3Charge : 0.39 0.77 0.60Température du cœur : 62.3 °Durée de fonctionnement : 14h 16m 35sLan adresse : 

OpenMPTCProuter platform: RPI3

Ysurac, 2021-02-08:
What is the VPN used ? By default it's Glorytun TCP.
What is the configuration you set in the wizard ?

babyss69, 2021-02-08:
I use OpenMPTCProuter VPS 0.1024
Yes i set by wizard

babyss69, 2021-02-08:
Oui j'ai utilisé le wizard et j'utilise Gloryturn TCP, dans l'interface web il est noté activé

Ysurac, 2021-02-08:
VPS was rebooted after install ?
All keys are available in the wizard ?
All in green in status page ?

babyss69, 2021-02-08:
Et ce qui est étrange aussi c'est que la configuration ADSL qui est en multipath , un coup tout est au vert et un coup c'est orange , et cela me dis que la reponse a dépassé 1 seconde.

Oui j'ai rebooter le serveur après l'installation.
Oui j'ai mis toutes les clés dans wizard.
non il y a des choses qui sont orange.
1) OpenMPTCProuter (192.168.100.1)Le VPN n'est pas lancé  et la croix est orange

2) par exemple sur le server , la croix est au vert , mais ca me dis que c'est impossible d'accédé au script d'administration

Ysurac, 2021-02-08:
Check in State->System log what you have.
What is the latency of your connections ?

babyss69, 2021-02-08:
openmptcprouter.wan1.latency_previous='199'
openmptcprouter.wan1.latency='127'

openmptcprouter.wan2.latency_previous='140'
openmptcprouter.wan2.latency='559'

Ysurac, 2021-02-08:
Latencies are really high...
You checked the log ?

babyss69, 2021-02-08:
oui j'ai checké le log.
J'essaye d'utilisé votre solution , car la je suis en Guadeloupe a la campagne.
j'ai une livebox , avec un débit de 1 megabits en descendant .
et j'essaye de le couplé à une 4 g  qui monte a 4 megabits via du roaming au caraibes

Ysurac, 2021-02-08:
Dans les logs il y a normalement des informations sur le soucis que rencontre le VPN ou les soucis détectés certainement avec par OMR-Tracker.
Il faut trouver un VPS avec la latence la plus faible possible, la ça me semble un peu haut.

babyss69, 2021-02-08:
OMR OpenMPTCProuter
Dashboard
État
Vue d’ensemble
Pare-feu
Routes
Journal système
Journal du noyau
Processus
Graphiques temps-réel
Surveillance du trafic VnStat
Système
OpenMPTCProuter
Système
Administration
Paramètres ACL
Logiciels
Démarrage
Tâches Régulières
Points de montage
Configuration des DEL
Sauvegarde / Mise à jour du micrologiciel
Redémarrage
Shutdown
Services
OMR-Tracker
iPerf
Proxy DNS Over HTTPS
Shadowsocks-libev
OMR-Bypass
Packet Capture
E-Mail
Wake on LAN
Terminal
uHTTPd
UPnP
DNS récursif
Réseau
MPTCP
Interfaces
Sans-fil
DHCP et DNS
Noms d'hôtes
Routes statiques
SQM QoS
Pare-feu
Diagnostiques
OMR-DSCP
Quota
SNMPd
VPN
MLVPN
Glorytun TCP
Glorytun UDP
DSVPN
OpenVPN
Déconnexion
Pas de mot de passe positionné !
Ce routeur n'a pas de mot de passe configuré. Veuillez configurer un mot de passe pour l'utilisateur root pour protéger l'accès de votre interface web.

Assistant de configuration
État
Paramètres avancés
Sauvegarde sur le serveur
Voir tous les paramètres
Tous les paramètres du routeur
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
[... 1577 lines cut ...]
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='11'
wireless.radio0.hwmode='11g'
wireless.radio0.path='platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='HT20'
wireless.radio0.country='00'
wireless.radio0.cell_density='0'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.mode='ap'
wireless.default_radio0.ssid='OpenWrt'
wireless.default_radio0.encryption='none'
wireless.default_radio0.skip_inactivity_poll='1'
wireless.default_radio0.network='wifi'
wireless.default_radio0.disabled='1'
wireless.wifinet1=wifi-iface
wireless.wifinet1.device='radio0'
wireless.wifinet1.mode='sta'
wireless.wifinet1.network='wwan'
wireless.wifinet1.ssid='fsa'
wireless.wifinet1.encryption='psk2'
wireless.wifinet1.key='fabrice971'
Powered by openmptcprouter v0.57.3 r0+15225-bfc433efd4

babyss69, 2021-02-28:
Bonjour, 
A force de chercher , j'ai trouvé la solution .
Le problème venais de la configuration de mon modem 4G , un TP Link; Archer  MR 200.
Celui ci était configuré sur  IPV4 & IPV6, des que j'ai configuré sur exclusivement IPV4 , la latence de la ligne est devenu normale , et la communication entre OMR et le VPS a pu être établi.
Le VPN fonctionne , tout est passé au vert.

closed 2021-06-03 by github-actions[bot]