Ysurac/openmptcprouter #1573: error -99

sonnyxp2008, 2021-02-08:
hi everyone, you know give me some indication about this error?

Mon Feb  8 14:32:46 2021 kern.err kernel: [579264.238842] __mptcp_init4_subsockets: token 0xfa8da51d bind() to 192.168.43.81 index 12 failed, error -99
Mon Feb  8 14:32:46 2021 kern.err kernel: [579264.486795] __mptcp_init4_subsockets: token 0xb1997929 bind() to 192.168.43.81 index 12 failed, error -99

Ysurac, 2021-02-08:
Not possible to create subflow binded on 192.168.43.81 ip.

sonnyxp2008, 2021-02-08:
excuse my ignorance but what is this flow?

Ysurac, 2021-02-08:
With MPTCP, each network interface with an IP is called a subflow. It's a connection (that can have a different internet path) identified by source and destination IP addresses.

sonnyxp2008, 2021-02-08:
so could this be the cause that the connection is not working? it gives me ping reply error after 1 second. I also tried other devices and mobiles. The problem remains

Ysurac, 2021-02-08:
Yes it's possible.

sonnyxp2008, 2021-02-08:
and how can I solve the problem? depends on what?

Ysurac, 2021-02-08:
The problem is not this error, but the connection configuration.

sonnyxp2008, 2021-02-08:
I also tried via mobile, the connection goes for about 30 seconds and then drops. I tried all the configurations but nothing to do
are you referring to the configuration of the omr?

Ysurac, 2021-02-08:
yes, but open a new issue as it's not related to this kernel error, and I will need OMR versions and the result of ```uci show network``` via SSH or Services->Terminal.

sonnyxp2008, 2021-02-08:
ok merci

closed 2021-05-14 by github-actions[bot]