Ysurac/openmptcprouter #1574: error -99 / down ping after 1 second

sonnyxp2008, 2021-02-08:
## Expected Behavior

create a connection with 4g device

## Current Behavior

the connection stays active for about 30 seconds and then drops. ping server error after one second.
kernel error -99

## Specifications

  - OpenMPTCProuter version:  0.57.3
  - OpenMPTCProuter VPS version:  0.1024 5.4.86
  - OpenMPTCProuter platform:  x86_64
  
paste result of the show network
root@OpenMPTCProuter:~# uci show network
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.macaddr='00:00:00:00:00:00'
network.loopback.metric='7'
network.globals=globals
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='blest'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='2'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.ula_prefix='fdd0:cdf4:aa30::/48'
network.globals.packet_steering='1'
network.globals.congestion='cubic'
network.globals.mptcp_checksum='1'
network.lan=interface
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.ipv6='0'
network.lan.delegate='0'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.macaddr='08:60:6e:75:b9:09'
network.lan.metric='8'
network.lan.defaultroute='0'
network.lan.peerdns='0'
network.lan.ifname='eth1'
network.lan.label='LAN1'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.defaultroute='0'
network.wan1.macaddr='50:3e:aa:0f:c3:78'
network.wan1.peerdns='0'
network.wan1.ipv6='0'
network.wan1.metric='9'
network.wan1.ifname='eth4'
network.wan1.label='CONNESSIONE W'
network.wan1.netmask='255.255.255.0'
network.wan1.addlatency='0'
network.wan1.dns='8.8.8.8' '1.1.1.1'
network.wan1.multipath='master'
network.wan1.ipaddr='192.168.1.143'
network.wan1.gateway='192.168.1.1'
network.wan2=interface
network.wan2.proto='static'
network.wan2.ip4table='wan'
network.wan2.defaultroute='0'
network.wan2.macaddr='3c:49:37:07:31:5d'
network.wan2.peerdns='0'
network.wan2.ipv6='0'
network.wan2.metric='10'
network.wan2.ifname='eth0'
network.wan2.label='CONNESSIONE T'
network.wan2.netmask='255.255.255.0'
network.wan2.addlatency='0'
network.wan2.dns='8.8.8.8' '1.1.1.1'
network.wan2.ipaddr='192.168.1.188'
network.wan2.gateway='192.168.1.1'
network.wan2.multipath='on'
network.omrvpn=interface
network.omrvpn.ifname='tun0'
network.omrvpn.ip4table='vpn'
network.omrvpn.leasetime='12h'
network.omrvpn.txqueuelen='100'
network.omrvpn.ipv6='0'
network.omrvpn.metric='1200'
network.omrvpn.defaultroute='0'
network.omrvpn.peerdns='0'
network.omrvpn.multipath='off'
network.omrvpn.addlatency='0'
network.omrvpn.proto='none'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.auto='0'
network.omr6in4.metric='1201'
network.omr6in4.ip6addr='fe80::a00:2/126'
network.omr6in4.gateway='fe80::a00:1/126'
network.omr6in4.ipaddr='10.255.255.2'
network.omr6in4.peeraddr='10.255.255.1'
network.4g=interface
network.4g.ifname='eth2'
network.4g.ipv6='0'
network.4g.proto='static'
network.4g.multipath='on'
network.4g.addlatency='0'
network.4g.ipaddr='192.168.88.250'
network.4g.netmask='255.255.255.0'
network.4g.gateway='192.168.88.1'
network.4g.metric='11'
network.4g.defaultroute='0'
network.4g.peerdns='0'
network.4g.macaddr='3c:49:37:07:31:5d'

Ysurac, 2021-02-08:
wan2 and 4g have same mac address set. Reset them to empty value in Network->Interface

sonnyxp2008, 2021-02-08:
fatto, una volta salvato ritornano gli stessi indirizzi mac, tutti e due uguali

Ysurac, 2021-02-08:
Try to change one to another mac.

sonnyxp2008, 2021-02-08:
one randomly chosen by me?

Ysurac, 2021-02-08:
yes, increase a number or letter

sonnyxp2008, 2021-02-08:
I'm sorry but despite everything it still doesn't work could it be a driver problem?

Ysurac, 2021-02-08:
You changed mac addresses ?
Check what you have in State->System log.

sonnyxp2008, 2021-02-08:
i checked and the address changed it. I noticed some strange behavior. In practice we have 4 network cards, one of which is disabled. Regardless of where I put the ethernet cable, it recognizes it as ethernet 2. Both on the activated and the deactivated port

sonnyxp2008, 2021-02-08:
I notice that it has difficulty with eth2 and eth3, now after fiddling with a little bit they both stopped working

Ysurac, 2021-02-08:
As I said, check logs.

Ysurac, 2021-02-08:
Many errors here.
This can be related to faulty hardware or driver.

sonnyxp2008, 2021-02-09:
here are the last things I did: I changed operator, I changed 4g device, I changed the network cards, I tried a usb card (the ones that don't work after a month) nothing to do, do you think that at this point a new installation clean could solve the problem? Or is it still possible that as soon as many requests arrive something goes haywire?

Ysurac, 2021-02-09:
No idea, check the log when you change something and have issue.

sonnyxp2008, 2021-02-09:
i checked the log it gives me error that it can't see the host. But I am able to ping the host. I am able to enter the interface without problems
Tue Feb  9 15:52:53 2021 user.notice post-tracking-post-tracking: 4g (eth5) switched off because ping from 192.168.88.250 error (8.8.4.4,1.1.1.1,8.8.4.4)
Tue Feb  9 15:52:53 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.88.1 dev eth5
Tue Feb  9 15:52:56 2021 daemon.err /usr/bin/ss-redir[20053]: server recv: Operation timed out
Tue Feb  9 15:53:10 2021 user.notice post-tracking-post-tracking: 4g (eth5) switched up
Tue Feb  9 15:54:01 2021 user.notice post-tracking-post-tracking: 4g (eth5) switched off because ping from 192.168.88.250 error (1.1.1.1,8.8.4.4,1.1.1.1)
Tue Feb  9 15:54:01 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.88.1 dev eth5
Tue Feb  9 15:54:47 2021 user.notice post-tracking-post-tracking: 4g (eth5) switched up

Ysurac, 2021-02-09:
What you mean by "enter the interface" ?

sonnyxp2008, 2021-02-09:
modem interface what you see marked as 192.168.88.1

Ysurac, 2021-02-09:
The error indicate that internet is not available using the modem, not that gateway doesn't answer.

sonnyxp2008, 2021-02-09:
but if I attach everything to the pc it works if I attach it to all omr it doesn't work

Ysurac, 2021-02-09:
What you mean but it works ?
As you can see ping doesn't answer using the interface... If the configuration is correct it's not related to OMR.
Check also kernel log, if you still have kernel errors

sonnyxp2008, 2021-02-09:
I have no errors in the kernel, I mean that if I insert the cable on the pc, the connection is there, and I have tried all day, if I insert it to the omr it gives me ok for a few seconds and then loses the connection

Ysurac, 2021-02-09:
Paste the full configuration in System->OpenMPTCProuter, last tab. Without keys and VPS IP.

sonnyxp2008, 2021-02-09:
it is very long but I will try to glue it. I was thinking of opting for a paid service. Do you always take care of it?

sonnyxp2008, 2021-02-09:
i deleted ip and keys, i hope i deleted them all
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/netflix.com/omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/nflxsearch.net/omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/nflxso.net/omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/btstatic.com/omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/customerevents.netflix.com/omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/movies.netflix.com/omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/nccp.netflix.com/omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/nflxext.com/omr_dst_bypass_eth4,omr6_dst_bypass_eth4,omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/nflximg.com/omr_dst_bypass_eth4,omr6_dst_bypass_eth4,omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/nflximg.net/omr_dst_bypass_eth4,omr6_dst_bypass_eth4,omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/nflxvideo.net/omr_dst_bypass_eth4,omr6_dst_bypass_eth4,omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/secure.netflix.com/omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/dazn.com/omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/dazn.com/en-IT/home/omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/13.225.65.122/omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/dazn.it/omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/xbox.com/omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/xboxlive.com/omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/xboxlive.com.akadns.net/omr_dst_bypass_eth4,omr6_dst_bypass_eth4' '/xboxlive.com.c.footprint.net/omr_dst_bypass_eth4,omr6_dst_bypass_eth4'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.tim4gnew=dhcp
dhcp.tim4gnew.interface='tim4gnew'
dhcp.tim4gnew.start='100'
dhcp.tim4gnew.limit='150'
dhcp.tim4gnew.leasetime='12h'
dhcp.tim4gnew.force='1'
dhcp.tim4gnew.dynamicdhcp='0'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
[... 1731 lines cut ...]
[... cut ...]
v2ray.omrrouting=routing_rule
v2ray.omrrouting.type='field'
v2ray.omrrouting.inbound_tag='omrbridge'
v2ray.omrrouting.outbound_tag='omrout_tunnel'
v2ray.omrrouting.domain='full:omr.lan'
v2ray.omrroutingo=routing_rule
v2ray.omrroutingo.type='field'
v2ray.omrroutingo.inbound_tag='omrbridge'
v2ray.omrroutingo.outbound_tag='out'
v2ray.main_routing=routing
v2ray.main_routing.enabled='1'
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth1' 'eth2' 'eth4'

Ysurac, 2021-02-09:
remove DNS on all interfaces.
Also why you removed IP on omr-tracker ? It need more than 2. There is also an error in omrvpn tracker, it should be set to none and not to ping.

sonnyxp2008, 2021-02-09:
you find those changes just because I've been going crazy for 3 weeks. I changed them but to no avail. From the 4g modem I see that there is a consumption and various requests. In practice, the download seems to work, the upload does not. I don't know if it creates other problems. But is it possible to request a quote to have a system based on our needs?

Ysurac, 2021-02-09:
Your problem seems to be hardware based, I don't provide any hardware.
It's working without the 4g ?

sonnyxp2008, 2021-02-09:
it could be a hardware problem even if I tried with 3 mobile phones 4 network cards 3 different operators. The only thing I haven't done yet is: I put a switch to be able to connect two routers with two different IPs. Delete this switch even if they work. Clean OMR re-installation because I made a lot of changes over the last few weeks. change pc even if it is a 6 core amd with 8 gb of ram. Beyond everything I would like an ad hoc paid program

Ysurac, 2021-02-09:
Kernel error was related to hardware issues.
You can check if your 4G connection allow ping or if it's blocked, but with 3 different operators this would be strange.
And again, it's working without the 4G connection or not ?
I can make you pay, but I don't know why. But if you don't want any free support, no problem.

sonnyxp2008, 2021-02-09:
the other 2 are two vdsl and always work. Only 4g doesn't work even though I need it a lot. For the question of payment I need a software for our needs, practical and fast. Then for the rest for me ORM is very easy and very functional

Ysurac, 2021-02-09:
Can you test a ping using 4g connection from another computer ? Ping can be blocked and in this case omr-tracker need to use another method than ping for this connection.

sonnyxp2008, 2021-02-09:
I did not understand your question you say to try another pc with only the 4g connection?

Ysurac, 2021-02-09:
Yes. And try to run a ping like ping 8.8.8.8 from this pc.
You can also add 4g interface in omr-tracker and set type to DNS (or none to disable tracker on 4g for test).

sonnyxp2008, 2021-02-09:
yes i was planning to do it tonight, several attempts and a clean install on ssd (i know you will take me for crazy). Beyond everything can we collaborate for a commercial function? obviously paying for your work

Ysurac, 2021-02-09:
You can contact me on contact@openmptcprouter.com for any commercial quote.

sonnyxp2008, 2021-02-09:
are you answering me yourself?

Ysurac, 2021-02-09:
As I'm alone yes, it's always me that answer to mails ;)

sonnyxp2008, 2021-02-09:
wonderful, tomorrow I'll send you the email so you understand better all the plot that I have combined

closed 2021-05-16 by github-actions[bot]