Ysurac/openmptcprouter #1575: please help not getting bonded speeds

Nickmcl024, 2021-02-08:
please help 

I have done the default set up 2 wan connections but am not getting a bonded speed  

router settings 
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301,65011'
dscp.@classify[3].comment='OMR vpn'
[... 1565 lines cut ...]
v2ray.omrrouting.domain='full:omr.lan'
v2ray.omrroutingo=routing_rule
v2ray.omrroutingo.type='field'
v2ray.omrroutingo.inbound_tag='omrbridge'
v2ray.omrroutingo.outbound_tag='out'
v2ray.main_routing=routing
v2ray.main_routing.enabled='1'
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.radio0.country='00'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.mode='ap'
wireless.default_radio0.ssid='OpenWrt'
wireless.default_radio0.encryption='none'
wireless.default_radio0.skip_inactivity_poll='1'
wireless.default_radio0.network='wifi'

Ysurac, 2021-02-08:
Please use the template when opening an issue.

Nickmcl024, 2021-02-08:
what template do i use? 

super new to this and just wana test out the software for bonding

Ysurac, 2021-02-08:
The template is the text you removed when you opened the issue that ask versions, platform,...
You can't expect to aggregate same speed on a RPI2B and a new x86_64 server, it's why I need some infos asked in template.
And as it's a speed problem, I would need some speed info ;)
To test aggregate speed you can use ```omr-test-speed``` on the router via SSH or Services->Terminal. To test speed for each wan ```omr-test-speed wan1``` (for the speed of wan1, check interfaces name using ```ip a```). You can stop speed after at least 2 minutes using "Ctrl+C".

Nickmcl024, 2021-02-08:
## Expected Behavior
not getting bonded speeds 

## Current Behavior
wan1 200dl 100up
wan2 200dl 200up 

only getting max 200dl and 200up

## Specifications

  - OpenMPTCProuter version Kernel Version：5.4.83 
                                               
  - OpenMPTCProuter VPS version: 5.4.86-mptcp 0.1024
  - OpenMPTCProuter platform: <Model：Raspberry Pi 4 Model B Rev 1.2

-

Nickmcl024, 2021-02-08:
i tested all it all in termail 

speeds are the same 
wan1 20MB
wan2 20MB

Ysurac, 2021-02-08:
I still need ```omr-test-speed``` results.
But as it's a RPI4, doesn't expect too much speed. Check RPI load average and VPS load average.
You can also try to disable shadowsocks encryption in System->OpenMPTCProuter, "advanced settings" checkbox, to test if CPU limit speed.

Adorfer, 2021-02-09:
>     * OpenMPTCProuter platform: <Model：Raspberry Pi 4 Model B Rev 1.2

200/200 is absolutly perfect for an RPI4. Sometimes 150MBit/s is the limit. 
You may disable SQM, but then the QoS will deteriorate.

Nickmcl024, 2021-02-09:
Best server is http://speedtest.tele2.net/1000GB.zip, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 1000G    0  996M    0     0  16.3M      0 17:23:47  0:01:00 17:22:47 17.0M
  0 1000G    0 1014M    0     0  16.3M      0 17:21:51  0:01:01 17:20:50 16.7M

  0 1000G    0 1033M    0     0  16.4M      0 17:19:35  0:01:02 17:18:33 16.8M

Adorfer, 2021-02-09:
for faster speeds look for a processor with AES-NI. And network cards on exclusive pcie lanes, not a sigle shared one (for all IO) like on the rpi.

Nickmcl024, 2021-02-09:
Hostname | OpenMPTCProuter
-- | --
Model | NF541 NF541
Architecture | Intel(R) Celeron(R) CPU J1900 @ 1.99GHz
Firmware Version | openmptcprouter v0.57.3 r0+15225-bfc433efd4 / LuCI Master git-20.339.75073-e54708a
Kernel Version | 5.4.83

did fresh install on a different system all stock settings just sent up the wans and vps 

speeds are 
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 1000G    0  605M    0     0  14.1M      0 20:07:31  0:00:42 20:06:49 18.1M
curl: (18) transfer closed with 1073106473049 bytes remaining to read

Nickmcl024, 2021-02-09:
are these setting corrects
MPTCP
Networks MPTCP settings. Visit http://multipath-tcp.org/pmwiki.php/Users/ConfigureMPTCP for help.
GLOBALS
Multipath TCP

enable
Multipath TCP checksum

disable
Multipath Debug

disable
Multipath TCP path-manager

default
Multipath TCP scheduler

default
Multipath TCP SYN retries
2
Congestion Control

bbr2

Default is bbr
Fullmesh subflows for each pair of IP addresses
1
Re-create fullmesh subflows after a timeout

enable
ndiffports subflows number
1
Interfaces Settings
LOOPBACK
Multipath TCP

disabled

One interface must be set as master
LAN
Multipath TCP

disabled

One interface must be set as master
WAN1
Multipath TCP

master

One interface must be set as master
WAN2
Multipath TCP

enabled

One interface must be set as master
OMRVPN
Multipath TCP

disabled

One interface must be set as master
OMR6IN4
Multipath TCP

disabled

One interface must be set as master

Ysurac, 2021-02-09:
Wait about 2 minutes for a test.
Check VPS internet speed and load average on both.

You seems to have disabled aggregation in MPTCP settings. Multipath TCP path-manager should be fullmesh.

Nickmcl024, 2021-02-09:
https://www.speedtest.net/result/10899616961.png

thats was it :) i never knew it had to be set to Multipath TCP path-manager should be fullmesh.

speeds are much better thanks 

how to i check vps loads and cpu . as well and loads on the pi4 

i did s speed test on the vps and it has 2000mbs so that side should be fine

Ysurac, 2021-02-09:
fullmesh is the default setting.
To get load average, you can use ```cat /proc/loadavg``` via SSH, but it's also available on status page.

Nickmcl024, 2021-02-09:
i promise i just did a fresh install on a pc and it was not on fullmesh  will now reset up the pi4 and test it with fullmesh , 
what is the best vpn setting to use for UDP streaming .

Nickmcl024, 2021-02-10:
Hey so sometimes on the raspberry pi 4 the file system says read only and one can't change anything, I have to flash the SD card agian and set it up all again is this a problem on the pi

Ysurac, 2021-02-10:
You can enable "Check filesystems before mount" in System->Mount Points
But best way is to use System->Shutdown to shutdown the RPI instead of removing power. RPI is very sensitive to power cut.

closed 2021-05-17 by github-actions[bot]

Bumpalert, 2021-05-18:
> i promise i just did a fresh install on a pc and it was not on fullmesh will now reset up the pi4 and test it with fullmesh ,
> what is the best vpn setting to use for UDP streaming .
@Nickmcl024 nice to know if you have some observations. I am also testing youtu.be streaming using 4g dongles

ithotline, 2021-07-24:
Is it possible to test omr-test-speed with smaller package from tele2.net? i want to test a 4g setup i'm working on but 1000GB would kill the 40GB dataplan i'm on.

Maybe 5MB test or something?

Ysurac, 2021-07-24:
Ctrl+C to stop test.