Ysurac/openmptcprouter #1577: No Keys after install

henry18hn, 2021-02-09:
## Expected Behavior =  Key issued after install to Vultr  & before reboot.

  = I should be able to receive key after install on Debian 10 vps

## Current Behavior
<I get error message > <No working domain detected not able to generate certificate for V2ray. You can set VPS_DOMAIN to a working domain if you want a certificate.>

## Specifications

  - OpenMPTCProuter version: <5.4.86>
  - OpenMPTCProuter VPS version: <0.1024-->
  - OpenMPTCProuter platform: <Debian 10 ,Vultr 1vCore 1024MB 25GB SSD /x86_64) -->

Ysurac, 2021-02-10:
And what you have at the end of VPS script ?

henry18hn, 2021-02-11:
<No working domain detected, not able to generate certificate for v2ray. You can set VPS_DOMAIN to a working domain if you want a certificate.>
   Thx, I must have accidentally deleted this part.

henry18hn, 2021-02-11:
Should I post a pic of my screen? I'm very new to all this.

Ysurac, 2021-02-11:
This is the latest message if it's not first time the script is launched.
Keys are in ```/root/openmptcprouter_config.txt```

henry18hn, 2021-02-11:
Thanks, I got it going now.

closed 2021-02-16 by henry18hn