Ysurac/openmptcprouter #1578: convert physical VPS to virtual VPS

Florent4223, 2021-02-10:
Hello

I have a VPS server under debian 10 but which is physical. It works very well but for reasons of ease to maintain I want to switch it to a KVM system.

I tried several procedures to transfer it to a VM, it starts but I have a problem with the network part.
Hence my question: can I go back to a clean instalaltion and transfer the keys (omr) to the new server?

thanks

Ysurac, 2021-02-10:
There is no easy way to transfer keys. But you can install VPS script and insert new key into the router.

spacedream, 2021-02-17:
another idea is not install on vps full linux+script but another openwrt image. cause its having patched same kernel, i guess. yes, its can be tricky but not impossible ;)

closed 2021-05-24 by github-actions[bot]