Ysurac/openmptcprouter #1579: [building from sources] ubnt-erx failed to build

Malaga82, 2021-02-11:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

I have an edgerouter-x and i really like to have OMR installed.
I saw yesterday a commit about edgerouter-x support so i wanted to test it.
i tried to build with instructions on related page using:
`OMR_TARGET="ubnt-erx" OMR_FEED_SRC="master" ./build.sh -j1 V=s`
There was a prompt asking things at some time, and leaved default answer ( bbr2 set to N )
But after many hours compiling i had a building error.

**Describe why you are asking for this support?**

building is failing

**Additional context**

Building error log:
```
mipsel-openwrt-linux-musl-ld: drivers/gpio/gpiolib-of.o: in function `of_gpio_export_probe':
gpiolib-of.c:(.text.of_gpio_export_probe+0x178): undefined reference to `__gpiod_export'
Makefile:1106: recipe for target 'vmlinux' failed
make[5]: *** [vmlinux] Error 1
make[5]: Leaving directory '/home/malaga/Downloads/openmptcprouter-develop/ubnt-erx/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.91'
Makefile:25: recipe for target '/home/malaga/Downloads/openmptcprouter-develop/ubnt-erx/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.91/.image' failed
make[4]: *** [/home/malaga/Downloads/openmptcprouter-develop/ubnt-erx/source/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.91/.image] Error 2
make[4]: Leaving directory '/home/malaga/Downloads/openmptcprouter-develop/ubnt-erx/source/target/linux/ramips'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/malaga/Downloads/openmptcprouter-develop/ubnt-erx/source/target/linux'
time: target/linux/install#521.93#82.55#610.66
    ERROR: target/linux failed to build.
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 1
make[2]: Leaving directory '/home/malaga/Downloads/openmptcprouter-develop/ubnt-erx/source'
target/Makefile:19: recipe for target '/home/malaga/Downloads/openmptcprouter-develop/ubnt-erx/source/staging_dir/target-mipsel_24kc_musl/stamp/.target_install' failed
make[1]: *** [/home/malaga/Downloads/openmptcprouter-develop/ubnt-erx/source/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/malaga/Downloads/openmptcprouter-develop/ubnt-erx/source'
/home/malaga/Downloads/openmptcprouter-develop/ubnt-erx/source/include/toplevel.mk:232: recipe for target 'world' failed
make: *** [world] Error 2
```

I'm not really skilled about openwrt compiling but i will do my best, in case of any tests.
Edgerouter-x is already a snapshot installed, so i just need to install the omr image then.
Thanks

Ysurac, 2021-02-11:
Support is still not finished. But this will be officially supported in next release.

Malaga82, 2021-02-11:
Ah, so it's more like "initial support", i was dreaming about testing it today, but nope 😢
Will check commits about it then.
Thanks

Ysurac, 2021-02-12:
The image can be build now, and it's build here: https://download.openmptcprouter.com/develop/5.4/ubnt-erx/targets/ramips/mt7621/ (based on develop branch, can be unstable)

Malaga82, 2021-02-12:
great!
will try it tomorrow, sadly i left the ubnt-x in my lab 😭 
i wanto to know if there is any issue with flow control.
[link1](https://forum.openwrt.org/t/mtk-soc-eth-watchdog-timeout-after-r11573/50000/30)
[link2](https://forum.openwrt.org/t/mt7621-mt7530-programming-disabling-flow-control-on-all-ports/76006)

While [HERE](https://forum.openwrt.org/t/ubiquiti-edgerouter-x-loading-openwrt-and-performance-numbers/27470) it's stated to enable HW offload for speed:

> 1. SQM works with software flow offload, allowing throughput of up to 200mbps with +3ms bufferbloat. Any thing faster maxes the CPU.
>2. HW flow offload resulting in 900mbps but the +200ms bufferbloat hurts
>3. NO HW flow offload, NO SQM, resulting in 550+ mbps with +80ms bufferbloat

Malaga82, 2021-02-13:
Hello,
gave it a try 👍 
installation ok, just had to sysupgrade twice to make installed but done.
As far as i tested, Hardware Offload has issues, if enabled i can't connect correctly to the VPS and after short time it reboots.
With Offload disabled and 3 WAN connected i have very poor performance:

- with NUC i have 90 to over 100 Mbps DOWNLOAD, 40/45 Mbps UPLOAD, ping 11-12 ms
- with ubnt-x i have 30/35 DOWNLOAD, 25/30 Mpbs UPLOAD, ping 13-14 ms

I can confirm that aggregation is working, i see all 3 WAN'S bandwidth usage in MPTCP section.

i saw these 2 processes hitting each one more than 20% CPU when doing a speedstest:
[image]
maybe the CPU is highly used:
[image]

Ysurac, 2021-02-13:
Processes are shadowsocks proxy. I think encryption use too much CPU.
For the error, I don't know, it's still develop branch, maybe an error somewhere ;)

Malaga82, 2021-02-13:
i used the default chacha20, i'll try to se to none if helps.

Malaga82, 2021-02-13:
```
OMR OpenMPTCProuter
Dashboard
Status
Overview
Firewall
Routes
System Log
Kernel Log
Processes
Realtime Graphs
vnStat Traffic Monitor
System
OpenMPTCProuter
Sysupgrade
System
Administration
ACL Settings
Software
Startup
Scheduled Tasks
Mount Points
Backup / Flash Firmware
Reboot
Shutdown
Services
OMR-Tracker
iPerf
Proxy DNS Over HTTPS
Shadowsocks-libev
OMR-Bypass
Packet Capture
E-Mail
Wake on LAN
Terminal
uHTTPd
UPnP
Recursive DNS
Network
MPTCP
Interfaces
DHCP and DNS
Hostnames
Static Routes
SQM QoS
Firewall
Diagnostics
OMR-DSCP
Quota
SNMPd
VPN
MLVPN
Glorytun TCP
Glorytun UDP
DSVPN
OpenVPN
Logout
System Log
Sat Feb 13 16:54:10 2021 daemon.info /usr/bin/ss-redir[9335]: tcp port reuse enabled
Sat Feb 13 16:54:10 2021 daemon.info /usr/bin/ss-redir[9335]: UDP relay enabled
Sat Feb 13 16:54:10 2021 daemon.info /usr/bin/ss-redir[9335]: udp port reuse enabled
Sat Feb 13 16:54:10 2021 daemon.info /usr/bin/ss-redir[9335]: running from root user
Sat Feb 13 16:54:10 2021 daemon.info /usr/bin/ss-local[9334]: using tcp fast open
Sat Feb 13 16:54:10 2021 daemon.info /usr/bin/ss-local[9334]: enable TCP no-delay
Sat Feb 13 16:54:10 2021 daemon.info /usr/bin/ss-local[9334]: initializing ciphers... none
Sat Feb 13 16:54:10 2021 daemon.info /usr/bin/ss-local[9334]: listening at 127.0.0.1:1111
Sat Feb 13 16:54:10 2021 daemon.info /usr/bin/ss-local[9334]: tcp port reuse enabled
Sat Feb 13 16:54:10 2021 daemon.info /usr/bin/ss-local[9334]: udprelay enabled
Sat Feb 13 16:54:10 2021 daemon.info /usr/bin/ss-local[9334]: udp port reuse enabled
Sat Feb 13 16:54:10 2021 daemon.info /usr/bin/ss-local[9334]: running from root user
[... 733 lines cut ...]
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-redir[13243]: enable TCP no-delay
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-redir[13243]: resolving hostname to IPv6 address first
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-redir[13243]: initializing ciphers... none
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-local[13242]: tcp port reuse enabled
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-local[13242]: udprelay enabled
Sat Feb 13 16:57:02 2021 user.info : using tcp fast open
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-redir[13243]: listening at :::1100
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-redir[13244]: enable TCP no-delay
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-redir[13243]: tcp port reuse enabled
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-redir[13244]: initializing ciphers... none
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-redir[13243]: UDP relay enabled
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-local[13242]: udp port reuse enabled
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-redir[13243]: udp port reuse enabled
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-redir[13244]: listening at 0.0.0.0:1101
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-redir[13243]: running from root user
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-redir[13244]: tcp port reuse enabled
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-redir[13244]: UDP relay enabled
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-local[13242]: running from root user
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-redir[13244]: udp port reuse enabled
Sat Feb 13 16:57:02 2021 daemon.info /usr/bin/ss-redir[13244]: running from root user
Sat Feb 13 16:57:02 2021 user.notice Shadowsocks: Rules DOWN
Sat Feb 13 16:57:04 2021 user.notice Shadowsocks: Rules UP
Powered by openmptcprouter v0.57.3 r0+15563-3bd8f660a4

```

Malaga82, 2021-02-13:
when trying to connect and it can't after some times it reboots.
for today the test finish here, back to the NUC :)
Will do more tests on work days.

Malaga82, 2021-02-13:
I have some time this evening and did some tests with new config.
if i have `When proxy shadowsocks is used, use it for UDP if VPN down` enabled shadowsocks get stucks and will never connect.
But the message `Can't get public IP address from ShadowSocks` pop up often, even if parameters are fine, after some time. Had to reboot for fixing if it doens't recover by itself.
All offloading must be OFF or sometimes there are random reboots.
When using wizard, passing IP and KEY, it fill shadowsocks, glotyrun values with wrong values, not the one of the server, had to insert manually.
And disabling encryption in wizard, in my previous installation i had "chacha" on, now it's off:
[image]
for now it's all :)

Malaga82, 2021-02-14:
updated to latest develop and can't get shadowsocks works fine.
[image]
Will downgrade tomorrow and check.

Ysurac, 2021-02-14:
Check State->System log.

Malaga82, 2021-02-14:
```
Sun Feb 14 20:34:12 2021 daemon.info /usr/bin/ss-redir[11879]: enable TCP no-delay
Sun Feb 14 20:34:12 2021 daemon.info /usr/bin/ss-redir[11879]: resolving hostname to IPv6 address first
Sun Feb 14 20:34:12 2021 daemon.info /usr/bin/ss-redir[11879]: initializing ciphers... none
Sun Feb 14 20:34:12 2021 daemon.info /usr/bin/ss-redir[11879]: listening at :::1101
Sun Feb 14 20:34:12 2021 daemon.info /usr/bin/ss-redir[11879]: tcp port reuse enabled
Sun Feb 14 20:34:12 2021 daemon.info /usr/bin/ss-redir[11879]: UDP relay enabled
Sun Feb 14 20:34:12 2021 daemon.info /usr/bin/ss-redir[11879]: udp port reuse enabled
Sun Feb 14 20:34:12 2021 daemon.info /usr/bin/ss-redir[11879]: running from root user
Sun Feb 14 20:34:12 2021 user.notice Shadowsocks: Rules DOWN
Sun Feb 14 20:34:14 2021 user.notice Shadowsocks: Rules UP
Sun Feb 14 20:34:14 2021 daemon.err omr-tracker[12165]: Flush terminated
Sun Feb 14 20:34:15 2021 user.notice Shadowsocks: Reload omr-bypass rules
Sun Feb 14 20:34:15 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Feb 14 20:34:19 2021 daemon.info omr-tracker-ss: Shadowsocks sss0 is down (can't contact via http 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 176.103.130.130, 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 176.103.130.130, 1.0.0.1, 212.27.48.10)
Sun Feb 14 20:34:22 2021 user.notice omr-bypass: Reload dnsmasq...
Sun Feb 14 20:34:23 2021 daemon.err omr-tracker[12166]: Command line is not complete. Try option "help"
Sun Feb 14 20:34:23 2021 daemon.err omr-tracker[12166]: Command line is not complete. Try option "help"
Sun Feb 14 20:34:24 2021 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Sun Feb 14 20:34:24 2021 daemon.info dnsmasq[6429]: read /etc/hosts - 4 addresses
Sun Feb 14 20:34:24 2021 daemon.info dnsmasq[6429]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sun Feb 14 20:34:24 2021 daemon.info dnsmasq-dhcp[6429]: read /etc/ethers - 0 addresses
Sun Feb 14 20:34:24 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Feb 14 20:34:24 2021 user.info : using tcp fast open
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-redir[14243]: enable TCP no-delay
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-redir[14243]: resolving hostname to IPv6 address first
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-redir[14243]: initializing ciphers... none
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-local[14242]: using tcp fast open
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-local[14242]: enable TCP no-delay
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-local[14242]: initializing ciphers... none
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-redir[14243]: listening at :::1100
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-redir[14243]: tcp port reuse enabled
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-local[14242]: listening at 127.0.0.1:1111
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-local[14242]: tcp port reuse enabled
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-local[14242]: udprelay enabled
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-redir[14243]: UDP relay enabled
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-local[14242]: udp port reuse enabled
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-redir[14243]: udp port reuse enabled
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-redir[14243]: running from root user
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-local[14242]: running from root user
Sun Feb 14 20:34:25 2021 user.info : using tcp fast open
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-redir[14244]: enable TCP no-delay
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-redir[14244]: resolving hostname to IPv6 address first
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-redir[14244]: initializing ciphers... none
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-redir[14244]: listening at :::1101
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-redir[14244]: tcp port reuse enabled
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-redir[14244]: UDP relay enabled
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-redir[14244]: udp port reuse enabled
Sun Feb 14 20:34:25 2021 daemon.info /usr/bin/ss-redir[14244]: running from root user
Sun Feb 14 20:34:25 2021 user.notice Shadowsocks: Rules DOWN
Sun Feb 14 20:34:26 2021 user.notice Shadowsocks: Rules UP
Sun Feb 14 20:34:27 2021 daemon.err omr-tracker[12163]: Flush terminated
Sun Feb 14 20:34:27 2021 user.notice Shadowsocks: Reload omr-bypass rules
Sun Feb 14 20:34:27 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Feb 14 20:34:34 2021 user.notice omr-bypass: Reload dnsmasq...
Sun Feb 14 20:34:35 2021 daemon.info dnsmasq[6429]: read /etc/hosts - 4 addresses
Sun Feb 14 20:34:35 2021 daemon.info dnsmasq[6429]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sun Feb 14 20:34:35 2021 daemon.info dnsmasq-dhcp[6429]: read /etc/ethers - 0 addresses
Sun Feb 14 20:34:35 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Feb 14 20:34:35 2021 daemon.err omr-tracker[12165]: Command line is not complete. Try option "help"
Sun Feb 14 20:34:35 2021 daemon.err omr-tracker[12165]: Command line is not complete. Try option "help"
Sun Feb 14 20:34:36 2021 user.info : using tcp fast open
Sun Feb 14 20:34:36 2021 daemon.info /usr/bin/ss-redir[16023]: enable TCP no-delay
Sun Feb 14 20:34:36 2021 daemon.info /usr/bin/ss-redir[16023]: resolving hostname to IPv6 address first
[... cut ...]
Sun Feb 14 20:38:29 2021 daemon.info /usr/bin/ss-redir[24630]: udp port reuse enabled
Sun Feb 14 20:38:29 2021 daemon.info /usr/bin/ss-redir[24630]: running from root user
Sun Feb 14 20:38:29 2021 user.notice Shadowsocks: Rules DOWN
Sun Feb 14 20:38:30 2021 user.notice Shadowsocks: Rules UP
Sun Feb 14 20:38:31 2021 daemon.err omr-tracker[12165]: Flush terminated
Sun Feb 14 20:38:32 2021 user.notice Shadowsocks: Reload omr-bypass rules
Sun Feb 14 20:38:32 2021 user.notice omr-bypass: Starting OMR-ByPass...
```

Malaga82, 2021-02-14:
just for info, upgrade from luci is not working, i updated from ssh with `sysupgrade -F /tmp/firmware.bin` note the -F switch. I can't upgrade without -F saying something about a mismatching firmware.

Malaga82, 2021-02-15:
sysupgraded to Feb 12th build ( first build ) and suddendly shadowsocks started to works fine, without touching anything.

extart, 2021-02-21:
> BTW with a speedtest all cpu cores are easly hitting 100% and often it disconnect from VPS, then when cores settle down a bit it reconnects fine.
> omr-tracker of wans is cpu intensive too, i see each of them take near 20%, for every wan.
Hardware encryption enabled? MT7621 supports hardware encryption. Maybe this is the problem?

Malaga82, 2021-02-21:
> Hardware encryption enabled? MT7621 supports hardware encryption. Maybe this is the problem?

What kind of hw encryption?
Other that chacha20 I can't select, I don't know what "other" mean.
There should be hw offloading but is not stable at all with that on.

Ysurac, 2021-02-21:
AES-256-GCM can be used, other is when an other encryption is set manually in shadowsocks.
AES is selected by default if router and VPS support AES-NI (```cat /proc/cpuinfo | grep aes```)

Adorfer, 2021-02-21:
> What kind of hw encryption?

https://deviwiki.com/wiki/MediaTek_MT7621

but seems to have issues: 
https://bugs.openwrt.org/index.php?do=details&task_id=2703

closed 2021-05-27 by github-actions[bot]