Ysurac/openmptcprouter #1580: Can't contact server admin script

luisglass, 2021-02-12:
This was working before and is not now. OMR router is no longer contacting with server. I've reinstalled and rebooted the server. Many errors of the following from Status - System Log:

Fri Feb 12 09:02:18 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)

From tail -f /var/log/syslog on vps server:

Feb 12 08:57:48 openmptcp-vps systemd[1]: Started OMR-Admin.
Feb 12 08:57:48 openmptcp-vps omr-service[24352]: parse error: Unmatched '}' at line 35, column 2
Feb 12 08:57:49 openmptcp-vps omr-admin.py[10923]: Traceback (most recent call last):
Feb 12 08:57:49 openmptcp-vps omr-admin.py[10923]:   File "/usr/local/bin/omr-admin.py", line 756, in <module>
Feb 12 08:57:49 openmptcp-vps omr-admin.py[10923]:     omr_config_data = json.load(f)
Feb 12 08:57:49 openmptcp-vps omr-admin.py[10923]:   File "/usr/lib/python3.7/json/__init__.py", line 296, in load
Feb 12 08:57:49 openmptcp-vps omr-admin.py[10923]:     parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)
Feb 12 08:57:49 openmptcp-vps omr-admin.py[10923]:   File "/usr/lib/python3.7/json/__init__.py", line 348, in loads
Feb 12 08:57:49 openmptcp-vps omr-admin.py[10923]:     return _default_decoder.decode(s)
Feb 12 08:57:49 openmptcp-vps omr-admin.py[10923]:   File "/usr/lib/python3.7/json/decoder.py", line 340, in decode
Feb 12 08:57:49 openmptcp-vps omr-admin.py[10923]:     raise JSONDecodeError("Extra data", s, end)
Feb 12 08:57:49 openmptcp-vps omr-admin.py[10923]: json.decoder.JSONDecodeError: Extra data: line 35 column 2 (char 1144)
Feb 12 08:57:49 openmptcp-vps systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
Feb 12 08:57:49 openmptcp-vps systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Feb 12 08:57:49 openmptcp-vps ss-server[24363]: server recv: Connection reset by peer
Feb 12 08:57:49 openmptcp-vps ss-server[24366]: server recv: Connection reset by peer
Feb 12 08:57:49 openmptcp-vps systemd[1]: omr-admin.service: Service RestartSec=100ms expired, scheduling restart.
Feb 12 08:57:49 openmptcp-vps systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 1.
Feb 12 08:57:49 openmptcp-vps systemd[1]: Stopped OMR-Admin.

Specifications
  - OpenMPTCProuter version:  v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: 0.1024, 5.4.86-mptcp on Debian 10
  - OpenMPTCProuter platform: PC Engines APU3, AMD GX-412TC SOC

Any help would be appreciated

Ysurac, 2021-02-12:
There is an issue in a json config.
Can you paste the result of ```/etc/openmptcprouter-vps-admin/omr-admin-config.json``` (without "user_password" entries) ?

luisglass, 2021-02-12:
Yes, here you go. 

`{
    "port": 65500,
    "users": [
        {
            "admin": {
                "username": "*****",
                "user_password": "*****",
                "permissions": "admin",
                "disabled": false
            },
            "openmptcprouter": {
                "username": "*****",
                "user_password": "*****",
                "shadowsocks_port": 65101,
                "disabled": false,
                "v2ray": {
                    "key": "50306195-84cf-4861-a60a-5721703fcc51",
                    "port": "65228"
                },
                "lanips": [
                    "192.168.100.1/255.255.255.0"
                ],
                "vpnremoteip": "10.255.255.2",
                "vpnlocalip": "10.255.255.1",
                "ula": "fd73:f887:473a::/48",
                "vpn": "glorytun_tcp"
            }
        }
    ],
    "debug": false,
    "allips": [],
    "ipv4": "52.183.30.89",
    "hostname": "52.183.30.89",
    "lastchange": 1612957368.542446
}}`

Ysurac, 2021-02-12:
There is an extra "}" at the end. Remove it.

luisglass, 2021-02-13:
That worked. Everything is green again.

closed 2021-05-19 by github-actions[bot]