Ysurac/openmptcprouter #1581: Localisation dans un autre pays avec VPS français

JeedHome44, 2021-02-12:
## Expected Behavior

 J'ai un VPS OVH dans le nord de la France mais pourtant, YouTube me met des publicités en espagnol. Mais la, le pire c'est que j'ai ouvert Google.com pour faire une recherche en wifi sur mon smartphone et je n'ai eu que des sites espagnols.

## Current Behavior

 Est-ce le VPN entre le routeur et le VPS qui pose problème?

## Specifications

  - OpenMPTCProuter version:  57.3
  - OpenMPTCProuter VPS version:  à jour
  - OpenMPTCProuter platform:  RPi 4B 2 Go

Ysurac, 2021-02-12:
Non. L'identification se fait au niveau de l'IP du VPS.
Possible que ça soit une plage d'IPs qui était avant localisée en espagne et les bases de Google ne sont pas à jour.

JeedHome44, 2021-02-12:
Merci pour la réponse rapide.

Y a-t-il un moyen pour dire définitivement que l'ip de mon VPS est en France et pourquoi la située sur la grande ville proche de chez moi?

C'est possible ça?

Adorfer, 2021-02-12:
that's an issue of your VPS hoster. And it's different on every VPS provider, if you have the choice of different ip localisations. Or even "how well respected" the AS/IP-range is. (e.g. if there are a lot of spammers and TOR exits in your virtual neighbourhood and you are blocked regularly by cloudflare etc.)

in other words: 
1) if you can not resolve it with OVH, take a different VPS somewhere else! 
2) openmptcprouter can not help you with that.

JeedHome44, 2021-02-13:
Est-il donc possible d'activer un VPN au lieu de l'IP externe du VPS et choisir le pays du VPN? Je suis abonné à CyberGhost et ca m'intéresse de pouvoir cacher mon IP publique par ce VPN mis je ne sais pas comment faire

Ysurac, 2021-02-13:
En installant un VPN sur le VPS, mais j'ai du mal à voir l'intérêt sauf si c'est pour contourner un bridage géographique.

JeedHome44, 2021-02-13:
Bah déjà, des fois mes Google home me disent que je ne suis pas dans le bon pays. Également, les pubs YouTube sont en espagnol ou autre.
C'est étrange quand même car le VPS OVH est bien marqué en France dans l'interface web du VPS.
Ou alors c'est OVH qui fait quelque chose sur le VPS qui ne devrait se faire...

JeedHome44, 2021-02-17:
OVH me dit que c'est sûrement Google qui n'est pas à jour avec RIPE.

Comment demander la mise à jour?

xillibit, 2021-02-20:
Il faut changer d'IP sur le VPS ou changer d'hébergeur

JeedHome44, 2021-02-21:
Bonjour,

J'ai trouvé un lien Google permettant de leur signaler que l'ip qu'ils considèrent en Espagne est en France.
OVH me confirme que mon IP est bien localisée en France. Ca ne sert donc à rien de changer d'IP car je peux retomber sur une autre qui sera encore considérée ailleurs par les services Google.

Pour le moment, j'ai trouvé une solution temporaire: je vais by-passer chaque GH de la maison et chaque smartphone, tablette, TV, Mi Box, console...

Par contre, puis je avoir une précision. Vous m'aviez dit qu'en faisant un By-pass sans préciser le wan, ca envoyait donc sur le maitre et si celui-ci est non dispo, sur l'autre. Mais si e souhaite envoyer sur le wan2 qui n'est pas le maitre, que se passe-t-il si le wan2 n'est plus dispo? OMR fait le By-pass sur le wan1 actif? J'aimerai vraiment pouvoir faire des by-pass "wan2 sinon wan1"

Merci

JeedHome44, 2021-02-21:
Si ca peut aider du monde, voici le lien google pour déclarer un ip localisé dans le mauvais pays:
https://support.google.com/websearch/workflow/9308722?hl=fr

Ysurac, 2021-02-22:
Si le bypass est vers le wan2 et que celui n'est plus dispo le trafic est bloqué.
Pas possible de préciser le wan que l'on veut en remplacement.

JeedHome44, 2021-02-22:
Est-ce possible de l'intégrer dans une future mise à jour?

JeedHome44, 2021-02-22:
Bon, j'ai tenté de passer le wan2 4G en mettre en faire des byPass pour mes 12 GH et d'autres appareils.
Le wan2 passe régulièrement à l'orange donc les GH n'aiment pas du tout.

Par contre, si je mets la Freebox en maître le wan 4G est très stable.

Il faudrait donc vraiment une règle dans les byPass qui permette donc de choisir l'ordre de priorité des wan. Si 1er non dispo, mettre sur le second, si pas dispo sur le troisième... tout en pouvant choisir un ordre différent pour chaque byPass

closed 2021-04-21 by JeedHome44