Ysurac/openmptcprouter #1582: How to expand OpenWRT Filesystem on the SD Card? [Question]

Jmmx1237, 2021-02-12:
My actual openmptcprouter configuration is working fine, but one thing is a bit confusing me.

How can i expand the filesystem to the whole SD card wich is represent the "Hard Disk", the openwrt installation is using about 1,82 GB disk space, after flashing the SD card with the image for a rasperry pi 4.

Please can you write a guide for this procedure in the documentation. Thanks for your work, this is a great projekt!

Ysurac, 2021-02-12:
There is a, "official" way indicated here: https://openwrt.org/docs/guide-user/additional-software/extroot_configuration
There is also some others available that you can find on google. But I never tested any.

Jmmx1237, 2021-02-12:
Thank you, i will this.

spacedream, 2021-02-13:
I wrote long ago some script to do this job for my gl-mifi router. so can be useful, but imho will be even *great* if you, @Ysurac can add this to your script, maybe with option "--upgrade" (or so). i'm sure you can wrote better code then me (im not a programmer, just copy/paste a little).
[build.sh.txt](https://github.com/Ysurac/openmptcprouter/files/5975710/build.sh.txt)
ty! <3

Jmmx1237, 2021-02-13:
good idea @spacedream, thanks for recommendation

spacedream, 2021-02-15:
i forgot to mentioned, you can also in openwrt gui backup your configs first, then unpack them in to "image-extras/etc/" folder before you build the image. then after run my script and its will upgrade firmware, reboot, expand the sd card and even install some packages (in my example was asterisk related). but you don't need to change your configs again after upgrade in that case.

closed 2021-05-21 by github-actions[bot]

andiohn, 2022-12-01:
To share what worked for me on OpenWRT 22.03.2 with SquashFS.

```
1. Flash image to SD as normal with the firmware selector (I added packages to mine)
2. ssh into router
3. "opkg update && opkg install cfdisk resize2fs"
4. "cfdisk /dev/mmcblk0"
5. Create a NEW PRIMARY partition in the free space at the end. Delete this new partition. Select /dev/mmcblk0p2 and RESIZE it. The default entries will take the whole space.
6. Reboot
7. ssh into router
8. "resize2fs /dev/loop0"
9. Success!
```

BigBobDalton, 2023-01-24:
I run into an error.  The cfdisk creates the correct size, utilizing the entire sdcard but 

resize2fs /dev/loop0

returns this error.

resize2fs 1.46.5 (30-Dec-2021)
resize2fs: Invalid argument while trying to open /dev/loop0
Couldn't find valid filesystem superblock.

> To share what worked for me on OpenWRT 22.03.2 with SquashFS.
>
> ```
> 1. Flash image to SD as normal with the firmware selector (I added packages to mine)
> 2. ssh into router
> 3. "opkg update && opkg [...]

sublime24, 2024-10-16:
> To share what worked for me on OpenWRT 22.03.2 with SquashFS.
>
> ```
> 1. Flash image to SD as normal with the firmware selector (I added packages to mine)
> 2. ssh into router
> 3. "opkg update && opkg [...]

Hello, I know this an old thread. But I wanted to ask about step 5. I'm confused in regards to what is created and deleted. I see create a NEW primary, then the instructions say to delete the new partition. When I try to resize mm...02 I get an error that it can only be 16m as it is. Any help is appreciated!

spacedream, 2024-12-15:
if you still interested to get help please show us what exactly you did and its responce from console, thanks.

Ryanhe521, 2025-02-11:
> To share what worked for me on OpenWRT 22.03.2 with SquashFS.
>
> ```
> 1. Flash image to SD as normal with the firmware selector (I added packages to mine)
> 2. ssh into router
> 3. "opkg update && opkg [...]

I had a 16gb SD card on Pi4 for openwrt.
I followed these steps but got the msg 

root@OpenWrt:~# resize2fs /dev/loop0
resize2fs 1.47.0 (5-Feb-2023)
The filesystem is already 98112 (1k) blocks long.  Nothing to do!

the space was not expanded.