Ysurac/openmptcprouter #1583: Orange 4G et limitation data

Ajowan84, 2021-02-13:
Bonjour,

Orange limite la quantité de donnée téléchargeable consécutive à 2Go. Le téléchargement est systématiquement coupé passé cette quantité de donnée.
La solution est d'utiliser le bypass, il semblerait que cette limitation s’effectue sur l'IP. Avec l'IP du VPS, la limitation disparait.
Le problème est qu'avec le bypass, je n'ai plus accès à Amazon vidéo ou a certains sites qui vérifient l'utilisation de VPN.

Existe t-il une autre solution que le bypass qui me permettrait  de corriger ces deux problèmes ?

Merci d'avance,

Ajowan84, 2021-02-13:
Changer de fournisseur VPS est la solution.
J'ai fait un test chez Firstheberg et plus aucun problème de géolocalisation et VPN...
J’étais chez OVH jusqu’à présent.

nicolasfrey, 2021-02-16:
J'ai pris un VPS chez https://www.pulseheberg.com et depuis je n'ai plus de problème de blocage. 

Par contre je ne comprends pas le problème des 2Go et orange. J'ai un agrégat ADSL free + 4G orange (sosh) et tout est parfait.

closed 2021-05-22 by github-actions[bot]