Ysurac/openmptcprouter #1584: USB connection sharing using an Android smartphone.

NicksQ69, 2021-02-13:
## Expected Behavior
My goal is to use my smartphone as a 4G modem and thus share the internet connection within the network to take advantage of aggregation.
I connected my smartphone to my Raspberry Pi 4 using a USB cable. This one is detected.
[image]

Next, i go to: System > OpenMPTCProuter > Configuration Wizard.
I configure the new "usb0" interface.
[image]

After this step, normally my aggregation system should benefit from the 4G connection in addition.

## Current Behavior
The aggregation system with the additional 4G does not work.
[image]

## Specifications

  - OpenMPTCProuter version: v0.57.3
  - OpenMPTCProuter VPS version: 0.1024
  - OpenMPTCProuter platform: Raspberry PI 4B (64-bits)

Ysurac, 2021-02-14:
How did you check that aggregation doesn't work ?

NicksQ69, 2021-02-14:
EN : I performed a throughput test using the speedtest.net website. With the aggregation of my 2 ADSL WANs, I get a downstream rate of about 20 mbps. When I test my internet speed with my smartphone alone on the 4G, I get a downlink speed of 30 mbps. After setting up with my two ADSL lines and my smartphone (4G), I have a downstream rate of about 20 mbps as before for my two ADSL lines.

FR : J'ai réalisé un test de débit à l'aide du site internet speedtest.net. Avec l'agrégation de mes 2 WAN ADSL, j'obtiens un débit descendant de 20 mbps environ. Lorsque je teste mon débit internet avec mon smartphone seul sur la 4G, j'ai un débit descendant de 30 mbps. Après la mise en place avec mes deux lignes ADSL et celle de mon smartphone (4G), j'ai un débit descendant d'environ 20 mbps comme précédemment pour mes deux lignes ADSL.

Ysurac, 2021-02-14:
Use ```omr-test-speed``` to test aggregated speed, and ```omr-test-speed wan1``` (for each interfaces), via SSH or Services->Terminal. Stop the speed test after at least 2 minutes using "Ctrl+C".

NicksQ69, 2021-02-14:
I had a very strange reaction.
Here are the screenshots before connecting my smartphone to the usb port as a modem :
[image]

[image]

[image]

And below are the screenshots when I plugged my smartphone on the usb port as a modem :
[image]

[image]

[image]

[image]

Ysurac, 2021-02-14:
You need to run test at least 2 minutes.
The syntax error is a know issue on some site that happen when a website answer with multiple IPs, it's solved in develop branch so this will be solved in next release.
For the last test, your connection seems to block ping. You should use another way to test if interface is up/down for this connection in omr-tracker. You can also check State->System log log about omr-tracker, it may say that connection is detected as down.

NicksQ69, 2021-02-14:
After adding it in OMR-Tracker, the result below (I did not understand everything) :
[image]

Ysurac, 2021-02-14:
It still use ping method in omr-tracker.

NicksQ69, 2021-02-14:
Yes that's right, I just corrected:
[image]

Ysurac, 2021-02-14:
httping doesn't seems to be better here... strange... you can disable omr-tracker for this connection only.

NicksQ69, 2021-02-14:
Anyway, no improvement ... do you have an idea how to solve this problem?

Ysurac, 2021-02-14:
You disabled omr-tracker on the connection only ? A least this should not switch off this connection.

NicksQ69, 2021-02-14:
Yes I removed the usb0 interface on OMR-Tracker.
I'll try to do some more tests, but I'm swimming in the unknown.

Ysurac, 2021-02-14:
removed ? It must be added and set to none.

NicksQ69, 2021-02-14:
Yes it's true, I did it by reflex.

NicksQ69, 2021-02-14:
I enabled the MPTCP over VPN option on the usb0 interface on the configuration wizard. And it seems to work now. I can't find the real download speed that my smartphone offers when I test the speed alone, but I gained like 6 mbps when downloading ... In my opinion the problem must come from my mobile provider.
[image]

[image]

Ysurac, 2021-02-14:
It's possible that it blovk MPTCP, but this seems strange.
Who is the provider ?

NicksQ69, 2021-02-14:
The provider is Bouygues Telecom.

Ysurac, 2021-02-15:
There is no know limit on Bouygues Telecom.
What is the router/dongle used ? Does it have any internal firewall ?

NicksQ69, 2021-02-15:
I tested on 3 other smartphones (based on Android) as a different brand modem, but we all have the same operator. And always the same problem.

Ysurac, 2021-02-15:
Maybe blocked by IPv6 APN ? Try to force APN mmsbouygtel.com to IPv4 only on your phone.

closed 2021-05-21 by github-actions[bot]