Ysurac/openmptcprouter #1585: How best to configure very asymmetric 3 connections

milankragujevic, 2021-02-13:
## Introduction
I have 3 connections. 

One is VDSL 20/2 Mbps, very stable, 5-8ms ping to VPS. 

Second is 100/8 Mbps cable (DOCSIS) but over a 2.8km 5GHz wireless link with MikroTik equipment. Connection is stable but takes time to "ramp up" (because the MCS is lowered if not in use for a long time). 
Ping to VPS is 10-20ms depending on the load on the coaxial networks. Ping on antennas is 1-2ms. 

Third is LTE limited to 100 GB per month, speed is around 30/30 Mbps, ping to VPS is 15-40ms depending on cell load and RF conditions.

## Expected Behavior
I want to have all download traffic over VDSL+DOCSIS and upload traffic mostly over DOCSIS and 4G.

## Explanation and reasoning
I have very fast download with cable, use DSL for realtime protocols (VoIP) and it's unmetered obviously because it's not mobile. However the upload speeds are pathetic. I have also 4G that has bad download speeds and is limited to 100 GB monthly, but of course it has great upload speeds compared to cable+DSL. 

So I want to move as much download traffic as possible over DSL+cable and move as much upload traffic over cable+4G (to keep DSL uncongested because download drops dramatically if upload utilization is >90% on VDSL due to physical layer specifics). 

**I am aware** it is impossible to move data in only one "direction" over a connection, but I want to avoid burning through the data cap for 4G for what can be done over cable and only use 4G for uploads which are more rare in my life, but it's important to do it quickly and not hog the line with uploads.

## Current Behavior
Currently I tried setting SQM on the LTE interface to 1024/30720 Kbps (so 1/30 Mbps DL:UL) and not using SQM on other interfaces, or setting them to their respective speeds, to basically try to limit the data flow by limiting speed on the interface and discouraging it's use for download with MPTCP. 

However, this creates instability (speeds vary and the connection "feels" slow and unstable) and a subpar experience compared to running with QoS set to 30/30 on LTE or without QoS or without LTE.

## Specifications

  - OpenMPTCProuter version: v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: 5.4.86-mptcp 0.1024
  - OpenMPTCProuter platform: x86_64 and also RPI3B

Ysurac, 2021-02-14:
You may try to use ```tc``` directly but not sure if this will give a better result than SQM.

milankragujevic, 2021-02-14:
Okay, but is the problem related to how MPTCP works and is implemented in OMR in any way, or is just the shaping that is the problem.

Because I do not know how it behaves when there are wildly different types of connections available.

Ysurac, 2021-02-14:
There is no problems with different types of connections.
I make all my tests using a very slow ADSL, fiber and a not so stable LTE connection.
SQM with LTE connections is know to make some problems.

closed 2021-02-23 by milankragujevic