Ysurac/openmptcprouter #1586: Can I use a router flashed with openmptcprouter as an aggregating VPS?

extart, 2021-02-14:
Can I use a router flashed with openmptcprouter and connected to a high-speed Internet channel as an aggregating VPS?
One router is at home, has a white IP, is connected to a high-speed channel. Another in the countryside. This is very convenient - providers will not block IP, no need to buy VPS.

Ysurac, 2021-02-14:
It's not a supported configuration.
This may work but need to manually set shadowsocks, vpn,...
The VPS side can be any x86_64 based computer.

extart, 2021-02-14:
> It's not a supported configuration.
> This may work but need to manually set shadowsocks, vpn,...
> The VPS side can be any x86_64 based computer.

The advantage of the router is that it is already installed at home and does not require money. It is also silent. Can you write what you need to configure and what settings to make?
This will be a universal and massive solution, without third-party VPS and problems with IP addresses.

extart, 2021-02-14:
If you bring the settings to the web interface, then everyone can configure such an installation by flashing it into the openmptcprouter router. If someone needs more speed, they can always upgrade to x86 ...

Ysurac, 2021-02-14:
It's not so easy, a shadowsocks server and a Glorytun TCP server need to be configured at least.
And many features will be missing because the VPS API is not available on the router.
There is some quite cheap barebones that would work really better as server side.

extart, 2021-02-14:
> It's not so easy, a shadowsocks server and a Glorytun TCP server need to be configured at least.
> And many features will be missing because the VPS API is not available on the router.
> There is some quite cheap barebones that would work really better as server side.

Ok, I'll take a look at the VPS installation scripts for now. If you have any questions, I will ask. I think a mass solution is better, with reduced functionality, with the possibility of expansion.

Nrakron, 2021-02-14:
> Can I use a router flashed with openmptcprouter and connected to a high-speed Internet channel as an aggregating VPS?
> One router is at home, has a white IP, is connected to a high-speed channel. Another in the countryside. This is very convenient - providers will not block IP, no need to buy VPS.

Support vps

extart, 2021-02-15:
> Support vps

Yes, no need to support VPS too.
As I say, a massive decision is needed. I bought a router, installed the firmware, pressed a few buttons in the web interface.
All. Otherwise, this beautiful project will first be available only to a select few, and then it will close.

closed 2021-05-21 by github-actions[bot]