Ysurac/openmptcprouter #1587: Burning hot router at 2 TB a day

rifaterdemsahin, 2021-02-14:
which process is overloading it ?

I suspect omr tracker right ?

[image]

Also asked here
https://twitter.com/rifaterdemsahin/status/1360881900655697922

Ysurac, 2021-02-14:
What is the problem ?

Ysurac, 2021-02-14:
CPU usage is low. If it's a Raspberry that is used, you can change "Scaling governor" in System->OpenMPTCProuter, "advanced settings" tab. But the main problem is a case ventilation/cpu heat sink issue.

rifaterdemsahin, 2021-02-14:
The protectci does not have an fan
I have installed one on top
As I have a daily 2 TB traffic
https://protectli.com/vault-6-port/

rifaterdemsahin, 2021-02-14:
i suspect the omr-trackers should be the main cpu users...is there a way I can see the overall usage of all processes in a month ?

Ysurac, 2021-02-14:
No.
Your problem is the traffic or hot CPU ?
omr-tracker only do ping to check if connections, proxy and VPN are UPs. It doesn't use much CPU or bandwidth.

Adorfer, 2021-02-15:
"burning hot": What is actually burning on the router? 
Or is it just the temperature? What is the value of that temperature?

hairydogltd, 2021-02-15:
It isn't clear (to me) what hardware you are running. That htop doesn't show that it is working at all hard, so I assume the problem is a lack of cooling, not an excess of heat. 
If it's a Pi4, I suggest the FLIRC cases: they have an aluminium heatsink that works well without the noise of a fan. The bottom is plastic, so I assume that WiFi and BT can get through (I've not used either on mine).
I have a different sort of PI case which works well, but its little fan makes an awful lot of noise in a household environment.

Adorfer, 2021-02-16:
> If it's a Pi4,

I do not think that the mentioned "protectli 6-port" is available with a broadcom core. 
But anyways, since we do not know a) what processor variant it is and b) what the acutual temperature is (and where this is meassured and how): perhaps it's totally normal!
I have some old Fujitsu Futro S900 who smell like an electrical space heater, after turning on the first time when summer is over. and they do that since several years.

spacedream, 2021-02-17:
> burning hot

i'm sure for fanless *that* type of powerfull machine to be hot is "normal" **by design**. 
to prove (or deny) that you can run regular OS on this machine for some time to prove it. 
so its not software problem but hardware itself, i'm sure 100%. 
ps: i will myself install big fan on top for sure, if i will have this piece of hardware. if noise is a problem then add speed control and lower speed of fan a little for optimal result. and problem must be disappear.

Adorfer, 2021-05-19:
this issue still has no problem description. 
- it totally lacks the metrik  (which material's [flash point](https://en.wikipedia.org/wiki/Flash_point) is meant by "burning hot"). Please use [SI units](https://en.wikipedia.org/wiki/International_System_of_Units)!
- "expected behaviour" vs "observed behavior": what would you expect
     - lower temperature
     - other throtteling methods (at same temperature, to mitigate impact of thermal throtteling)
     - less traffic than 2TB

closed 2021-08-24 by github-actions[bot]