Ysurac/openmptcprouter #1591: can´t contact Server Admin Script (No result - No API script Answer)

Roc2707, 2021-02-15:
## Expected Behavior
Hello, when running mptcprouter, all arrows turn green except for Direct Output which has an exclamation point triangle, which indicates the following problem: Unable to contact the server admin script (No result - no API script response)

## Current Behavior
The router is unable to contact the server

## Specifications

  - OpenMPTCProuter version: 0.57.3
  - OpenMPTCProuter VPS version: 0.1024 Debian 10
  - OpenMPTCProuter platform: x86_64 VirtualBox

Installation on the VPS took place through the command: 
wget -O - https://www.openmptcprouter.com/server/debian10-x86_64.sh | sh

Read several similar problems but no solution worked for me.

Ysurac, 2021-02-15:
Direct output mean this doesn't work, it use directly available connections.
Did you rebooted the VPS ?
What you have in State->System log about VPS ?

Roc2707, 2021-02-15:
VPS rebooted several times. Reinstalled several times. Nothing changes. 

> Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.258740] Atheros(R) L2 Ethernet Driver - version 2.2.3
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.259036] Copyright (c) 2007 Atheros Corporation.
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.259851] usbcore: registered new interface driver cdc_acm
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.260147] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.260801] usbcore: registered new interface driver cdc_wdm
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.261260] Loading modules backported from Linux version v5.8-0-gbcf876870b95
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.261654] Backport generated by backports.git v5.8-1-0-g79400d9e
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.266634] e100: Intel(R) PRO/100 Network Driver, 3.5.24-k2-NAPI
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.266947] e100: Copyright(c) 1999-2006 Intel Corporation
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.267814] e1000e: Intel(R) PRO/1000 Network Driver - 3.2.6-k
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.268115] e1000e: Copyright(c) 1999 - 2015 Intel Corporation.
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.273748] hso: drivers/net/usb/hso.c: Option Wireless
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.274058] usbcore: registered new interface driver hso
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.275344] usbcore: registered new interface driver ipheth
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.276711] usbcore: registered new interface driver kaweth
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.279143] ne2k-pci.c:v1.03 9/22/2003 D. Becker/P. Gortmaker
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.299121] pcnet32: pcnet32.c:v1.35 21.Apr.2008 tsbogend@alpha.franken.de
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.299797] pegasus: v0.9.3 (2013/04/25), Pegasus/Pegasus II USB Ethernet driver
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.300235] usbcore: registered new interface driver pegasus
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.304701] usbcore: registered new interface driver r8152
Mon Feb 15 14:24:07 2021 kern.warn kernel: [    8.306360] r8712u: module is from the staging directory, the quality is unknown, you have been warned.
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.310001] usbcore: registered new interface driver r8712u
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.310541] usbcore: registered new interface driver rtl8150
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.311706] sis900.c: v1.08.10 Apr. 2 2006
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.312507] sky2: driver version 1.30
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.315843] Solos PCI Driver Version 1.04
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.316670] tulip: Linux Tulip driver version 1.1.15-NAPI (Feb 27, 2007)
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.320864] usbcore: registered new interface driver usbserial_generic
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.321205] usbserial: USB Serial support registered for generic
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.322653] via_rhine: v1.10-LK1.5.1 2010-10-09 Written by Donald Becker
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.323570] usbcore: registered new interface driver visor
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.323869] usbserial: USB Serial support registered for Handspring Visor / Palm OS
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.324298] usbserial: USB Serial support registered for Sony Clie 5.0
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.324620] usbserial: USB Serial support registered for Sony Clie 3.5
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.374225] xt_ndpi v1.2 ndpi 3.5.0 IPv6=YES debug_message=no
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.374225]  BT: hash_size 0k, hash_expiation 0 sec, log_size 128kb
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.374225]  sizeof hash_ip4p_node=44 id_struct=256
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.374225]  flow_struct=2272 packet_struct=1416
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.374225]    flow_tcp_struct=112 flow_udp_struct=40 int_one_line_struct=16
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.374225]  ndpi_ip_addr_t=16 ndpi_protocol=4 nf_ct_ext_ndpi=176
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.374225]  flow_info=112 spinlock_t=4  NF_EXT_ID 7
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.375903] xt_ndpi: MAX_PROTOCOLS 320 LAST_PROTOCOL 254
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.376180] xt_ndpi: flow acctounting OFF
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.386678] xt_time: kernel timezone is -0000
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.387138] usbcore: registered new interface driver ark3116
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.387445] usbserial: USB Serial support registered for ark3116
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.390869] usbcore: registered new interface driver asix
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.391670] usbcore: registered new interface driver ax88179_178a
Mon Feb 15 14:24:07 2021 kern.info kernel: [    8.392183] usbcore: registered new interface driver belkin_sa
[... cut ...]
Mon Feb 15 14:26:15 2021 daemon.notice ttyd[27420]:   close signal: SIGHUP (1)
Mon Feb 15 14:26:15 2021 daemon.notice ttyd[27420]:   terminal type: xterm-256color
Mon Feb 15 14:26:15 2021 daemon.notice ttyd[27420]:  Using foreign event loop...
Mon Feb 15 14:26:15 2021 daemon.notice ttyd[27420]: Creating Vhost 'default' port 7681, 2 protocols, IPv6 off
Mon Feb 15 14:26:15 2021 daemon.notice ttyd[27420]:  Listening on port: 7681
Mon Feb 15 14:26:15 2021 daemon.notice procd: /etc/rc.d/S99v2ray: Flush terminated
Mon Feb 15 14:26:15 2021 daemon.info procd: - init complete -

Roc2707, 2021-02-15:
/var/log/daemon.log

> Feb 15 14:24:05 xxxxxxxxxx glorytun-tcp-run[3335]: INITIALIZED gt-tun0
Feb 15 14:24:05 xxxxxxxxxx glorytun-tcp-run[3335]: STARTED gt-tun0
Feb 15 14:24:05 xxxxxxxxxx glorytun-tcp-run[3335]: STOPPED gt-tun0
Feb 15 14:24:05 xxxxxxxxxx glorytun-tcp-run[3335]: STARTED gt-tun0
Feb 15 14:24:05 xxxxxxxxxx glorytun-tcp-run[3335]: STOPPED gt-tun0
Feb 15 14:24:05 xxxxxxxxxx systemd-networkd[190]: gt-tun0: Lost carrier
Feb 15 14:24:05 xxxxxxxxxx systemd[1]: Stopping Glorytun TCP on tun0...
Feb 15 14:24:05 xxxxxxxxxx systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Feb 15 14:24:05 xxxxxxxxxx systemd[1]: Stopped Glorytun TCP on tun0.
Feb 15 14:24:05 xxxxxxxxxx systemd[1]: Starting Glorytun TCP on tun0...
Feb 15 14:24:05 xxxxxxxxxx systemd-networkd[190]: gt-tun0: Gained carrier
Feb 15 14:24:05 xxxxxxxxxx systemd-udevd[8622]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Feb 15 14:24:05 xxxxxxxxxx systemd[1]: Started Glorytun TCP on tun0.
Feb 15 14:24:16 xxxxxxxxxx glorytun-tcp-run[8624]: 80.83.xxx.xxx.22207: connected
Feb 15 14:24:35 xxxxxxxxxx glorytun-tcp-run[8624]: 80.83.xxx.xxx.64427: connected
Feb 15 14:26:51 xxxxxxxxxx ss-server[11156]: getpeername: Transport endpoint is not connected
Feb 15 14:28:53 xxxxxxxxxx ss-server[11156]: getpeername: Transport endpoint is not connected

Ysurac, 2021-02-15:
Did you set an interface as master ?
I would need your configuration, available in last tab of System->OpenMPTCProuter (without VPS IP/key)

Roc2707, 2021-02-15:
wan1 selected as master.

> dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301,65011'
dscp.@classify[3].comment='OMR vpn'
dscp.@classify[4]=classify
dscp.@classify[4].direction='both'
dscp.@classify[4].proto='udp'
[... 1456 lines cut ...]
v2ray.omrtest.listen='127.0.0.1'
v2ray.omrtest.s_socks_auth='noauth'
v2ray.omrtest.s_socks_udp='1'
v2ray.omrtest.s_socks_ip='127.0.0.1'
v2ray.omrtest.s_socks_userlevel='0'
v2ray.main_reverse=reverse
v2ray.main_reverse.enabled='1'
v2ray.main_reverse.bridges='omrbridge|omr.lan'
v2ray.omrexit=outbound
v2ray.omrexit.protocol='freedom'
v2ray.omrexit.tag='out'
v2ray.omrrouting=routing_rule
v2ray.omrrouting.type='field'
v2ray.omrrouting.inbound_tag='omrbridge'
v2ray.omrrouting.outbound_tag='omrout_tunnel'
v2ray.omrrouting.domain='full:omr.lan'
v2ray.omrroutingo=routing_rule
v2ray.omrroutingo.type='field'
v2ray.omrroutingo.inbound_tag='omrbridge'
v2ray.omrroutingo.outbound_tag='out'
v2ray.main_routing=routing
v2ray.main_routing.enabled='1'
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'

Roc2707, 2021-02-16:
I added a second provider and made it master, but nothing changed. 
[image]
[image]

Ysurac, 2021-02-16:
There is at least a big latency problem.
Why are you using shadowsocks obfuscation ?

Roc2707, 2021-02-16:
Latency is usually not a problem. obusfaction is selected by default. 
If you select in the menu Advanced Settings-Enable ShadowSocks Obfuscating-OFF, it does not help. 
In addition, if you select in the menu Obfuscating plugin-Simple-obfs, then this also does not help.

Ysurac, 2021-02-16:
obfuscation is disabled by default. Set plugin to v2ray and uncheck the box.
Latency is usually a problem with MPTCP.

Roc2707, 2021-02-16:
I set plugin to v2ray and uncheck the box Enable ShadowSocks Obfuscating. It did not help.
After installation on the VPS through the script, there was a message at the end: 

> No working domain detected, not able to generate certificate for v2ray.
You can set VPS_DOMAIN to a working domain if you want a certificate.

Is this normal? 

Errors were found in the logs /var/log/daemon.log:

> Feb 16 12:18:30 vps ss-server[12195]: getpeername: Transport endpoint is not connected
Feb 16 12:19:23 vps ss-server[12195]: getpeername: Transport endpoint is not connected
Feb 16 12:20:22 vps ss-server[12195]: getpeername: Transport endpoint is not connected
Feb 16 12:21:19 vps ss-server[12195]: getpeername: Transport endpoint is not connected
Feb 16 12:22:30 vps ss-server[12195]: getpeername: Transport endpoint is not connected
Feb 16 12:23:25 vps ss-server[12195]: getpeername: Transport endpoint is not connected

Could this be a problem?

Ysurac, 2021-02-16:
The message is normal, this mean that https can't be used as obfuscation type.

Error are not normal, but this can be related to obfuscation.

Roc2707, 2021-02-16:
What can be done about this error? How to find the reason? If I turn off the router, then the errors on the VPS stop.
What is the password in the remote shadowsocks servers settings? I don't have it.

Ysurac, 2021-02-16:
You can force disable obfs manually via SSH or Services->Terminal on the router:
```
uci set shadowsocks-libev.sss0.obfs=0
uci set shadowsocks-libev.sss0.obfs_plugin='v2ray'
uci commit shadowsocks-libev
/etc/init.d/openmptcprouter-vps restart
```
All keys are in /root/openmptcprouter-config.txt on the VPS.

closed 2021-05-22 by github-actions[bot]