Ysurac/openmptcprouter #1592: QUIC/HTTP & UDP

iDustbin, 2021-02-15:
**Is your feature request related to a problem? Please describe.**

Hi, first your Software is really Awesome :) 👍 
actually QUIC Support is disabled.

Im thinking about: IP Passthrough & MAC-Addr. Spoofing.

**Describe the solution you'd like**
Whats about: 
Wireline ( IP ) IP Passthrough to all LTE Devices: 
and all LTE Network Devices get the same MAC-Address like wireline HWAddr. 
Did you ever tried todo this "ugly" workaround to test UDP/QUIC Connections ? 
I have an working ( h20 ) / QUIC Server inplace if you want to test. 
Do you think it is possible ?

Ysurac, 2021-02-15:
What is the problem ?
QUIC is disabled via firewall rules because UDP is slower than TCP on OpenMPTCProuter.

iDustbin, 2021-02-15:
Theres not a real Problem ! 
UDP Connections are Port Based and not IP based. 
QUIC/HTTP3.0 is also UDP based. 
Im not sure and its just conceptional Brainstorming.
Whats about you have only 1x Public ( Source ( openmptcprouter ) IP ) instead - 2x or 3x IPs.
Does it matter to UDP Connections ? 
Maybe an big buffer could also help to include the Future Web3.0 into your Software. 
Whats about Online Gaming ? Online Gaming is mostly UDP based. For inital Connections a few Gameserver support TCP Connections. 
When you contact GameServer with multiply Pub IPs it could be a problem. 
Call of Duty for example is only UDP based. 
Conceptional Request. Not an bug or something like this !

Ysurac, 2021-02-15:
By default, OpenMPTCProuter use Multipath TCP to aggregate connections, so it's TCP based. To aggregate UDP, by default, a TCP VPN is used, so it's slower than TCP here.
I will add Multipath QUIC when this will be usable and fast for OpenMPTCProuter usage.
V2Ray proxy can provide better UDP speed/latency, Glorytun UDP (that use own aggregation method) offer sometimes a better speed than default OpenMPTCProuter settings.

With OpenMPTCProuter, when you use the VPS part, you don't have multiple output IP, you have only VPS public IP as output.

closed 2021-02-15 by iDustbin