Ysurac/openmptcprouter #1593: Managed DNSMasq / TFTPBoot

iDustbin, 2021-02-15:
**Is your feature request related to a problem? Please describe.**
Hi,
im the Founder of Systronaut.de, Systronaut is based on DNSMasq and offer an Restful Backend for managed Operating System installations. 
I want to contribute with your OpenMPTCPRouter Software. 

**Describe the solution you'd like**
Systronaut will be installed on an Rasperry PI which is an normal Network Device on an stupid/unmanaged Switch. ( LAN - side )  
OpenMPTCPRouter have to disable the local DNSMasq or Relay to Managed DNSMasq ( Rasperry ) 
Systronaut is able to be the local DNS & TFTPServer for PXELinux/Syslinux Operating System installations.
NFS or SMB is also required. 
Do you see any way to disable DNSMasq or Relay DNS/TFTP Requests in your Software ? 
Best regards and thanks in advance
Dustbin

Ysurac, 2021-02-15:
In Network->DHCP&DNS, "DNS forwardings", you can transmit DNS queries to any DNS server.

iDustbin, 2021-02-15:
> In Network->DHCP&DNS, "DNS forwardings", you can transmit DNS queries to any DNS server.

sry my fault: i meant dhcp requests.

thanks

closed 2021-02-15 by iDustbin

Adorfer, 2021-02-16:
i really do not get it. 
Please use the template to describe a) what the scenario is b) what the expected/wanted behaviour is and c) what the current behaviour is.
(For me it sounds like "i can not decide which of the different methods i should describe. at the moment i now at least 5 different methods how to achieve what you ask for. But probably one one of them is what you really need)