Ysurac/openmptcprouter #1594: Plusieurs IP sur VPS

seb-65, 2021-02-16:
Bonjour,

Est-il possible sur le script d'installation du VPS de pouvoir gérer plusieurs box openmptcprouter ?

Par exemple mon VPS dispose de 5 IP publique, je souhaiterai savoir s'il est possible d'utiliser 5 box openmptcprouter avec chacune une IP publique qui sera associée depuis le VPS.

Merci de la  réponse

Adorfer, 2021-02-16:
additional: this would be REALLY nice to have the LAN-subnets of the OMRs routed between each others, to realize a site2site-vpn via this method.

CecilioGovain, 2021-02-23:
> Bonjour,
>
> Est-il possible sur le script d'installation du VPS de pouvoir gérer plusieurs box openmptcprouter ?
>
> Par exemple mon VPS dispose de 5 IP publique, je souhaiterai savoir s'il est possible [...]

Translated to English by Bing:
Good morning

Is it possible on the VPS installation script to be able to handle multiple openmptcprouter boxes?

For example my VPS has 5 public IPs, I would like to know if it is possible to use 5 openmptcprouter box with each a public IP that will be associated from the VPS.

Thanks for the response

Ysurac, 2021-02-23:
Support of multi OpenMPTCProuter and Lan 2 Lan is available via the API but not documented yet.
Duplicate of #1058

closed 2021-02-23 by Ysurac