Ysurac/openmptcprouter #1598: Concerns with DNS performance

milankragujevic, 2021-02-16:
I have noticed many times that sites load slowly while speed is very good and ping also. 

On speedtest.net I have ping 8ms and download 110 Mb/s upload 9 Mb/s (combined ~94/7.8 Mb/s and ~19/1.7 Mb/s connections). 

I think it is DNS because it happens that sites load in chunks, i.e. some parts do load some parts don't. Refreshing it fixes the problem. 

I am not sure whether this is due to problem with the stability of bonding or the fact DNS is UDP or something else. I have tried using both custom DNS and also the OMR gateway as DNS, it is the same. 

I have not touched any settings regarding DNS on OMR.

Malaga82, 2021-02-17:
well, just manually set DNS in one of your clients and check if sites load faster that way.
in DHCP/DNS you can set forward DNS resover, and there set like 8.8.8.8 or 1.1.1.1 or whatever you want to use.

milankragujevic, 2021-02-17:
I did.

> I have tried using both custom DNS and also the OMR gateway as DNS, it is the same.

Sadly it doesn't work as good as without OMR.

Jmmx1237, 2021-02-17:
I run my DNS via Pi-Hole which does not connect over the bond to the Internet, the DNS request directly connect over one of the WAN Router's. This is performing well and provide lower latency.

Jmmx1237, 2021-02-17:
An alternativ is resolve DNS via one of the WAN Router which directly connect to the internet... e.g. Fritzbox can host an dns service.

what do you mean with "sites load in chunks" exactly? This does not sound to DNS, is you download speed constant at bigger files?

milankragujevic, 2021-02-18:
> what do you mean with "sites load in chunks" exactly? This does not sound to DNS, is you download speed constant at bigger files?

Yes, download speed is constant. I mean exactly what I said, and if you ever had a DNS problem in your life it is obvious: 

For example, on YouTube thumbnails do not load but CSS does. On news sites, banners and ads do not load and some news pictures do and some don't. On some sites CSS is missing until refresh. Download speed, when downloading a file or running a speedtest, is constant at around 110 Mbps. 

This doesn't happen all the time but very often.

> An alternativ is resolve DNS via one of the WAN Router which directly connect to the internet... e.g. Fritzbox can host an dns service.

I know what you mean. I have not thought of this, I will try that today. Sounds genius and very simple. I had been trying to use OMR-Bypass before but it is buggy (so doesn't work often) in the latest release, a fix was acknowledged by Ysurac for next version.

Will try and report back info.

kams19, 2021-02-18:
the other thing that i can think of is, that as you are using direct WAN for resolving DNS, the response might be the specific to the ISP - think of CDN hosted on that ISP. try using PiHole on VPS and see if that makes the difference.

milankragujevic, 2021-02-18:
VPS is hosted at my ISP very close to me physically, so that ping from PC to VPS is ~5ms (on VDSL, on fiber it would be <1ms if I had fiber).

Resolving DNS on modem does _resolve_ (pun intended 😁) the problem. It works fine. 

For @Ysurac: maybe it is problematic that DNS is UDP and VPN is problematic or something? maybe look into it...

closed 2021-02-18 by milankragujevic