Ysurac/openmptcprouter #1599: ShadowSocks is not running and Cant get public IP from ShadowSocks

kams19, 2021-02-17:
Can you please help me.
[image]

system log:
Wed Feb 17 12:35:57 2021 daemon.err omr-tracker[11916]: Flush terminated
Wed Feb 17 12:35:57 2021 user.notice Shadowsocks: Reload omr-bypass rules
Wed Feb 17 12:35:57 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed Feb 17 12:35:59 2021 daemon.err omr-tracker[11916]: sed: to: No such file or directory
Wed Feb 17 12:36:00 2021 daemon.err omr-tracker[11916]: sed: to: No such file or directory
Wed Feb 17 12:36:02 2021 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 176.103.130.130, 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 176.103.130.130, 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 176.103.130.130, 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 176.103.130.130, 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 176.103.130.130, 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 176.103.130.130, 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 176.103.130.130, 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42)
Wed Feb 17 12:36:03 2021 daemon.err omr-tracker[11916]: sed: to: No such file or directory
Wed Feb 17 12:36:06 2021 daemon.err omr-tracker[11916]: sed: to: No such file or directory
Wed Feb 17 12:36:06 2021 daemon.err omr-tracker[11916]: sed: to: No such file or directory
Wed Feb 17 12:36:06 2021 daemon.err omr-tracker[11916]: sed: to: No such file or directory
Wed Feb 17 12:36:06 2021 daemon.err omr-tracker[11916]: sed: to: No such file or directory
Wed Feb 17 12:36:07 2021 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Wed Feb 17 12:36:08 2021 user.notice omr-bypass: Reload dnsmasq...
Wed Feb 17 12:36:08 2021 daemon.info dnsmasq[8603]: read /etc/hosts - 4 addresses
Wed Feb 17 12:36:08 2021 daemon.info dnsmasq[8603]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Wed Feb 17 12:36:08 2021 daemon.info dnsmasq-dhcp[8603]: read /etc/ethers - 0 addresses
Wed Feb 17 12:36:08 2021 user.notice omr-bypass: OMR-ByPass is running
Wed Feb 17 12:36:09 2021 user.info : using tcp fast open
Wed Feb 17 12:36:09 2021 user.info : using tcp fast open
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-redir[8493]: enable TCP no-delay
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-redir[8493]: resolving hostname to IPv6 address first
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-redir[8493]: initializing ciphers... chacha20-ietf-poly1305
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-local[8492]: using tcp fast open
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-local[8492]: initializing ciphers... chacha20-ietf-poly1305
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-redir[8494]: enable TCP no-delay
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-redir[8494]: resolving hostname to IPv6 address first
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-redir[8494]: initializing ciphers... chacha20-ietf-poly1305
Wed Feb 17 12:36:09 2021 user.notice Shadowsocks: Rules DOWN
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-redir[8493]: listening at :::1100
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-redir[8493]: tcp port reuse enabled
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-redir[8493]: UDP relay enabled
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-redir[8494]: listening at 0.0.0.0:1100
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-redir[8494]: tcp port reuse enabled
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-redir[8494]: UDP relay enabled
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-local[8492]: listening at 127.0.0.1:1111
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-local[8492]: tcp port reuse enabled
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-local[8492]: udprelay enabled
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-redir[8493]: udp port reuse enabled
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-redir[8493]: running from root user
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-redir[8494]: udp port reuse enabled
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-redir[8494]: running from root user
Wed Feb 17 12:36:09 2021 user.notice Shadowsocks: Rules UP
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-local[8492]: udp port reuse enabled
Wed Feb 17 12:36:09 2021 daemon.info /usr/bin/ss-local[8492]: running from root user
Wed Feb 17 12:36:09 2021 daemon.err omr-tracker[11913]: Flush terminated
Wed Feb 17 12:36:09 2021 user.notice Shadowsocks: Reload omr-bypass rules
Wed Feb 17 12:36:09 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed Feb 17 12:36:11 2021 daemon.err omr-tracker[11913]: sed: to: No such file or directory
Wed Feb 17 12:36:12 2021 daemon.err omr-tracker[11913]: sed: to: No such file or directory
Wed Feb 17 12:36:14 2021 daemon.err omr-tracker[11913]: sed: to: No such file or directory
Wed Feb 17 12:36:16 2021 daemon.err omr-tracker[11913]: sed: to: No such file or directory
Wed Feb 17 12:36:17 2021 daemon.err omr-tracker[11913]: sed: to: No such file or directory
Wed Feb 17 12:36:17 2021 daemon.err omr-tracker[11913]: sed: to: No such file or directory
Wed Feb 17 12:36:17 2021 daemon.err omr-tracker[11913]: sed: to: No such file or directory
Wed Feb 17 12:36:18 2021 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Wed Feb 17 12:36:18 2021 user.notice omr-bypass: Reload dnsmasq...
Wed Feb 17 12:36:18 2021 daemon.info dnsmasq[8603]: read /etc/hosts - 4 addresses
Wed Feb 17 12:36:18 2021 daemon.info dnsmasq[8603]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Wed Feb 17 12:36:18 2021 daemon.info dnsmasq-dhcp[8603]: read /etc/ethers - 0 addresses
Wed Feb 17 12:36:18 2021 user.notice omr-bypass: OMR-ByPass is running
Wed Feb 17 12:36:18 2021 user.info : using tcp fast open
Wed Feb 17 12:36:18 2021 daemon.info /usr/bin/ss-redir[11262]: enable TCP no-delay
Wed Feb 17 12:36:18 2021 daemon.info /usr/bin/ss-redir[11262]: resolving hostname to IPv6 address first
Wed Feb 17 12:36:18 2021 daemon.info /usr/bin/ss-redir[11262]: initializing ciphers... chacha20-ietf-poly1305
Wed Feb 17 12:36:18 2021 daemon.info /usr/bin/ss-local[11261]: using tcp fast open
Wed Feb 17 12:36:18 2021 user.info : using tcp fast open
[... 46 lines cut ...]
Wed Feb 17 12:36:28 2021 daemon.info /usr/bin/ss-redir[14095]: initializing ciphers... chacha20-ietf-poly1305
Wed Feb 17 12:36:28 2021 daemon.info /usr/bin/ss-local[14094]: initializing ciphers... chacha20-ietf-poly1305
Wed Feb 17 12:36:29 2021 user.notice Shadowsocks: Rules DOWN
Wed Feb 17 12:36:29 2021 user.notice Shadowsocks: Rules UP
Wed Feb 17 12:36:29 2021 daemon.info /usr/bin/ss-redir[14096]: listening at 0.0.0.0:1100
Wed Feb 17 12:36:29 2021 daemon.info /usr/bin/ss-redir[14096]: tcp port reuse enabled
Wed Feb 17 12:36:29 2021 daemon.info /usr/bin/ss-redir[14096]: UDP relay enabled
Wed Feb 17 12:36:29 2021 daemon.info /usr/bin/ss-redir[14095]: listening at :::1100
Wed Feb 17 12:36:29 2021 daemon.info /usr/bin/ss-redir[14095]: tcp port reuse enabled
Wed Feb 17 12:36:29 2021 daemon.info /usr/bin/ss-redir[14095]: UDP relay enabled
Wed Feb 17 12:36:29 2021 daemon.info /usr/bin/ss-redir[14096]: udp port reuse enabled
Wed Feb 17 12:36:29 2021 daemon.info /usr/bin/ss-redir[14096]: running from root user
Wed Feb 17 12:36:29 2021 daemon.info /usr/bin/ss-redir[14095]: udp port reuse enabled
Wed Feb 17 12:36:29 2021 daemon.info /usr/bin/ss-redir[14095]: running from root user
Wed Feb 17 12:36:29 2021 daemon.err omr-tracker[11916]: Flush terminated
Wed Feb 17 12:36:29 2021 user.notice Shadowsocks: Reload omr-bypass rules
Wed Feb 17 12:36:29 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed Feb 17 12:36:31 2021 daemon.err omr-tracker[11916]: sed: to: No such file or directory

Troubleshooting:
Restarted OpenMPTCP
Restarted VPS
Changed master interface

Let me know if you need any other info

Ysurac, 2021-02-17:
I need the configuration, in System->OpenMPTCProuter, last tab (without VPS IP/Key).

kams19, 2021-02-17:
> OpenMPTCProuter

[config.txt](https://github.com/Ysurac/openmptcprouter/files/5995892/config.txt)
@Ysurac please see attached

Ysurac, 2021-02-17:
Please copy/paste in issue, I don't download file.

kams19, 2021-02-17:
l

> Please copy/paste in issue, I don't download file.

sure, done that, please check

Ysurac, 2021-02-17:
Check that shadowsocks is running on VPS side using  ```ps aux | grep ss``` and check if you have errors in ```/var/log/daemon.log```.

kams19, 2021-02-17:
> Check that shadowsocks is running on VPS side using `ps aux | grep ss` and check if you have errors in `/var/log/daemon.log`.
Here is what i can see -

user1@localhost:~$ ps aux | grep ss
root        751  0.0  0.3   7676  1536 ?        Ss   Feb16   0:00 /usr/bin/ss-manager -c /etc/shadowsocks-libev/manager.json --manager-address 127.0.0.1:8839
root        764  0.0  1.8  17248  8544 ?        Ss   Feb16   0:21 ss-server --manager-address 127.0.0.1:8839 -f /root/.shadowsocks/.shadowsocks_65101.pid -c /root/.shadowsocks/.shadowsocks_65101.conf --acl /etc/shadowsocks-libev/local.acl -t 1000 -u --fast-open --no-delay --mptcp -6 -s 0.0.0.0
message+    844  0.0  0.3   8696  1708 ?        Ss   Feb16   0:00 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only
root        974  0.0  0.5  15848  2748 ?        Ss   Feb16   0:00 /usr/sbin/sshd -D
root      81566  0.0  1.6  16648  7852 ?        Ss   14:23   0:00 sshd: user1 [priv]
user1   81773  0.0  0.9  16648  4724 ?        S    14:23   0:00 sshd: user1@pts/0
user1    85694  0.0  0.1   8600   824 pts/0    S+   14:26   0:00 grep ss

user1@localhost:~$ sudo tail -150  /var/log/daemon.log
Feb 17 14:22:48 localhost omr-service[841]: RTNETLINK answers: File exists
Feb 17 14:22:48 localhost omr-service[841]: RTNETLINK answers: File exists
Feb 17 14:22:48 localhost omr-service[841]: RTNETLINK answers: File exists
Feb 17 14:22:53 localhost dhclient[610]: XMT: Solicit on ens192, interval 109690ms.
Feb 17 14:22:53 localhost dhclient[610]: RCV: Advertise message on ens192 from fe80::250:56ff:feaf:ff68.
Feb 17 14:22:53 localhost dhclient[610]: RCV: Advertise message on ens192 from fe80::250:56ff:febb:223f.
Feb 17 14:22:58 localhost ss-server[764]: getpeername: Transport endpoint is not connected
Feb 17 14:23:02 localhost omr-service[841]: /usr/local/bin/omr-service: line 80: echo: write error: Device or resource busy
Feb 17 14:23:02 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:23:02 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:23:02 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:23:02 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:23:02 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:23:02 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:23:02 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:23:02 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:23:02 localhost omr-service[841]: RTNETLINK answers: File exists
Feb 17 14:23:02 localhost omr-service[841]: RTNETLINK answers: File exists
Feb 17 14:23:02 localhost omr-service[841]: RTNETLINK answers: File exists
Feb 17 14:23:07 localhost systemd[1]: Created slice User Slice of UID 1000.
Feb 17 14:23:07 localhost systemd[1]: Starting User Runtime Directory /run/user/1000...
Feb 17 14:23:07 localhost systemd[1]: Started User Runtime Directory /run/user/1000.
Feb 17 14:23:07 localhost systemd[1]: Starting User Manager for UID 1000...
Feb 17 14:23:07 localhost systemd[81758]: Listening on GnuPG network certificate management daemon.
Feb 17 14:23:07 localhost systemd[81758]: Listening on GnuPG cryptographic agent and passphrase cache.
Feb 17 14:23:07 localhost systemd[81758]: Reached target Timers.
Feb 17 14:23:07 localhost systemd[81758]: Listening on GnuPG cryptographic agent and passphrase cache (restricted).
Feb 17 14:23:07 localhost systemd[81758]: Reached target Paths.
Feb 17 14:23:07 localhost systemd[81758]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
Feb 17 14:23:07 localhost systemd[81758]: Listening on GnuPG cryptographic agent and passphrase cache (access for web browsers).
Feb 17 14:23:07 localhost systemd[81758]: Reached target Sockets.
Feb 17 14:23:07 localhost systemd[81758]: Reached target Basic System.
Feb 17 14:23:07 localhost systemd[1]: Started User Manager for UID 1000.
Feb 17 14:23:07 localhost systemd[1]: Started Session 201 of user kams19.
Feb 17 14:23:07 localhost systemd[81758]: Reached target Default.
Feb 17 14:23:07 localhost systemd[81758]: Startup finished in 82ms.
Feb 17 14:23:16 localhost omr-service[841]: /usr/local/bin/omr-service: line 80: echo: write error: Device or resource busy
Feb 17 14:23:16 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:23:16 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:23:16 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:23:16 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:23:16 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:23:16 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:23:16 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:23:16 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:23:16 localhost omr-service[841]: RTNETLINK answers: File exists
Feb 17 14:23:16 localhost omr-service[841]: RTNETLINK answers: File exists
Feb 17 14:23:16 localhost omr-service[841]: RTNETLINK answers: File exists
Feb 17 14:23:30 localhost omr-service[841]: /usr/local/bin/omr-service: line 80: echo: write error: Device or resource busy
[... cut ...]
Feb 17 14:25:10 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:25:10 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:25:10 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:25:10 localhost omr-service[841]: Error: Device can not be enslaved while up.
Feb 17 14:25:10 localhost omr-service[841]: RTNETLINK answers: File exists
Feb 17 14:25:10 localhost omr-service[841]: RTNETLINK answers: File exists
Feb 17 14:25:10 localhost omr-service[841]: RTNETLINK answers: File exists

kams19, 2021-02-17:
@Ysurac 
I have re run the VPS script and rebooted it. The Shadowsocks error goes away but i get an error exclaimation next to the VPS under the status page, and it seems the traffic is still going direct and not via VPS.
[image]

My VPS is set to auto update critical debian patches and the following updates have been done in the last few days, do you think this can cause the issue?

2021-02-03 06:10:02,809 INFO Packages that will be upgraded: linux-image-amd64 linux-libc-dev
2021-02-05 06:25:07,159 INFO Packages that will be upgraded: libldap-2.4-2 libldap-common
2021-02-07 06:12:13,062 INFO Packages that will be upgraded: base-files ca-certificates file grub-common grub-pc grub-pc-bin grub2-common iproute2 libgnutls30 libmagic-mgc libmagic1 libpam-systemd libsystemd0 libudev1 systemd systemd-sysv tzdata udev unzip
2021-02-11 06:40:08,778 INFO Packages that will be upgraded: libzstd1

Ysurac, 2021-02-17:
Maybe, what you get via "uname -a" on the VPS ?

kams19, 2021-02-17:
> uname -a

Linux localhost 5.4.86-mptcp #1 SMP Mon Jan 4 16:33:37 UTC 2021 x86_64 GNU/Linux

Ysurac, 2021-02-17:
So no problem here.
It seems to be a problem with shadowsocks only.
What do you have on the VPS, in ```/etc/shadowsocks-libev/manager.json``` ? (remove keys)

kams19, 2021-02-17:
> /etc/shadowsocks-libev/manager.json

cat /etc/shadowsocks-libev/manager.json
{
    "server": "0.0.0.0",
    "port_key": {
        "65101": "###########################"
    },
    "local_port": 1081,
    "mode": "tcp_and_udp",
    "timeout": 1000,
    "method": "chacha20-ietf-poly1305",
    "verbose": 0,
    "ipv6_first": true,
    "prefer_ipv6": false,
    "fast_open": true,
    "no_delay": true,
    "reuse_port": true,
    "mptcp": true,
    "ebpf": 0,
    "acl": "/etc/shadowsocks-libev/local.acl"

do you see any issue here?

Ysurac, 2021-02-17:
If there is a "}" at the end of the file that was not pasted, no. Else it's missing a "}"
Did you try another master interface ?

kams19, 2021-02-17:
There is  a  "}" this was omitted in my reply. Also i have tried making the other interface master - but still the issue persists.
do you think i will have to rebuild OpenMPTCP router?

Ysurac, 2021-02-17:
You can try in System->OpenMPTCProuter, "advanced settings" checkbox, to use v2ray instead of shadowsocks as proxy.
If this work, the issue is really shadowsocks here, else there is another issue.

kams19, 2021-02-17:
> You can try in System->OpenMPTCProuter, "advanced settings" checkbox, to use v2ray instead of shadowsocks as proxy.
> If this work, the issue is really shadowsocks here, else there is another issue.

did that, but its still the same issue :(
[image]

system logs:
Wed Feb 17 19:25:58 2021 daemon.err /usr/bin/ss-redir[10716]: send: Connection reset by peer
Wed Feb 17 20:00:00 2021 cron.err crond[5721]: USER root pid 5311 cmd /etc/init.d/vnstat_backup backup
Wed Feb 17 20:00:20 2021 user.notice post-tracking-post-tracking: Check API configuration...
Wed Feb 17 20:00:25 2021 user.notice post-tracking-post-tracking: Check API configuration... Done
Wed Feb 17 20:11:22 2021 daemon.err /usr/bin/ss-redir[10716]: server recv: Host is unreachable
Wed Feb 17 20:11:38 2021 daemon.err /usr/bin/ss-redir[10716]: server recv: Host is unreachable
Wed Feb 17 20:20:14 2021 user.notice MPTCP: Flush route cache
Wed Feb 17 20:20:17 2021 daemon.info v2ray: DNS disabled: main_dns
Wed Feb 17 20:20:17 2021 daemon.info v2ray: Setting transparent proxy on port: 1897
Wed Feb 17 20:20:17 2021 daemon.info v2ray: Transparent proxy mode: default
Wed Feb 17 20:20:17 2021 user.notice v2ray: add rules
Wed Feb 17 20:20:17 2021 user.notice v2ray: Reload omr-bypass rules
Wed Feb 17 20:20:17 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed Feb 17 20:20:21 2021 user.notice omr-bypass: Reload dnsmasq...
Wed Feb 17 20:20:21 2021 daemon.info dnsmasq[9543]: read /etc/hosts - 4 addresses
Wed Feb 17 20:20:21 2021 daemon.info dnsmasq[9543]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Wed Feb 17 20:20:21 2021 daemon.info dnsmasq-dhcp[9543]: read /etc/ethers - 0 addresses
Wed Feb 17 20:20:21 2021 user.notice omr-bypass: OMR-ByPass is running
Wed Feb 17 20:20:21 2021 daemon.info v2ray[28046]: V2Ray 4.31.0 (V2Fly, a community-driven edition of V2Ray.) OpenWrt R1 (go1.15.2 linux/amd64)
Wed Feb 17 20:20:21 2021 daemon.info v2ray[28046]: A unified platform for anti-censorship.
Wed Feb 17 20:20:22 2021 daemon.info v2ray[28046]: 2021/02/17 20:20:22 [Info] v2ray.com/core/common/platform/ctlcmd: <v2ctl message>
Wed Feb 17 20:20:22 2021 daemon.info v2ray[28046]: v2ctl> Read config:  /var/etc/v2ray/v2ray.main.json
Wed Feb 17 20:20:23 2021 user.notice MPTCP: Add route 192.168.1.0/24 via 192.168.100.212 dev eth0
Wed Feb 17 20:20:24 2021 user.notice Shadowsocks: Rules DOWN
Wed Feb 17 20:20:24 2021 user.notice MPTCP: Flush route cache
Wed Feb 17 20:20:24 2021 daemon.notice netifd: Network device 'tun0' link is down
Wed Feb 17 20:20:24 2021 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Wed Feb 17 20:20:24 2021 daemon.notice netifd: Interface 'omrvpn' is now down
Wed Feb 17 20:20:24 2021 daemon.info glorytun[31122]: STOPPED tun0
Wed Feb 17 20:20:24 2021 daemon.debug dnsmasq[9543]: stopped listening on tun0(#26): 10.255.255.2 port 53
Wed Feb 17 20:20:24 2021 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Wed Feb 17 20:20:24 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Wed Feb 17 20:20:24 2021 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Wed Feb 17 20:20:24 2021 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed Feb 17 20:20:24 2021 user.notice omr-tracker: Launching...
Wed Feb 17 20:20:29 2021 user.notice omr-tracker: Launched
Wed Feb 17 20:20:29 2021 user.notice post-tracking-post-tracking: omrvpn down
Wed Feb 17 20:20:29 2021 daemon.info vnstatd[31601]: Info: SIGTERM received, exiting.
Wed Feb 17 20:20:29 2021 daemon.err glorytun[28753]: the name server returned a failure
Wed Feb 17 20:20:29 2021 user.notice post-tracking-post-tracking: Glorytun VPN down, restart it
Wed Feb 17 20:20:29 2021 daemon.info vnstatd[29637]: Info: vnStat daemon 2.6 started. (pid:29637 uid:0 gid:0 64-bit)
Wed Feb 17 20:20:29 2021 daemon.info vnstatd[29637]: Info: Monitoring (2): eth2 (1000 Mbit) eth1 (1000 Mbit)
Wed Feb 17 20:20:29 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Wed Feb 17 20:20:29 2021 user.notice post-tracking-post-tracking: Disable Pi-Hole...
Wed Feb 17 20:20:29 2021 daemon.info dnsmasq[9543]: exiting on receipt of SIGTERM
Wed Feb 17 20:20:29 2021 daemon.info v2ray: DNS disabled: main_dns
Wed Feb 17 20:20:29 2021 daemon.info dnsmasq[29872]: started, version 2.82 cachesize 150
Wed Feb 17 20:20:29 2021 daemon.info dnsmasq[29872]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Wed Feb 17 20:20:29 2021 daemon.info dnsmasq-dhcp[29872]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Wed Feb 17 20:20:29 2021 daemon.info dnsmasq[29872]: using only locally-known addresses for domain test
Wed Feb 17 20:20:29 2021 daemon.info dnsmasq[29872]: using only locally-known addresses for domain onion
Wed Feb 17 20:20:29 2021 daemon.info dnsmasq[29872]: using only locally-known addresses for domain localhost
Wed Feb 17 20:20:29 2021 daemon.info dnsmasq[29872]: using only locally-known addresses for domain local
Wed Feb 17 20:20:29 2021 daemon.info dnsmasq[29872]: using only locally-known addresses for domain invalid
Wed Feb 17 20:20:29 2021 daemon.info dnsmasq[29872]: using only locally-known addresses for domain bind
Wed Feb 17 20:20:29 2021 daemon.info dnsmasq[29872]: using nameserver 127.0.0.1#5353
[... cut ...]
Wed Feb 17 20:23:09 2021 daemon.info dnsmasq[8712]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Wed Feb 17 20:23:09 2021 daemon.info dnsmasq-dhcp[8712]: read /etc/ethers - 0 addresses
Wed Feb 17 20:23:13 2021 user.notice MPTCP: Flush route cache
Wed Feb 17 20:23:24 2021 user.notice MPTCP: Add route 192.168.1.0/24 via 192.168.100.212 dev eth0
Wed Feb 17 20:23:24 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Wed Feb 17 20:23:28 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0

kams19, 2021-02-18:
@Ysurac can you think of anything else that I can try before rebuilding the VPS.

Also is there any easy way of updating OMR-Bypass or just doing back up of OMR Bypass,
I dont want to do full back-up and restore for the rebuild.

kams19, 2021-02-18:
just to update on any future reader - i had to rebuild the OpenMPTCP router to fix this issue!
[image]

closed 2021-02-18 by kams19