Ysurac/openmptcprouter #1600: How to setup with EasyTether (which does not allow ping?)

Fffrank, 2021-02-18:
Looking to use EasyTether as a usb connection but it does not allow ping.... Is it possible to set up and still detect if the connection is up/down?

vinceducat, 2021-02-20:
no need of any apps on android..

search in the issue for Thethering....

vinceducat, 2021-02-20:
https://github.com/Ysurac/openmptcprouter/issues?q=is%3Aissue+tethering

vinceducat, 2021-02-20:
https://github.com/Ysurac/openmptcprouter/issues/1584

Fffrank, 2021-02-22:
EasyTether prevents your provider from detecting tethered data.....

Ysurac, 2021-02-22:
You can try DNS or HTTP mode in omr-tracker.

vinceducat, 2021-02-22:
@Fffrank 

> EasyTether prevents your provider from detecting tethered data.....

could you explain ? OMR create VPN over the link

mrconnerton, 2021-04-22:
I too would like to use EasyTether as one of my wan sources. Easytether provides the ability to use 4g lte data as "phone data" and not "hotspot" data that android tethering would do.

closed 2021-07-27 by github-actions[bot]