Ysurac/openmptcprouter #1602: Control of Download and Upload rate

rnunes77, 2021-02-18:
## Expected Behavior

I am using 3 4G modems (individually powered) from different operators and all are recognized by the system. However, I realized that during use the balancing occurs in a strange way, honoring only one of the modems and leaving the other two almost unused. I would like to place a limit to see if this way the modems work more together, adding a link, bringing redundancy and load balancing.
## Current Behavior

I need to add links, have redundancy and load balancing
## Specifications

  - OpenMPTCProuter version: 57.3
  - OpenMPTCProuter VPS version: 0.1024
  - OpenMPTCProuter platform: RPI3

Ysurac, 2021-02-18:
Wan connections are used based on latency and only when needed. It's not load balancing at all but aggregation.
In Network->MPTCP, "MPTCP scheduler" you can choose redundancy or load balancing mode that may give what you want but will be slower and use more data.

rnunes77, 2021-02-18:
Thanks for listening! I would like to use for video transmission, I usually traffic videos with 6mbps bitrate, in your opinion what would be the best configuration to achieve a stable connection?

Ysurac, 2021-02-18:
If you use UDP for transmission, then you try others VPN (like Glorytun UDP) in System->OpenMPTCProuter, "advanced settings" checkbox

rnunes77, 2021-02-19:
[image]

Today I tried to change to UDP, but I started having problems with the VPN. I went back to GLORYTUN TCP and while using it I received poor connection information.

closed 2021-05-25 by github-actions[bot]

Bumpalert, 2021-06-02:
> 
> 
> If you use UDP for transmission, then you try others VPN (like Glorytun UDP) in System->OpenMPTCProuter, "advanced settings" checkbox

@Ysurac  did you mean to enable "When proxy shadowsocks is used, use it for UDP if VPN down" setting for udp for streaming?

[image]

Ysurac, 2021-06-02:
No. I mean what I write. In wizard page, using advanced settings **checkbox**