Ysurac/openmptcprouter #1603: Bandwidth issue

jpradelle, 2021-02-18:
## Expected Behavior

Stable bandwhidth

## Current Behavior

I made following test : disable VPS and enable only ADSL stable connection :
[image]

With direct ADSL connection without openmtcprouter I get stable speed tests running at 4.4Mb/s
With openmtcprouter and ADSL as direct output speed starts at 4.4Mb/s and decrease with the time to around 1MB/s

Here is a screen shot during speed test, it starts download at 4.4Mb/s and after few secondes speed decreases.
[image]

When using multipath with ADSL and 4G connection I can reach up to 25Mb/s so I think I can eliminate problems like bad wifi and so on

Using omr-test-speed with the setup of only ADSL as direct output I get a stable download speed of 550Ko/s (4.4Mb/s) :
[image]

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: (not used here) 5.4.74-mptcp 0.1022
  - OpenMPTCProuter platform: VirtualBox on AMD A4-5000 APU with Radeon(TM) HD Graphics

Ysurac, 2021-02-18:
You can try in Network->MPTCP, another TCP Congestion Control and MPTCP scheduler.

jpradelle, 2021-02-18:
Thank you for the answer, I tried all congestion control and scheduler but I'm getting similar results with all, still with the issue.
I also tried to set "Multipath TCP" to disabled, same result

jpradelle, 2021-02-18:
I'm wondering if there is something wrong with my VM network setup. I spent lot of time trying to have something working, not sure this is the expected setup: the computer hosting the VM is linked via ethernet to ADSL router and via WiFi to the 4G router, connected on both networks.
In virtalbox I created 3 network interfaces (3 as bridge, 82545EM, promiscuous allow all):
- 1st linked to ethernet interface
- 2nd linked to same ethernet interface too
- 3rd linked to wifi interface

In OMR
- 1st is used as lan for 192.168.100.1
- 2nd is used as WAN1 with a static IP on ADSL router
- 3rd is used as WAN2 with a static IP on 4G routeur

Is it the expected setup ? Or is wrong and it could introduce this instability ?

Ysurac, 2021-02-19:
You can set SQM on ADSL connection.
Wifi is never good on 4G connection, because Wifi can loose packets and 4G too.

jpradelle, 2021-02-23:
I know WiFi is bad, but I have a cheap 4G router without ethernet as this was planned to be a very temporary solution.

I made setup from scratch several times, I managed to fix the issue by changing setup:
Create only 2 interfaces on virtualbox, 1 for Ethernet, the other for WiFi. Configure then in OMR ADSL based on ethernet interface using MacVLAN and WiFi interface as bridge.
On the host, connected it to OMR router and connect it to WiFi too so connection can be bridged.
But after I did not get full speed of 4G (WiFi) and sometimes I completely loose connection, on OMR interfaces are streaming at full speed, I think I have routing loop or something like that.

Then I took time to find back my Raspberry PI 3B+, setup is much easier, and work exactly as expected !
Thanks a lot for this awesome project !

closed 2021-02-23 by jpradelle