Ysurac/openmptcprouter #1605: Virtual dynamic interface no ipv6 address

changloong, 2021-02-19:
Unlike Openwrt,  IPv6 Virtual dynamic interface  only has IPv6-PD, no IPv6 address.

I try last version with X86,  `openmptcprouter.settings.disable_ipv6=0`.

I need it has   IPv6 address for public access.

Ysurac, 2021-02-19:
If you use MacVLAN it's normal: it's exactly the same problem with DHCP.

changloong, 2021-02-19:
I use same setting like OpenWrt, vlan normal wan.   Is there a workaround to fix the problem ?

Is the mptcp script remove the address or some kernel config disable it get address ?

Ysurac, 2021-02-19:
True, IPv6 support is disabled on wan interfaces in current release.
WAN IPv6 support will be available in next release with full VPS IPv6 connections support.

closed 2021-05-25 by github-actions[bot]