Ysurac/openmptcprouter #1606: daemon.notice procd: /etc/rc.d/S90mptcp: Device 'bonding_masters' can't found!

jfh75, 2021-02-19:
## Expected Behavior
Bonding masters would be found 

## Current Behavior
I have following notice I never think I saw before : 
Fri Feb 19 07:41:17 2021 daemon.notice procd: /etc/rc.d/S90mptcp: Device 'bonding_masters' can't found!
Fri Feb 19 07:41:17 2021 daemon.notice procd: /etc/rc.d/S90mptcp: Use the hardware name like in ifconfig
It is a long time I have not doing speedtest and I get 55 Mb down and 24Mb up. Sometimes with previous releases I was reaching 185 Mb down. With one ADSL 10Mb and one 4G. Today I have one 4G more. 
Testing My 4G new Huawei router, it reaches 150 Mb when directly connected to its LAN port.
Is something not working with bonding ?
[image]

## Specifications

  - OpenMPTCProuter version:  57.3
  - OpenMPTCProuter VPS version: 1024
  - OpenMPTCProuter platform:

Ysurac, 2021-02-19:
Bonding-master is related to OpenVPN bonding VPN.
You use same ISP for both LTE connection, antenna can have a limited bandwidth.
To test speed, use ```omr-test-speed``` for aggregated speed and ```omr-test-speed eth1```... for each interfaces speed, via SSH or Services->Terminal (ctrl+c to stop after at least 2 minutes).

jfh75, 2021-02-19:
Thank you for you quick answer.
My question is is there something not working I should fix on a software point of view or should I ignore it as every lights are green ? If device "bonding master not found" happens, is it working then or not working ?
Thank you for your help,

JFH

Ysurac, 2021-02-19:
It's not a problem, the device really doesn't exist. I will disable this log info in next release.
All lights are green.

jfh75, 2021-02-19:
Ok thank you, indeed it is reported as info, not error. Yes i know for the 2 4G's ISIMS on the same ISP/antenna, it is stupid, desperate to get some bandwith over here ;-)
JFH

closed 2021-02-19 by jfh75