Ysurac/openmptcprouter #1607: Gateway down USB m.2 modem

techdeisel, 2021-02-19:
## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform:

henry18hn, 2021-02-22:
On mine it changed my Eth2 to eth2.1 during reboot, I went to the Wizard and changed protocol til I found the correct one. 
O am also using 2 Goldenorb/Rooter modems. O see you using ModemManger protocol, I'm using, DHCP. What changes using Modem Manager?

techdeisel, 2021-02-22:
> 
> 
> On mine it changed my Eth2 to eth2.1 during reboot, I went to the Wizard and changed protocol til I found the correct one.
> O am also using 2 Goldenorb/Rooter modems. O see you using ModemManger protocol, I'm using, DHCP. What changes using Modem Manager?

not sure i cant use static or dhcp with a usb modem tho i think

techdeisel, 2021-02-22:
> 
> 
> > On mine it changed my Eth2 to eth2.1 during reboot, I went to the Wizard and changed protocol til I found the correct one.
> > O am also using 2 Goldenorb/Rooter modems. O see you using ModemManger protocol, I'm using, DHCP. What changes using Modem Manager?
> 
> not sure i cant use static or dhcp with a usb modem tho i think,are you using a rpi4?and you using dhcp wiht usb modem?

using rpi4 yes cant use anything but modem manager

closed 2021-05-28 by github-actions[bot]