Ysurac/openmptcprouter #1608: VPS chez soi ?

tboome33, 2021-02-19:
Bonjour,
j'ai un premier site où je bénéficie de la Fibre (local pro) et un second site (domicile) où je peux agréger de l'adsl et de la 4G. Je voudrais installer le VPS sur le 1er site (pour faire l'économie du vps, pour avoir une ip grand public et plus une ip blacklistée un peu partout). Comment procéder pour configurer le VPS sur une le 1er site ? Une machine virtuelle avec Debian 10 dessus ? Comment la configurer ? Quelles redirections à faire entre la box et la machine virtuelle ? J'ai un peu de mal à voir par où commencer ...
Merci pour vos idées....

Roland

Ysurac, 2021-02-19:
Oui une Debian 10 avec le script d'installation VPS lancé et redirection des ports > 65000 de la box vers la machine.

tboome33, 2021-02-20:
Ok merci c'est une bonne piste. Est-ce qu'un Rasberry Pi 4 peut tenir la route pour cet usage, est-ce assez puissant ? Sinon si j'ai bien compris, je connecte mon Rasberry Pi et je configure mon Debian 10 pour avoir une IP locale données par le serveur DHCP de ma Box Orange Fibre. Ensuite dans la box, je paramètre une redirection de tous les ports >65000 vers l'IP de mon Rasberry. C'est bien ça ?

Sinon est-ce qu'il y a un moyen que les deux réseaux locaux, celui à mon domicile et celui chez moi puissent se voir ? 

Désolé pour toutes ces questions mais je débute dans tout ça.

Merci

Ysurac, 2021-02-20:
Le raspberry n'est pas supporté niveau VPS car le noyau patché est uniquement pour x86_64.
Il y a moyen que les réseaux puissent se voir en mettant la route qui va bien sur l'OMR chez toi et pareil de l'autre côté.

tboome33, 2021-02-20:
ah mince. Bon est une VM sous vmware c'est envisageable ? Est-ce qu'il existe une image disque pour Vmware à télécharger avec toute ce qui va bien dessus (configurer un Linux c'est tout nouveau pour moi)?

Ysurac, 2021-02-20:
Oui c'est envisageable, mais non pas d'image toute prête pour le VPS. Il faut installer ou prendre une image Debian 10.

tboome33, 2021-02-21:
Bonjour, je progresse. J'ai installé une version Ubuntu sous Vmware. J'ai installé la version Desktop (je sais c'est pas bien mais j'avais pas trop le temps de chercher à configurer tout ça à la main, je ferai ça dans un second temps). J'ai ensuite fixé l'adresse ip de la machine ubuntu en fonction de l'adresse MAC sur la Box Orange et j'ai redirigé les ports de 65000 à 65535. J'ai installé la version serveur de OpenMPTCProuter avec le script d'installation puis sur un Rasberry Pi j'ai tout configuré. Tout semble fonctionner parfaitement, les débits sous nPerf montrent que les connexions sont bien agrégées mais j'ai des message d'erreurs sur toutes mes connexion ! Que peut-il bien se passer ?

[image]

tboome33, 2021-02-21:
Le fait que les deux réseaux locaux coté serveur sur mon lieu pro et coté client à mon domicile ont la même plage d'adresse  peut mettre la pagaille ?

closed 2021-05-27 by github-actions[bot]