Ysurac/openmptcprouter #1609: Setup OMR with 1 WAN link (initially)

bdruth, 2021-02-20:
## Expected Behavior
Find documentation for setting up OMR with single WAN connection

## Current Behavior
No documentation found (wizard expects two WAN connections)

## Specifications

  - OpenMPTCProuter version: v0.57.3-r0
  - OpenMPTCProuter VPS version: debian10
  - OpenMPTCProuter platform: rpi-4

I'm expecting a 2nd WAN link to be installed this week - but I'd like to get started with setting up OMR before then because I have time now and I would like to get everything working "as is" so that the only thing I'm changing is adding an additional WAN link once it's setup. The wizard requires two WAN connections, so I'm not sure how best to start with just a single WAN link. I imagine it's trivial, but I'm new to OMR, so I'm not sure where to start.

Thanks!

techdeisel, 2021-02-21:
setup like you wud for two disable wan2 in  omr settings until you need it

vinceducat, 2021-02-21:
just "add" the second wan via the wizard ...;-)

bdruth, 2021-02-21:
K, thx, will try!

henry18hn, 2021-02-22:
I started mine with 1 wan.

hairydogltd, 2021-02-23:
You can ignore the second WAN or you can delete it. The wizard won't mind.
I ran with one at first, to see how it was.

closed 2021-03-17 by bdruth