Ysurac/openmptcprouter #1612: Streaming Blocked

henry18hn, 2021-02-21:
## Expected Behavior
Stream online movies using my paid account.

## Current Behavior
Netflix, Amazon, and ESPN.com etc blocked from streaming.
Services>OMRBypass>Domains I added every variance of ESPN.com I could see on my screen >trying to connect..... in both Interface Eth0 which is my LAN and unspecified and none worked. Also AmazonPrime.com, Netflix.com. I have not tried to access anything I dont have a subscription to.
OMRBypass>MAC Address> I added the MAC address to my television I was trying to stream these on.
OMRBypass>Protocols and services> I added Disney Plus (ESPN parent company) and Amazon Video from menu but would not work.
I tried accessing these on my pc as well.
Youtube did play the 1 video I tried. 
I see a DNS Zones available in Vultr/Debian10 server I have not enabled, dont know if I should.

## Specifications

  - OpenMPTCProuter version: 0.57.3
  - OpenMPTCProuter VPS version: 0.1024  5.4.86-mptcp Vultr Debian 10
  - OpenMPTCProuter platform: Rpi4b

Ysurac, 2021-02-21:
Did you use OpenMPTCProuter IP as DNS ?
What you mean by "interface eth0" ? If you try to exit via lan interface in omr-bypass this can't work.

henry18hn, 2021-02-21:
Do you mean change the DNS settings in the television to my Openmptcrouters  ip address?, no i did not but I can if I should.

Services>OMRBypass> I should pick which WAN i want to use , not "eth0". Thanks I shouldve been able to figure that out.

henry18hn, 2021-02-21:
Services \OmrBypass correcting interfaces worked , thanks.

closed 2021-02-21 by henry18hn